fesch/CanZE #527: Character set/font is too big?

Albino-Zoe, 2019-02-07:
In V1.29 are suddenly in different screen different font sizes!
Is that a problem that I can solve on the smartphone (SM-G930F; Android 9.0.0) or has something gone wrong here?
[image]
[image]
[image]

fesch, 2019-02-07:
I changed the fixed size font to some pdi-related untis.
The problem is, that some screen are meant to be looked at in landscape, while others are more for portrait usage. So this is why some of the text is to long. Please rotate your phone ;)

yoh-there, 2019-02-09:
@fesch are the properties of the widgets, at least the ones in the layout xmls, programatically changeable and would the widgets play nice with that? If so, would that be an option? TBH, I myself run the consumption screen almost always and in portrait. It would be kind of nice if this could be faciliated if it is not crazy much work

fesch, 2019-02-10:
Orientation can be detected, so a simple if statement should make the trick to make a difference between portrait and landscape...

fesch, 2019-02-10:
Fixed in the latest release.
Feel free to re-open if you think the font in portrait mode is still too big!

closed 2019-02-10 by fesch