fesch/CanZE #530: Clipping bug in the graphs?

yoh-there, 2019-02-16:
I am not 100% sure where the issue is. Looks like on the widgets nothing is drawn when either the value is above the scale or exactly on the top (maybe also bottom) scale. Here are a few examples. It is most noticeable at the gradients, but it might be a common graphing problem.

[image]
(middle graph)

[image]
(bottom graph)

Let me know if there is something I can do please.

fesch, 2019-02-18:
I got this now twice too. First thought about a problem of the new value limiting code I've added, but ended it in rejecting this idea, because missing values, which we know from other cases, result in big triangles so a simple empty space should not be there.

So finally this seams to be a drawing problem and I need to investigate into that direction.

A workaround, increasing the scale and does avoiding getting value right on the limits of the graph will probably do the trick as well ;)

yoh-there, 2019-02-18:
Scaling would work for the delta, but not for the power. I will try to recreate this using the http dongle emulator with stable, fixed values and will let you know.

Speaking about triangles: I still see them occasionally. Do you think it is caused by Nan's?

yoh-there, 2019-02-18:
I think I found it. A regression of trying to get rid of the triangles, combined with faulty boolean logic. I will test and push if fixed.

yoh-there, 2019-02-18:
Tested. #a92e956 Closing

commit a92e956c (2019-02-18): Fixed #530 clipping bug. Code reformat messes the diff, rea change line 477 and 366
closed 2019-02-18 by yoh-there