fesch/CanZE #531: Improvement for Can ZE

AngeloCalvo, 2019-02-17:
I´d like to suggest you an upgrade do Can ZE, that i saw on Fluence APP and it will be grater to have it on CanZe:
- Totalizer of Kwh
- Number of slow charges
- Number of quick charges
- Number of full charges
- Number of partial charges  
Thanks, and i hope you continue the great job.

yoh-there, 2019-02-17:
Totalizer I can understand. I will see if such a field exists. I tried finding it for energy taken from the grid and could not find a reliable one. For consumption I can have another look.

I don't understand why those counts would be interesting. Why is it?

AngeloCalvo, 2019-02-17:
Hello,

 

I send you a snapshot of the Fluence ZE APP:

 

 

 

It will be in many ways, if you buy na used car you could know how the car been treated, if you buy a new car you can see everytime you want, how much Kwh have you charge, and how …

 

Thanks,

Ângelo

Enviada: 17 de fevereiro de 2019 20:51
Para: fesch/CanZE
Cc: AngeloCalvo; Author

yoh-there, 2019-02-18:
Fair enough. I'll see if these fields are present in the ZOE too.

yoh-there, 2019-02-20:
Total kWh:  7bb.2161.120 *
Quick counter: 7bb.2166:16
Normal counter: 7bb.2166.32
Full counter: 7bb.2166.48 *
Partial counter: 2166.64 *

For my Q210 ZOE, full and partial give relevant numbers, but normal is just the sum of thos and quick is nothing, even if I charge it sometimes @43 kW. So what I might do is put those 3 number in the Charging History screen. There is enough room there and it makes sense.

AngeloCalvo, 2019-02-20:
I that what you tell will be very nice on CanZe.

I was thinking of a new improvment if the user could define the price for
kW the APP could tell how much i spend last month, last year and from the
beggining...

Thanks
Ângelo

yoh-there, 2019-02-20:
Haha, no no, weŕe not getting there. Then we get discussions over which tariff to use to which date, etc. and indeed history graphs.

But your original request is done. Closing commit #9ce57c5

closed 2019-02-20 by yoh-there

AngeloCalvo, 2019-02-24:
Dear Sirs,

 

I’ve just update my CanZe APP and saw the new menu that i thanks, but in my car (ZOE Q210) i can´t see the number of slow charges (PCN) and the number of fast chargers (PCR) :

 

 

What i saw on FLuence APP:

 

 

The need for this information is to know how the battery was treated, because they like more of slow charges than fast charges…

 

Thanks.

 

Best regards

Ângelo

Enviada: 20 de fevereiro de 2019 22:42

yoh-there, 2019-02-25:
Dear Angelo,

I retried just now, using after re-installing from the Play store, both with the new CanSee dongle and my KONNWEI dongle, on my Q210. Both show the new fields. See this link for an example https://canze.fisch.lu/new-release-coming-up-4/

Make sure the car is "on" please.

AngeloCalvo, 2019-02-25:
Dear Sirs,

I think i don't explain correctly, i send you an image of My CanZe and It
show the three new subjects, but It don't show how much rápid charges and
how much slow charges that have been done, like show on Fluente APP (send a
Snapshot too)...

Thanks.

Regards
Ângelo

yoh-there, 2019-02-25:
See my previous comment

> For my Q210 ZOE, full and partial give relevant numbers, but normal is just the sum of thos and quick is nothing, even if I charge it sometimes @43 kW. So what I might do is put those 3 number in the Charging History screen. There is enough room there and it makes sense.

The numbers just don't make sense for quick and slow.

The snapshots somehow didn't save on github.

BTW, you are not correct slow charging is better for your battery. Fast charging is actually better, but it needs very good thermal management (which the ZOE has).

AngeloCalvo, 2019-02-25:
Dear Sir,

I respect your opinnion, but on Portugal, i belong to a Facebook group of
Renault Zoe, and everybody says that if possible is better to charge at low
speed, and stay between 20 to 80%, if possible of course. I sorry that you
couldn't join this information on CanZe, but thanks for what you have done.

Regards,
Ângelo

fesch, 2019-02-25:
Still being at 105% SOH and nearly 22kW/h capacity during the last 5 years, I can't complain about my 3.7kW/h slow charger that much ...

AngeloCalvo, 2019-02-25:
Its ok, but if you use a 43kw (rapid) charger almost the time you maybe
have 90% ir less in your battery SOH...

yoh-there, 2019-02-25:
@fesch Mine is worse than yours (still pretty good, and I am almost always charging 3.7 kW. QED).

@AngeloCalvo Forums are echo chambers of opinions, but that doesn't mean it is true. I apologize if that sounds arrogant. But let's indeed agree to disagree.

As for the counters, again, as I said, at least on my ZOE, while I do fast charge occasionally, that counter is on 1, and the slow counter is is the same as the partial+full (minus 1), so while I can fetch the same values as in FluenceSpy, the numbers are totally useless at least on my ZOE.

@fesch could you do me a big favor and make an LBC All Data dump of yours (assuming you once in a while charge on 43 kW) and check for string 21_66_03? I am totally happy to add the two fields, but not if their values are misleading, which they are for my Q210.

AngeloCalvo, 2019-02-25:
Ok i understand that if numbers are not correct It can't be use on APP.

Thanks

fesch, 2019-02-25:
@yoh-there : The fastest I did was 22kW ... sorry :-/

yoh-there, 2019-02-25:
No worries. I checked with somebody else who does almost exclusively fast charging. Same result.

@AngeloCalvo you can get the data in a bit more complicated way, DTC > LBC > All data, then search the log file starting with LBC in the Cance folder for "Quick". I you do have meaningful results, let us know please.