fesch/CanZE #534: Charging prediction now shows 1kw dc power

fgmccrea, 2019-02-20:
After version 1.29 my charge prediction changed. 
With a 41kwh battery and cansee dongle the soc and range is now sensible but the dc power sticks at 1kw
[image]

yoh-there, 2019-02-20:
Thank you. I assume the car setting is on Q90/R90/R110 right? If so it's probably a bug in the new battery modelling code.

yoh-there, 2019-02-20:
I had a quick look and it seems there is indeed a bug but it should only show if the car has a 22 kWh battery. Can you confirm you had this setting, so car type Q210 or R240?

fgmccrea, 2019-02-20:
Reasonably certain I had it on r90, possibly confirmed by miles range. Sorry car is away today so I can't try again right now.
ThanksGordon

yoh-there, 2019-02-20:
Same here (no car). Will try later too. I agree re. range, in which case there is another bug to chase. Hmmmm. BTW, if you have your phone, you can simply check in the settings; no car needed.

yoh-there, 2019-02-20:
F31. Print one of the two PDF's in this page and put it in your car ;-) 

https://canze.fisch.lu/the-fuses-of-the-zoe/

fgmccrea, 2019-02-20:
Phone is set on r90/110 
On another topic, do you think it is possible to switch off the day running lights when in drive mode?

yoh-there, 2019-02-20:
Thanks. Need to dig deeper then.
Running lights: No idea. You could see if there is something in DDT4All, either the EVC or the BCM/UCH (body control module)

yoh-there, 2019-02-20:
Yeah, mixed up coefficients pretty badly when implementing ZE40 and generalizing the battery model. I am pretty sure I fixed it. Will commit tonight after test and it will make it into the next release.

fgmccrea, 2019-02-20:
Can you send me a couple of .java files and I can try it later please

yoh-there, 2019-02-20:
Sure, but I will commit pretty soon (read: when SWMBO is back), and you can pull the all the commits in the current development branch. It would be easier and a fairer test. And you get the goodies earlier ;-)

fgmccrea, 2019-02-20:
Gr8 thanks

yoh-there, 2019-02-20:
Fix in development branch is pushed. Let me know please if your numbers look at least somewhat reasonable

fgmccrea, 2019-02-20:
Sorry I'm new to this, Where is the development branch?

fgmccrea, 2019-02-20:
Cancel that I have found it.....

fgmccrea, 2019-02-20:
And it seems much better now thank you.
Only tested on home charger so the charge power is populated with 5's but it sorts itself out nicely and the range fits well. 
I'll get it checked on a public 3 phase 22kw charge before the weekend.

yoh-there, 2019-02-20:
Thank you for your effort and the bug report. You probably noticed some graphical cleanup too, hehe. While I will refine the battery model a bit further (SOH), I will close this issue now, but if things turn up after all, feel free to reopen.

Closing with commit #4f59b88

closed 2019-02-20 by yoh-there