fesch/CanZE #542: A few more cleanup thoughts

yoh-there, 2019-03-11:
Please do comment. All very much "IMHO" of course.

1) Remove Experimental > BCB and move it's content to a new section under Technical > Charging (yes it will become a long page, but it is more logical)

2) Remove "All Data" from Technical DTC (makes no sense at all) and move it to a new activity Technical > Data. In that section also integrate what is now under "ELM DUMP" (makes no sense either, it has nothing to do anymore with the ELM), instead simply add an another "ECU" "All Free Frames". 

3) Remove Tests Bob (does nothing now), LeafSpy (obsolete and confusing) and Fluence Kangoo Temps (in planning anyway for aprx July)

4) Fixup Technical > Range to indicate a bit more what is shown, maybe move to main screen.

yoh-there, 2019-03-14:
On the technical side:
- convert to sdk-29 and androidx instead of compat, at the same time matching compileSdk and tarketSdK.
- maybe have a look at kotlin / ktx (not sure about this one).

Calling @fesch for input on this and the main post above, which is more "UI" :-)

fesch, 2019-03-14:
- remove "Tests Bob" >> OK
- remove "LeafSpy" >> **NOK**, I need this. Rename to "Test Bob" ;-)

About the API level: I have older devices and don't to kick myself of. Targeting a higher level will make us loose some users. I don't think this is clever ...

yoh-there, 2019-03-14:
A-OK on the UI thingies

I agree on the API, that is why I am NOT suggesting to step up the minSdkVersion at all. BTW I did some counts on version usage, see the blog. https://canze.fisch.lu/statistics/ There are 60 devices on Android 4. Our current minSdk of 15 means Android 4.0.3.

yoh-there, 2019-03-15:
In a (still private) branch I set the Compile and Target Sdk to 28 and migrated to AndroidX.

Seems to run fine on my Samsung S3 (Android 4.4, API 19) and Cubot X18 (Android 7.0 API 24).

I will try to run them both tonight in my car.

yoh-there, 2019-03-16:
1) How about calling the old "Leaf Spy"=> "Dash mode"? :-)

2) I had no problem whatsoever with SdkCompile and SdkTarget at 26 and MinSdk (unchanged) at 15. Studio did a decent job of downloading the androidx libraries and that didn't give a single issue compiletime or runtime. I propose to merge this change into development.

3) I also did a massive cleanup of unused resources.

4) Can you explain "Range" to me? It gives the weirdest numbers (like > 800) in the "by CanZE" part  on my end. Experimental? ;-)

yoh-there, 2019-03-16:
Closed: 
1. Remove Experimental > BCB and move it's content to a new section under Technical > Charging (yes it will become a long page, but it is more logical): done #eb78835
2. Remove Tests Bob (does nothing now): done #90c8ad8
3. Range Activity Questions / bugs: done/fixed #a4124dd

Open:
1. rename Leafspy to Dash mode and move to main screen: to decide
2. Kotlin: Meh, maybe, maybe not.
3: Move DTC all data / integrate with ELM DUMP: to do.

yoh-there, 2019-03-20:
Update and closing:
- Leafspy: rename yes, relocation no
- Kotlin: leaving it on the backburner.

All else done.

closed 2019-03-20 by yoh-there