fesch/CanZE #548: Allow CanZE to be stored on SD memory

yoh-there, 2019-04-05:
Hallo,

ich habe APP CanZE von euch aufs Handy geladen um von der ZOE  ein paar mehr Infos zu bekommen.

Leider lässt sich die App nicht auf die SD Karte des Handys verschieben. Samsung meckert jetzt immer, dass der Gerätespeicher voll ist und Updates nicht gemacht werden können.

Bitte gebt doch eurer App diese Funktion, damit sie auf der SD Karte abgelegt werden kann und den Gerätespeicher nicht unnötig belastet.

yoh-there, 2019-04-05:
From Android docs, seems simply a manifest setting

> Beginning with API Level 8, you can allow your application to be installed on the external storage (for example, the device's SD card). This is an optional feature you can declare for your application with the android:installLocationmanifest attribute. If you do not declare this attribute, your application will be installed on the internal storage only and it cannot be moved to the external storage.

fesch, 2019-04-05:
Thanks to the link. 

This will be published with the next release!

closed 2019-04-05 by fesch