fesch/CanZE #552: on abort not fixed

yoh-there, 2019-04-17:
Fora suggested this problem fixed by avoiding hardware acceleration on 8.1. Seems like this is either ineffective or I did not code it properly.

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
pid: 0, tid: 0 >>> lu.fisch.canze <<<

backtrace:
  #00  pc 000000000001e668  /system/lib64/libc.so (abort+120)
  #01  pc 0000000000008348  /system/lib64/liblog.so (__android_log_assert+296)
  #02  pc 00000000000700f8  /system/lib64/libhwui.so (android::uirenderer::renderthread::EglManager::createSurface(ANativeWindow*, bool)+332)
  #03  pc 000000000006de6c  /system/lib64/libhwui.so (android::uirenderer::renderthread::OpenGLPipeline::setSurface(android::Surface*, android::uirenderer::renderthread::SwapBehavior, android::uirenderer::renderthread::ColorMode)+76)
  #04  pc 000000000006a73c  /system/lib64/libhwui.so (android::uirenderer::renderthread::CanvasContext::setSurface(android::Surface*)+144)
  #05  pc 0000000000073828  /system/lib64/libhwui.so (android::uirenderer::renderthread::Bridge_initialize(android::uirenderer::renderthread::initializeArgs*)+16)
  #06  pc 000000000007547c  /system/lib64/libhwui.so (android::uirenderer::renderthread::MethodInvokeRenderTask::run()+24)
  #07  pc 0000000000076774  /system/lib64/libhwui.so (android::uirenderer::renderthread::RenderThread::threadLoop()+336)
  #08  pc 00000000000114e8  /system/lib64/libutils.so (android::Thread::_threadLoop(void*)+344)
  #09  pc 00000000000abcf4  /system/lib64/libandroid_runtime.so (android::AndroidRuntime::javaThreadShell(void*)+136)
  #10  pc 000000000007472c  /system/lib64/libc.so (__pthread_start(void*)+36)
  #11  pc 000000000001fbb4  /system/lib64/libc.so (__start_thread+68)

yoh-there, 2019-05-26:
This issue is by far the worst offender, >60% of all crash instances, pushing our ratings down. Any help on the cause - solutions is appreciated

yoh-there, 2019-05-28:
Switched of hardware accelleration for all screens having widgets or editable fields in the manifest and moved switching it back on programmatically to CanzeActivity. Need to wait for next release to see if this fixes the issue.

fesch, 2019-05-28:
Should I roll-up the next release ?

yoh-there, 2019-05-28:
Let me update the changelog (my achilles heel) ;-)

yoh-there, 2019-05-28:
Done. I think it's a good idea to release. Seen this crash 10 times now in the last release only. It's a nasty problem, hidden deep inside Android 8 code, so I am in favor. While not sure I hope it fixes it. Some other more minor issues committed too.

#f7c42565

yoh-there, 2019-07-08:
I consider this problem unfixable. It's an issue in Android 8.1. It seems to have subsided somewhat. I am going to close it, more or less in desparation.

closed 2019-07-08 by yoh-there