fesch/CanZE #553: CanzeDataSource instance null

yoh-there, 2019-04-18:
I have to give up on this one, sorry....

Found in the crash log. getInstance is called in MainActivity.onCreate with getBaseContext(), so I don't see a mechanism where the property instance can be null causing this Throw.

java.lang.Error: 
  at lu.fisch.canze.database.CanzeDataSource.getInstance (CanzeDataSource.java:49)
  at lu.fisch.canze.activities.SettingsActivity$7.onClick (SettingsActivity.java:466)

If we can't pinpoint this, I suggest a mechanism where it doesn't crash, but then again, the context of the error is Android 6 on S10, which looks weird to me. So, we should consider a "dontfix"

fesch, 2019-04-18:
I suppose this is due to the following, which is line #49:

`if (instance == null) throw new Error("Must call at least once with given context!");`

So there must be some place where the getInstance Method is being called before the context has been set. If we change this and return "null" instead, the above situation should produce a NullPointerException and we could find out where exactly the call is being made ...

yoh-there, 2019-04-18:
ok, will do

closed 2019-04-18 by yoh-there

yoh-there, 2019-04-20:
BTW, as I suspect this happening because of the threaded hourly cleanup, I have added an extra check there. Commit #203f9016