fesch/CanZE #554: Tire settings cannot be written anymore v.1.37

Ulf-D, 2019-04-28:
Hi, since the latest update (CanZE v. 1.37, build:2019.04.15), I cannot write the Tire sensor-ID settings (TPMS) anymore.  Only can read it.  I get the error  "atm..673" all the time. With a former version from January this year (1.29) it works.
Using a Konnwei ELM327 at a ZOE R90 car.
Thanks for your help!

yoh-there, 2019-04-28:
Thanks, I will delve into it (though it might be hard, I don't have a TPMS equiped car, thought I had it nailed)

yoh-there, 2019-04-28:
Trying to find this. Can you please insert a screenshot of the actual error? I have a hint of what it might be, but I don't want to just release "guesses" too much. Thank you!

Ulf-D, 2019-04-28:
Hi,

thanks a lot for the very quick response!

Send you the screenshot later, if available.

 

BR

Ulf

 

Gesendet: Sonntag, 28. April 2019 um 15:15 Uhr

Ulf-D, 2019-04-29:
Hi, 

 

finally I had the chance for the screenshoots- please see attached.

Hope that helps.

If you need further info, let me know please.

 

Looking forward to hear from you.

 

Best regards

Ulf

 

Gesendet: Sonntag, 28. April 2019 um 15:15 Uhr

yoh-there, 2019-04-29:
Sorry ulf, files don't work through email I think. Can you please try to add them here? https://github.com/fesch/CanZE/issues/554 Otherwise maybe drop a public link where we can view. Thanks.

Blaubaer306, 2019-05-02:
Guys, good day!
I yesterday used the v.1.37 with an ELM327 on a Zoe-Q90 to finally register the Valves of my winter tires.
In the beginning I could only read, then my ELM327 crashed and after a new start of phone and ELM I was able to read and overwrite the figures. It now works perfectly and CanZE shows the different pressures. Zoe's continuous warnings are gone, too. So maybe it's not the CanZE version.

But finally I wanted to give a big "Thank you" to all the guys contributing in developing this amazing tool CanZE!

yoh-there, 2019-05-02:
Thank you @Blaubaer306 for letting us know. Maybe, for the sake of at least @Ulf-D can you state what dongle you are using? I'll leave the issue open for a bit more.

Ulf-D, 2019-05-03:
Hi,

I´m using the Konnwei KW 902 bluetooth OBD-adapter. I just tried it again. I can read the valve-ID´s without problems, but never write. Everytime the message "Could not write TPMS-values" and sometimes the error. Please follow-up, as this function is important for all users with the valve sensors.

Thanks a lot!

Best regards

Ulf

 

Gesendet: Donnerstag, 02. Mai 2019 um 23:52 Uhr

Ulf-D, 2019-05-10:
Hi, did you have a chance to get in there?

Thanks and best regards 

Ulf

fgmccrea, 2019-06-04:
When trying to read the IDs, do you get the error message 'These are not all valid hex values other than 000000'

Ulf-D, 2019-06-05:
No, while reading no error messages occure. This is possible without any issues. 
 

Gesendet: Dienstag, 04. Juni 2019 um 08:11 Uhr

Aquaerics, 2019-06-08:
I have the same problem. I cannont write the id values. Reading works under android without problems.
I use a ODB2 Dongle wich is labeled with XiTECH XR7.
The error Message is Error {atcra673] [?]

yoh-there, 2019-06-13:
This issue has not slipped our attention.

fgmccrea, 2019-06-22:
So sorry I have only just noticed that the 'read' button is now marked 'write', so I CAN read the ids if I click the 'write' button.

yoh-there, 2019-06-22:
This is incredibly daft and I have no idea how I could have that slip to production. The button texts are indeed swapped. Since I have no TPMS myself, could you please check if the read button successfully writes?

fgmccrea, 2019-06-22:
I hope people realise that the wheels need to move to switch on the pressure sensors. I have never tried changing an id so I just tried editing the top one. I see it displays hex as lower case letters but the keypad insists on upper case for that data. So I tried numbers only but failed there as well.

yoh-there, 2019-06-22:
Ah thank you. Yes, moving, but that is just for the pressure to register, not for reading or writing the IDs.

Lower or uppercase should not matter is it is converted to binary internally, but I agree it is sloppy and I will change it.

Please elaborate a bit on "failed": did you get an error, and if so, which one? Did it appear to work but when read back the original data re-appeared?

fgmccrea, 2019-06-22:
Does 'Debug 673.0 ... fail'  help

fgmccrea, 2019-06-22:
Or in grey 'could not write TPMS valves'

yoh-there, 2019-06-22:
It, does, thanks again. I will have a look tonight at the code.

fgmccrea, 2019-06-22:
Like '50c0' ?

yoh-there, 2019-06-22:
Yup, that's the stuff I am looking for. It's a bit hard as I don't own a TPMS model car! Any and every error displayed when you hit "write" (with reasonable data of course) is appreciated. "Too much" does not exists :-)

fgmccrea, 2019-06-22:
Also got '7e'

Ulf-D, 2019-06-27:
Dear all,

Sorry for the delay, was out. I tried again several times and would summarise as follows:

1) Read and write buttons are interchanged.

2) No issue to read the IDs 

3) While writing got the following errors: atcra5d7, atcra654, atcra1fd, atcra1f8 and atsh7e4.

4) Sometimes it's possible to write. 

Don't know, if this has to do with the timeouts I get often at all screens and why is that.

Hope that helps. 

Best regards 

Ulf

yoh-there, 2019-06-27:
Very helpful, thank you.
1) is fixed in code and will be included in next release
2) Good to know!!
3) and 4) helpful, will dig into that. I must say though that cra errors hint to dongle problems, and should (!!) not be related to TMPD IDs specifically really.

I will try to do a few tests soon, but as said, need to have access to a car.

yoh-there, 2019-07-04:
Said release was set free a few days ago. I would be interested in updates, as I suspect at least part of the "it isn't working" was because of the silly button swapping.

fgmccrea, 2019-07-04:
E-books.rtm.empty

fgmccrea, 2019-07-04:
Spell check sorry E-BobDue.rtm.empty is now a consistent response to an attempt to write

yoh-there, 2019-07-04:
Oh I didn't realize you used a CanSee dongle. That changes the dynamics slightly. Thanks.

fgmccrea, 2019-07-04:
Sorry I should have mentioned that.

yoh-there, 2019-07-05:
No worries. Just a thought. I have been pretty careless with decent version management on the CanSee firmware and most certainly the master branch has been not OK for a while. In January I fixed a few nasty bugs in ISO-TP long frames handling that made it to master end of February. Can I assume you took the firmware after that? Just making sure as otherwise I am going completely up the wrong road with this.

fgmccrea, 2019-07-05:
Aha. I must try that. It's possible that I have the older version. Thanks!

fgmccrea, 2019-07-06:
You are correct my cansee firmware was the problem. I can now read and write perfectly.

yoh-there, 2019-07-06:
Thank you VERY much reporting back. Was a bit of a head scratcher!!

@Ulf-D @Aquaerics @Blaubaer306 How are things looking for you please?

Ulf-D, 2019-07-08:
Hi, I'm on holidays right now, will have a look after my return ASAP. 

Regards 

Ulf

E-Gerd-21, 2019-08-08:
There is a defect in the German translation string for the two buttons on the "Tires info" page. At least for the latest version 1.42.

Button lower left "READ" should be translated with "ID LESEN"
Button lower right "WRITE" should be translated with "ID SCHREIBEN"
Obviously "ID LESEN" and "ID SCHREIBEN" are interchanged.
The wrong translation leads to wrong user action and malfunction.
See https://www.goingelectric.de/forum/viewtopic.php?f=64&t=14853&p=1061088#p1061088

Fix should be easy and straight forward.

Furthermore usability can be improved, if SPACES in the four German translation strings for "ID vorne rechts     " etc.  will be removed to avoid unnecessary CR. Goal should be to have all information (4 x pressure and 4 x Sensor ID) on one page. Same as for English page.  See the screenshots in the goingelectric link.
Regards Gerd

E-Gerd-21, 2019-08-09:
Hello,
defect is in the file CanZE/app/src/main/res/values-de/strings.xml in lines 659 and 660. 
German text string is interchanged.
Gerd

yoh-there, 2019-08-11:
Fixed #a68434a9 Thanks Gerd. Will have a peek on the scroll of the IDs later.

yoh-there, 2019-08-11:
Fixed #de8876ad (layout) Thanks Gerd

yoh-there, 2019-08-11:
I think the TPMS issues have now been fixed and the button swap had caused a lot of confusions. Therefor I will close this issue. If another issue pops up in the next release (this will be upcoming 1.43, when the last two fixes will be incorporated), please do open a NEW issue, as this one has become a bit messy. Thank you.

closed 2019-08-11 by yoh-there