fesch/CanZE #566: Renault Kangoo Z.E. Support

DerArtem, 2019-07-13:
Hello,

what is necessary to add support for Kangoo Z.E. to CanZE?

The Kangoo Z.E. is not in the list of the car type, but when disabling the "Save driving mode" in the app the most things do work with setting to Zoe R90/110.

My main issue is that reading the fault memory does not work. I need to troubleshoot the electic fault error message in my Kangoo.

How can I assist you to add this feature?

fesch, 2019-07-13:
Needed would be a Kangoo owner with development competencies. Next step would be to determine the different PID's that circulate on the CAN bus and get detailed information about ISO-TP frames that the Kangoo uses.

Some of the Zoé Settings work for the Kangoo, yes. This is because some PID's seam to be Renault-internally "standard", but as the car's are not the same, things differ in details ;)

DerArtem, 2019-07-13:
Ok, great, I am also a developer and have a Kangoo :) but have not much experience in android development. Can you give me a quick start how to compile CanZE and where I should look into?

fesch, 2019-07-13:
Perfect ;-)

So, first do and download Android Studio with Android SDK. Next you need to clone the present repository, branch "Development". Once cloned, you can simply open it in Android Studio an hit the "compile" button. That should be all.

After having done changes (and having tested them) you can then push your changes into a merge request in order to integrate them back into the repository.

landswellsong, 2019-08-13:
Copying my comment from another issue, my friend runs an EV repair garage and has complained ever since Kangoo/Fluence support was dropped that he is forced to use outdated versions for them. He routinely has access to both (or at least their ECUs ;) ) so he can do testing and some QA if that was to return eventually. Subscribing with my comment for updates.

[image]

DerArtem, 2019-08-13:
I have issues with the Can Bus on my Kangoo and need to fix it first.

yoh-there, 2019-08-13:
I don't think there has ever been any specific Fluence/Kangoo code in the DTC activity. Under the hood we simply only have managed to roll in the definitions of ZOE ECU's  but as fesch said, there are many commands that seemed to work "well enough" on a Fluence/Kangoo. So my first advice to your EV mate is to simply choose ZOE and try.

solmoller, 2019-08-29:
I really, really miss my driving screen in the Fluence - sad it had to go

yoh-there, 2019-08-29:
If you literally mean the "driving" activity, absolutely nothing has changed. Put the car type on ZOE and you should be fine. :-)

FYI, the only affected activities are
- Settings and Main (tech stuff)
- TempHeatmap
- Clima
- Charging tech
- The specific Kangoo/FLuence temperatures screen.

All other screens are unchanged and if they don't work now, they should not have worked before.

Happy driving!

landswellsong, 2019-08-29:
@yoh-there got a specific feedback, the new chargning details screen has SIDs which are unavailable on Kangoos and so the whole operation is either locked or very much delayed while it retries to get those SIDs.

* I've not personally tried it because I don't sadly have a Kangoo, just relaying

yoh-there, 2019-08-29:
Yeah, that is probably the charging tech screen. Since a few releases it features BCB parameters showing the quality of the power line (so, before the charger). That change was made quite a while before stopping Fluence / Kangoo support and illustrates perfectly what this is about: we have no way to test and thus get bad results.

The simple charging screen probably works fine as it doe not display any BCB parameters.

Thank you for your feedback, appreciated.

landswellsong, 2019-08-29:
Yeah I believe it's about the advanced charging screen since it got that voltages, resistance etc thing recently.

solmoller, 2019-09-22:
Confirming, that the driving screen still works with Fluence ZE, thanks

yoh-there, 2019-10-19:
Closing, as it has gone stale. Feel free to re-open

closed 2019-10-19 by yoh-there

KarelSvo, 2020-01-05:
> My main issue is that reading the fault memory does not work. I need to troubleshoot the electic fault error message in my Kangoo.

[click here](https://www.drive2.ru/b/501493851572142178/) ;-)