fesch/CanZE #571: Black mode

solmoller, 2019-09-22:
We should consider doing a dark mode, the bright screen may be too bright in some scenarios, people are getting bigger screens in their cars.

yoh-there, 2019-09-22:
I agree and have started a small experiment. While Android does most of the hard work, it does require quite a bit of going through all the code and resources, getting rid of all hardcoded colors.

yoh-there, 2019-09-29:
Status update: all done in dev branch, with the following remarks
- I used ?attr/colorButtonNormal for all "patch" blocks, ie block in the charging screen. The contrast between the styles backgrounds was not enough;
- the widgets have not been done yet. (help @fesch requested);
- where alarms (ie tyres) or dynamic (heatmaps) backgrounds are needed, I start of with ?attr/colorButtonNormal and modify it with reasonable assumptions;
- lots of cleaning up. Not all may be liked (ie, the previously yellow boxes and header bar). See for yourself;
- the theme is NOT settable but follows the system settings in Android 9 ("night") or 10 ("dark"). Debatable if a setting is needed IMHO.

yoh-there, 2019-10-01:
Status update: all widgets are done using
- custom:backgroundColor="?android:attr/colorBackground"
- custom:foregroundColor="?android:attr/colorForeground"
- custom:intermediateColor="?android:attr/colorForeground" (for tacho only)

Also updated label,plotter,tacho to listen to these settings (just like timeplot)

yoh-there, 2019-10-01:
I think done. Closing issues, will most probably go out next release. Closing, feel free to test in branch development, or to reopen when issues come up in production.

closed 2019-10-01 by yoh-there

solmoller, 2019-10-08:
Brilliant, I am quite impressd - kudos!

Henrik