fesch/CanZE #574: elm connection error

Rraets, 2019-10-14:
When I connect ELM327 ODB scanner I alway get following message :
ELM ready, version 2.x, will problably not work, please report it if does

yoh-there, 2019-10-14:
Great (as I assume it DOES work). What type of dongle is it? "elm327" is insanely generic and can mean anything. Link, or pic?

Rraets, 2019-10-15:
Hello,

I is this one:
https://www.bol.com/nl/p/obd2-scanner-bluetooth-elm327-obdii-auto-computer-uitlezen-car-reader-diagnose-foutmeldingen/9200000088401970/?s2a=

There is not a lot of information on it..

greetings

René

yoh-there, 2019-10-15:
Thanks. I assume that despite the warning, it does work normally?

Rraets, 2019-10-15:
No it doesnt work at all. After I give the pincode 1234 for the BT connection it gives that message. Then it looses the BT connection

Outlook voor Android downloaden<https://aka.ms/ghei36>

yoh-there, 2019-10-15:
Ah, my bad. Well, sorry to say the only dongles we pledge to do our stinking best to run are the KONNWEI902 and CanSee. See https://canze.fisch.lu/elm327/ and https://gitlab.com/jeroenmeijer/cansee

closed 2019-10-15 by yoh-there

Rraets, 2019-10-15:
Is this a good one ?
https://www.amazon.de/Diagnoseger%C3%A4t-KONNWEI-Autoscanner-Unterst%C3%BCtzung-OBD2-Protokolle/dp/B077XGSV83/ref=sr_1_8?__mk_nl_NL=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=MaxiScan+Konnwei+902&qid=1571157970&sr=8-8

fesch, 2019-10-15:
It might ...

Rraets, 2019-10-15:
Do you sell them of do you have a link to a good one ??