fesch/CanZE #576: Dark mode, black not grey?

KotekBesar, 2019-10-28:
Hi! Loving the new dark mode. One question, is it possible that, instead of dark grey, it can be black? The reason is, completely black would be better for us to have OLED screens. Thank you!

fesch, 2019-10-28:
The colors are defined by the system, not by us. So I think this will not be possible ...

yoh-there, 2019-10-28:
Just for my own curiousity I figured the colors that are used by the standard themes on the Pixel 3a API 29 emulator
```
Foreground colors: Light mode ff000000, dark mode ffffffff (aka hard black, white)
Background colors: Light mode fffafafa, dark mode ff303030 (aka almost white, dark grey)
```

So the settings as described by @KotekBesar are real.

Implementing two custom CanZE themes is no option now (too much work really), so it is only fair to say there won't come a solution from the developers.

A working solution might be to install one of the custom themes from the Play store but I am not sure if this will work at all, as Android "tweaks" dark/light for version 8 and older.

KotekBesar, 2019-10-28:
Understood. Thank you both for taking the time to look in to it. :)

closed 2019-11-03 by yoh-there