fesch/CanZE #577: Overwriting TPMS IDs via CanZE

minibiti, 2019-10-28:
Hi,
I have the latest version of CanZE (1.45) on Android but when I try to overwrite the tires pressure sensors registered with new IDs so they match the winter tires ones, I get an error.
"Could not read write valves: E-ISOTP tx flow error"
Any idea?

yoh-there, 2019-10-29:
Thanks for reporting. I am assuming you are using a KONNWEI902 dongle? If so, there is not a lot we can do in the short term I am afraid, as both developers own older (non TPMS) cars; we literally have to borrow another more recent ZOE to debug and test.

minibiti, 2019-10-29:
I have the ELM327 Bluetooth dongle, like this one: http://www.urlaub-in-istanbul.com/images/1/en32862623124.jpg

yoh-there, 2019-10-29:
Ah, that will almost certainly not work, sorry. See https://canze.fisch.lu/hardware/ and https://canze.fisch.lu/elm327/ for the explanation. Feel free to re-open when tried with a KONNWEI902 or a CanSee ( https://gitlab.com/jeroenmeijer/cansee ) dongle.

minibiti, 2019-10-29:
OK thanks for the info. Sorry :)

kler, 2019-10-31:
I actually have the Konwei, bought earlier this year. I get the same error.

Can I extract some information somehow that would help?

[image]

[image]

[image]

yoh-there, 2019-10-31:
Thank you for the offer. I need to go over the code, but it is hard as I do not have a car with TPMS. We need to consider adding some below-the-bonnet logging to get this right. If that works out, I might get back to you to see if you can do a few test.

E-Gerd-21, 2019-11-08:
I can confirm the same defect as kler did 8 days ago. Unable to Write TPMS ID mit Error Msg "Could not read write valves:-E-ISOTP tx flow Error" as response to "Debug 673.11".
I am using KONNWEI 902 (black case)  for a Zoe R90 from 2017. CanZE ver.1.45.

yoh-there, 2019-11-09:
Thanks for reporting. The cause it not very clear yet, but might be related to timing. I made a small change in the ELM driver which will come out in the next release. Expect it to be in the play store in under a week.

In the mean time, you MIGHT want to try with the car not "started" and see if that makes a difference. I would appreciate the results of both.

yoh-there, 2019-11-13:
https://github.com/fesch/CanZE/issues/581 confirmed also in the latest update, so the timing change did not work. I have limited resources now (time) but will work on it ASAP

kler, 2019-11-18:
> Thanks for reporting. The cause it not very clear yet, but might be related to timing. I made a small change in the ELM driver which will come out in the next release. Expect it to be in the play [...]

I've now tried with version 1.46 and the car not started, and I receive an error, slightly different than before:

    Could not write walves: E-ISOTP tx flow Error:STOPPED

FredLeudon, 2019-11-18:
Hi all,

It seems there are two ways to write TPMS IDS

1 : Writing all TPMS ID at same time

	Request[765.3B5D000000000000000000000000] : Write_TPMS_Configuration_4      (ignorée)
		> Send       [765.3B5D000000000000000000000000.3.0]*: CodeIdentite(1)
		> Send       [765.3B5D000000000000000000000000.6.0]*: CodeIdentite(2)
		> Send       [765.3B5D000000000000000000000000.9.0]*: CodeIdentite(3)
		> Send       [765.3B5D000000000000000000000000.12.0]*: CodeIdentite(4)
2 : Writing TPMS one wheel at a time

		Request[765.3B5E00000000] : Write_TPMS_Configuration_12      (ignorée)
		> Send       [765.3B5E00000000.3.0]*: Position_roue
			Values list : 
				1	AVG
				3	ARD
				4	ARG
				2	AVD
				0	Aucune
		> Send       [765.3B5E00000000.4.0]*: CodeIdentite

In CanZE for iOS, I use second method.

Bytes to send are : 3B5E00000000

Where 00 equals Wheel ID (01 Front Left, 02 Front Right, 03 Rear Right and 04 Rear Left)

And 000000 are TPMS ID (AB4C37) in the log below

Below, CanZE log for writing TPMS ID AB4C37 on Rear Right Wheel

>	:requestField(ecu[26],request[objet][Write_TPMS_Configuration_12]	Virtual[0]
		SID[765.3B5E00000000.3.0]	Data[Position_roue]	Value[ARD]	Raw Value[3]
		SID[765.3B5E00000000.4.0]	Data[CodeIdentite]	Value[11226167]	Raw Value[11226167]
		Construction Bytes To Send	Ecu bytes to send (original) = [3B5E00000000]	ECU bytes to send (calculé) = [3B5E03AB4C37]
	>	:sendELM(elm[objet], ecu_request[objet], command[3B5E03AB4C37], auto[1]
		       	CanBus[1]	>	Ecriture ELM : data[063B5E03AB4C37<\n>]
			CanBus[1]	<	Ecriture ELM : Ok

Hope it’s help ;-)

Tried on my non TPMS equipped 2013’s Zoé and no error occurs; but no data where written...

Cordialement,

Frédéric RICHARD

yoh-there, 2019-11-18:
That is GREAT info as I can (temporarily) avoid the long frame issue on the ELMs. Thank you.

FredLeudon, 2019-11-18:
You’r welcome

Cordialement,

Frédéric RICHARD

yoh-there, 2019-11-18:
I have implemented @FredLeudon  suggestion, and @fesch released. I really hope this is fixed now. Thank you all for reporting. I am closing the issue, but please reopen if this release did not solve the issue.

kler, 2019-11-18:
Thx for the rapid work!

However, I believe that the release caused a problem, namely that the TPMS are _wiped from the car_ if a **write** is performed :(

I had working TPMS values programmed in the car, and after writing I get the error:

    Could not write valves:-E-message.isotp.startswith7f

And

    Timeout on [Atma] []

And after that

    No TPMS valves found

[image]

yoh-there, 2019-11-18:
Oh bloody hell :-(

kler, 2019-11-18:
Well, I didn't try it, but I _believe_ the car stills drives  xD

yoh-there, 2019-11-18:
I have no TPMS, but I will try to fix ASAP. I hope you have a screen print or paper with the proper (current) values?

kler, 2019-11-18:
I do have screenshot from when they were attached last winter.
However, I was stupid enough to _not_ screenshot before this operation I did today though.
The reason it bugs me is that realised later on I'm _pretty sure_ that the values I saw today before writing _are different_ from the ones I have taken note of last winter...

The reason I believe this is the case is because I today changed the last letter of front left from **A** to **B** before pressing write.
But in my notes the last letter of front left is **7**. And even if the wheels would have been mounted differently I have no TPMS from notes taken last winter that ends with **A**.

I really can't explain this.
I know for a fact that I'm using the same tires, and that they most likely are mounted in the same place _(I had tags on them)_. There is of course the possibility that I'm confused and am remembering the change I did today incorrectly. But if not I was thinking if there could be a possibility that CanZE would decode or display the value differently between now and around March last year?

Anyhow - I'll happily test the next fix, and hopefully the values I took note of last winter will work :)

yoh-there, 2019-11-18:
OK, thanks for being a good sport. I went over the code and it looks quite OK, so this is a tough one. And I did test things, though a while ago and with a CanSee dongle. I'll dig deeper.

bjaolsen, 2019-11-20:
Not that it helps CanZE, but if you had issues with the TPMS-IDs you can write them back with ddt4all and a suitable database for renault. I have used this the last 3 seasons without issue even with a cheapo ELM327 from Ali.

yoh-there, 2019-11-20:
That's true indeed and it does work. Would also be a great way to ensure
the ID's you have listed are correct.

kler, 2019-11-21:
Thx for input!
I wasn't aware of this tool, and after reading up on it, it felt a little bit too much hardcore for me right now - especially since it requires a separate DDT2000 database that has to be downloaded illegally somewhere because it's copyrighted.

Irregardless of how stupid it is to actually copyright such a database, I couldn't find any trustworthy source for downloading it, and I don't want to do _too much_ fiddling with neither the computer nor the car :)
My hope is still that CanZE will manage this.

softamateur, 2019-11-24:
Regarding Write tire ID's:
I was able to write the winter tires ID numbers in to the ODB port of the ZOE.  I did this with version 1.47, not fully aware of the bug warning issued for this version. The TPMS error was shining in the ZOE. Afterwards the warning was magically unlit and the sensors started to report after half a minutes driving in approximate 50 km/h.   I did not have to do the "Learning tyre pressure" procedure.

My ODB dong is a Vgate Icar Bluetooth 3. I noted that the CanZE needed a minute of operation before it starts to read data. The setting  "use ISOPT fields" was on. To be sure CanZE had communication, I checked "Battery voltage map" was working. Write button seems not perfectly stable and different warnings and errors was output in the debug field. But when the Read button returns what was written with the Write button, it seems to work. I pushed the Write button a couple of times and the the Read button.

More details regarding ID's:
I bought winter tires and rims myself from the net (ZOE R110 41 kW -2019) and realized afterwards the need of TPMS sensors. So I bought them as well, original VDO/Continental. Took photos of the printed 8 digit Hex ID numbers on the sensors. Only the six to the right is used as ID. With a service tool, Autel  MaxiTPMS TS401, I could se that the printed ID´s on the sensors had several singel character errors. The printed markings was then useless! Always check with a service tool. The local Tyre shop put on the sensors and swapped the tires for the winter.   

Recommendations for winter tires:
Buy winter tires with programmable sensors from your local tire shop, and let them clone your summer tires ID and none of the above have to be done ;-) Just swap tires on your own will hereafter. Second recommendation, buy a TS401 (Ebay 120-130 Euros) and be in control. It can both read all sensor data in a comprehensive way and program Autel MX sensors. But no ODB functionality. But cloned sensors don't need ODB.   

Thanks a lot for CanZE.

yoh-there, 2019-11-24:
@softamateur  Thank you so much or your report. It really helps.

I put the warning in because of the issue reported by @kler above. Maybe that was a bit premature. Anyway, the next release should be more stable and also added quite a bit of checks.

What a mess with the prints on the packaging, who would expect that??

@kler Above suggests "try again", using latest version and "Use ISOTP fields" on, though the latter should not have any influence on reading and writing ID's (it does on pressure readings)? Of course you can wait until the next release, which is due quite soon.

yoh-there, 2019-11-24:
Release is in the play store now. I would appreciate a retry as soon as it's in. If it works I can then change the news bar and remove the warning. Thanks!

minibiti, 2019-11-24:
I just tried the new CanZE 1.48 and I get this error now, both with and without ISOTP fields:
Could not write TPMS valves:-E-ISOTP rx unexpected 1st nibble of 1st frame: NO DATA
On the plus side, it did update the first TPMS ID but not the last 3.
So progress but not fully working yet.

yoh-there, 2019-11-24:
Thank you for testing @minibiti . You can skip valves by entering value 0. So, if you can write all valves by doing them one by one, it is a timing issue that I think we can fairly easily fix.

yoh-there, 2019-11-24:
I have create a public beta that should be in the play store within hours. I will keep an eye on that.The only change is in the tires part. It will write, with a delay, the ID's one by one and will NOT stop if there is an error in one of the writes. You can enter 0 to skip a write.

minibiti, 2019-11-24:
I tried one valve at a time and it works great! :)
One little bug: looks like the rear left and rear right are inverted. You need to write to the rear left to update the rear right, and vice versa.

yoh-there, 2019-11-24:
tataa, getting there. Are you sure about the positions? I saw the switched 3rd and 4th position and therefor double-checked the documentation multiple times?

softamateur, 2019-11-24:
I can confirm, reading the sensors direct, with the TMPS service tool TS401, that the writing is correct and indeed the Reading of RR and RL are switched.  I can se that in the pressure difference between the tires compared to CanZE (In the 1.47 version)

kler, 2019-11-26:
I've tried now again with the Konwei dongle, and write still fails for me.
I get 2 error messages:

    Your car has no TPMS system

    Could not write TPMS valves:-E-ISOTP rx unexpected 1st nibble of 1st frame: NO DATA

The first value however seems to be correct - but I don't know if CanZE actually managed to write that value, or if that particular value just wasn't overwritten when I initially tried this.

Also the tire input form is locked for long amounts of time, so I need to struggle to actually write the IDs into it. When switching app _(to get to the app where I took note of the IDs)_ the connection breaks and the form goes into RO mode. Often if goes back to write mode in a few seconds, but sometimes it seems to stuck in RO mode. So the best option is to have the notes beside the phone, which is a shame because copy paste from another app is more convenient :)
This however is not related to the bug, and just a side note.

yoh-there, 2019-11-26:
Hi Patrick, thanks for trying. NO DATA suggests a problem between the car and the dongle or the dongle being overwhelmed. Hmmm. I take it this was done with 1.49beta2 right?

I would appreciate if the other participants in this thread can give that a spin too.

I need to borrow a car with TPMS.......

kler, 2019-11-26:
I will order a Vgate **Icar Bluetooth 3**, since that seems to have worked for @softamateur

And I was not the beta - sry :/
I was using **v.1.48**
How do I install the beta? 🤔

yoh-there, 2019-11-26:
Some extra info: in the Tires screen, the app queries the car every 6 seconds for the pressures, wheel state and if there is a TPMS installed. If there is no TPMS installed, it will not enable the read, write and 4 fields. So, if this goes "back and forth", that query gives strange results. But what I fail to understand is, if there is a faulty response, values (including the state) should not be updated at all.

yoh-there, 2019-11-26:
If you go to the play store >> My apps >> installed and open the CanZE screen, there is an option to enroll in Beta testing. It  will take a while but after a while you will get an update. You can always leave the beta. See also https://canze.fisch.lu/open-beta-for-android/

yoh-there, 2019-11-26:
Hopefully fixed issue when switching to other app. 1.49.beta3 in play store

kler, 2019-11-26:
**Success _(more or less)_!**

Thx for the instruction, and the hard work!
Sad that the community need to bridge this crappy design from Renault, but what an adventure ;)

With 1.49.beta2 it actually worked _(I'm on beta2, that's what I got after signing up for beta)_.

There are some strange behaviours in the form still - but it's better than in 1.48.
What's happening is the same, it disables the input, and then enables it again. Also at least once it got stuck so that I had to restart the app.
The period of time that the form is disabled is much shorter than in 1.48, so it's really not a problem. But when switching between my notes and CanZE, I just need to wait (unless it actually get stuck).

However, from my point of view these are minors, and CanZE managed the task from what I can tell.

Now I just need to drive like 10 km or so to verify that it **really** works - that I managed to get the right ID in the right place. But from a CanZE perspective it seems like WIN.

Again, thank you! 🎆

yoh-there, 2019-11-26:
Ah great! You will get beta3 pretty soon (usually in less than an hour) and it should have shortened the waiting time significantly and not repeat the disabled fields when switching apps.

I can't determine if this is "real release ready" and would appreciate input after checking beta3.

The one time that I drove in a ZOE with TPMS the pressures were shown in less than a minute. I think as soon as the valves detect motion they are triggered to transmit state

kler, 2019-11-26:
I know that when the shop failed to write TPMS (they failed 3 times on 2 different Zoes) it took around 8-10 km for the car to come to the conclusion that it cannot read the TPMS.

- If not all 4 TPMS values are programmed, the car will show an error when starting
- If all 4 are programmed, but it for some reason can't read it, the error isn't displayed until 8-10 km _(or at least that's my experience)_

I will try beta3 once I get it, I will change to an incorrect value and the back again.

yoh-there, 2019-11-26:
Got it. Slightly different scenario I guess. What I meant was the pressures showing after a minutes, as in, valves transmitted, car received, CanZE displayed.

From what I understand from your comment is the car throws a fit after 10 minutes if **not** all 4 are received. That makes sense. Combining the above, I would bet that if you see 4 pressures in CanZE, by definition no error will pop up on the dashboard.

But hey, you never know ;-)

kler, 2019-11-26:
So I have read values in the app, so from my perspective this works.

Pressure low, which is also confirmed by the car.

[image]

yoh-there, 2019-11-26:
And you could program all 4 ID's without having to do them one-by-one? If so I think it's releasable.

ps: Google is still not rolling out beta3 it seems. Changes were small.

kler, 2019-11-26:
How would I program then one by one?

I just filled out all values and pressed write, however one or two of them happened to be right from the beginning (I suspect that they either weren't completely wiped, or CanZE managed to write some of the IDs when I tried using 1.48).

yoh-there, 2019-11-26:
If you enter zero, that valve is skipped while writing. I think I wrote that somewhere earlier but it doesn't matter. So I think you are saying is now works as it should. If not, please let me know.

@minibiti - @E-Gerd-21  - @softamateur maybe want to chime in on the beta3? It would be nice if we can finally close this chapter!

One word about Open Beta. As soon as I release a new production version, that will overwrite your beta, so no worries about being left behind. The only thing is if we decide to launch something else to test in the Beta track and if you are still enrolled, you will get that.

minibiti, 2019-11-26:
I joined the beta testing :) I will retest as soon as I can, hopefully tomorrow.
But now that it is all fine for my car, what should I try?
Write bad ID values and then write the good ones again? All the 4 of them at the same time?

yoh-there, 2019-11-26:
Yes please!

softamateur, 2019-11-26:
Hi there!

Beta test repport:
1.  Perfectly working winter tires TPMS with correct ID written according to Canze. 
2.   Did read.  Ok.
3.   Did write (still with the ok ID:s)
4.    Pressure displays was wiped out. 
5.    Drove in about 30- 60 seconds at 50 km/h
6.    All sensor started to display again. 
7.    Entered wrong ID.  
8.    Write.   ok 
9.    Read.    ok.
10.  Pressure display was wiped out ( as in point 4)
11.  Didn´t bother to drive 
12.  Entered correct ID's again. 
13.  Read ok  
14.  After 30 seconds of driving all pressure was displayed 
(as in point 6.)

So far so good :-)

But there is a message coming up when phone is rotated or
if going out of "Tires" and the on re-enter to "Tires"  
It says, a little worrying, "Your car has no TPMS system".

I will report back if any problems occurs,  if the warning lamp on the Zoe dashboard lights up. I will also check with CanZE to se if it still display the message above. 

When writing there was some text flickering in the debug field. 
But the writing seamed to proceed well. 

For the protocol:  Vgate Icar Bluetooth 3,  ZOE R110 41 kW 2019
CanZE: 1.49beta3  2019-11-26 @10:16

Happy driving everyone !

yoh-there, 2019-11-26:
Thank you! Rotation issue solved in source code.

minibiti, 2019-11-27:
So I re-tested this morning with 1.49beta3.
I tried to add 4 random values and then write the correct ones back. I did that 2/3 times.
At least once or twice, the 4th values does not get written and ends up getting 000000 when I read again.
I tried with and without ISOTP fields.
I have the ELM327 Bluetooth dongle (Mini) (https://www.ebay.com.au/itm/113815327984)

yoh-there, 2019-11-27:
Thank you for testing again! I have to say, that dongle, or I must actually say, most of those dongles, are awfully crappy and don't perform well at all. It's hard to say, because these blue shells are very generic and house many, many different designs. The other day I [got one with a real PIC controller](https://canze.fisch.lu/el-cheapo-dongle/) in it that was fine, but I had another with two stacked PCB's that was pure evil.

The code is 100% the same for each wheel (it's actually a loop), so there is not a lot I can do, other than write-read-compare maybe, retry once or twice if it fails. Hmmmm, let me think about that for a little bit.

softamateur, 2019-11-27:
Hi there,

It works alright after 25 minutes of driving.  Nothing to note at all. Perfect. 

But some points just to remind and perhaps add/change: 

1. "Your cars has no TPMS"  can be annoying for cars with TPMS  :-)

2. "Please don't write Valves etc"  Don't forget to remove this text !

3. Strictly, the correct denomination is "Sensors" not "Valves".  There are TPMS sensors, from
Continental, that are not in contact with the valves at all, they are glued on to the inside rubber 
of the tire.  You read and write to the sensor, not the valve.  (Just important for the nerds ;-)

4. A note to the user about waiting with further actions until  CanZE is communication correctly? Checking the battery map for instance?    (This is perhaps overkill?) 

5. A note to the user not to have "Learning tyre pressure" activated  ?  Just to be sure things not mess upp.  (This is perhaps overkill ?) 

6. Decide the spelling of "Tires" is not easy: Googled: "Tire and tyre both mean a covering for a wheel, usually made of rubber. Tire is the preferred spelling in the U.S. and Canada. Tyre is preferred in most varieties of English outside North America. Of course, all English speakers use tire in the sense to grow weary."  No problem either way I think  :-)  

And again, thank you very much for the development of this very useful app, I will use it at least twice a year, changing the winter tires ;-)

softamateur, 2019-11-27:
Important one almost forgot to mention is: 

7.  In the display of pressures,  Rear righ and Rear left still seem to be switched comparing to the Autel TS-401 reading. 

Cheers!

yoh-there, 2019-11-27:
Great news!
1. Should be gone. I will upload beta4 in a few minutes
2. Sure. This is fetched online and I will remove it as soon as this is in the production release
3. Agreed and done in beta4
4. I hear you loud and clear, but won't do. If one decides to mess with wheels, itś  up to the user. I fif build in the compare thought (see comment @minibiti )
5. Same ;-)
6. I know. Consensus is: Tyres is UK spelling. Everywhere else and growing also in the UK is Tires. So as the code was inconsistent I had to do something and decided to go for Brexit Tires ;-)

Thank for testing and the heads up!

yoh-there, 2019-11-27:
7. Triple checked again with the docs, again now for the state and pressures too. Truth I guess is "in the middle"; either documentation or Autel is wrong :-( . I'll leave it for now.

yoh-there, 2019-11-27:
Thinking about the  swapping of pressures (ID is fine now I think?) some more. As said, I triple checked both the ISOTP as well as the free frame definitions, and they both are most definitely according to three sets of documentation.

If you have time, you might want to check pressures with Use ISOTP Fields setting off. I would suggest that:
- If they are then the same as the Autel, it would suggest the ISOTP documentation might be in error.
- If they are again in disagreement, I would tend to think the Autel firmware is wrong, as three sets of documentation from reputable source would then disagree with it. I wouldn't blame them, it is a very easy error to make, said the  guy who did!

softamateur, 2019-11-27:
...or just drain  Right Rear of 0.2 Bar and se what will be reported on the CanZE pressure display? :-)
I am a little bit curious so I will try this in late evening....

yoh-there, 2019-11-27:
Yes please!

yoh-there, 2019-11-27:
But if you do, please check both with ISOTP on and off, because those are different queries to the car. Thanks!

[33 more replies]

closed 2019-12-01 by yoh-there