fesch/CanZE #578: Clima RPM

gpezzarini, 2019-11-07:
I dont'understand the value, I can read only "5,0", it don't change when the Engine Fan Speed work
ZOE R240 2015

[image]
Best regards

yoh-there, 2019-11-10:
Engine fan speed should indicate the speed of the beefy radiator blower at the front, cooling (or heating) the big climate system radiator. I think in this picture you are charging and the heat pump is not working. Are you sure said fan was running? (it is pretty loud and easily detectable)

yoh-there, 2019-11-10:
Yes, that was an error. The value is given as %. There is a release out just a few hours ago, so the fix is made, but it will have to wait until the next release in a couple of weeks. Thanks for reporting.

closed 2019-11-10 by yoh-there