fesch/CanZE #579: Conversion of accelerometer data 12E

asmusskar, 2019-11-08:
Hi, 

there seems to be an issue reading accelerometer data (12E). 

Example: for the raw data 12E#C77FFC7FE0FFFF00 i get 17,-3 and -0.1 whereas others report values ranging from 200-4000. The output is far from the expected values (units m/s, g and deg/s assumed, respectively). Could anyone provide the correct conversion of this example?

Thanks a lot!

Asmus

yoh-there, 2019-11-08:
For a ZOE, in 12E there is only one valid value, being the YawRate. Unit is degrees per second.

`Definition: ,12e,24,35,0.1,2047,1,deg/s,,,ff,Yaw rate`

(see https://github.com/fesch/CanZE/blob/Development/app/src/main/assets/FFC_Fields.csv )

Your data: 0x7FE = 2046 => minus 2047 = -1 => times 0.1 = -0.1 °/s

I know there are definitions like
```
,12e,0,7,1,198,0,,,,ff,LongitudinalAccelerationProc
,12e,8,23,1,32768,0,,,,ff,TransversalAcceleration
```
But I can't remeber wher they came from and they are not in my formal documentation.

Please close this issue if this si what you needed.

closed 2019-11-09 by asmusskar

asmusskar, 2019-11-09:
Thanks. Is there a list available of sensor data applicable to ZOE?

yoh-there, 2019-11-10:
Please don't double post your questions, or add new questions on an existing, closed issue. All definitions can be found in the source code, assets folder where I linked to above.