fesch/CanZE #581: Tire valve ID's cannot be written...

AndreasBahrt, 2019-11-12:
... From CanZE v.1.43 (Android 7.1.2).

When write of the TPMS ID's are tried, message **Could not read write valves:-E-ISOTP tx flow error** appears...

Thank you!

[image]
[image]

yoh-there, 2019-11-12:
Duplicate of #554. Will you please try with the latest update, and report back in that issue? It's possibly a timing issue. Thank you!

closed 2019-11-12 by yoh-there

AndreasBahrt, 2019-11-13:
CanZE is now updated to version 1.46. Please look at the following messages which are revealed in CanZE, when the write button is pressed...
[image]
[image]
[image]

AndreasBahrt, 2019-11-13:
Read does not work either...
[image]

AndreasBahrt, 2019-11-13:
... I intend to try another Android device, when I find one...

yoh-there, 2019-11-13:
https://github.com/fesch/CanZE/issues/577