fesch/CanZE #582: Support tire pressure sensor ID readout and edit in iOS CanZE app

PetMei, 2019-11-12:
See title - I was informed that this is possible with the Android app but not with the iOS app which I use. Please add this functionality so I can update the IDs after summer/winter tire change (Zoe) on my own.

Cheers -
Peter

FredLeudon, 2019-11-12:
I’ll do that as soon I have time to work n CanZE for iOS.

Be patient, it will arrive on beta version soon. 

Frédéric RICHARD

PetMei, 2019-11-12:
OK - looking forward to version 4.x  ;-)

Cheers –
Peter

yoh-there, 2019-11-12:
@FredLeudon Please be aware that **sending** multi-frame ISOTP messages is another ball game. I am not 100% sure we got it right in Android as we still get issues reported.

FredLeudon, 2019-11-12:
I saw.

I have to make some tests but my poor (~ 296000 km Zoé) doesn’t have tires pressure calculator. 

I will perhaps test  isotp frame sending on ecu « display » and play with clock & temperature display.

Will report if it will be successful or not ;-)

Frédéric RICHARD

yoh-there, 2019-11-12:
Whoa, you pushed her hard!! Same here Q210. If you have a working test, I would happily repeat it on mine. Of course I would need a functional description on what you send to the display cluster ;-)

yoh-there, 2019-11-13:
@AndreasBahrt confirmed this issue on the latest update too, so the timing change did not work. I have limited resources (time) at this moment but will work on it when I can.

FredLeudon, 2019-11-15:
Hello.

I successfully display/hide clock and/or temperature on cluster.

Sending iso-tp frames with data works great for me*.
I just have to add  a cluster soft reboot after writing process.

About TPMS captors, how users can get new IDs?
Are IDs directly written on captors ? Are they numeric or hexadecimal ?

I will add tpms writing capability  in next beta, but I won’t be able to test :-(

If beta testers return it’s working fine, I’ll share :-) 

* I actually can send simple data (data items with no scale and or offset). I have to work on this ;-)

Regards

Frédéric RICHARD

yoh-there, 2019-11-15:
Unrelated to this issue: can you please email what messages you sent to the cluster, just to fool around a bit ;-)

Sometimes valves come with printout, but usually you have to have them read by the dealer at least once. See https://canze.fisch.lu/tpms-ids-can-now-be-read-and-written/ (you need a LF transmitter to force the valves to transmit their ID, and also set the car in "learn" mode).

TMPS write does **not** work in Android yet. I still have issues with long ISO-TP messages, I will try tomorrow. Luckily I KNOW the message itself is OK, I tested that with the CanSee dongle on a TPMS equiped car.

FredLeudon, 2019-11-18:
@yoh-there, same as TPMS ;-)

For example :

`18/11/2019	08:34:43,02 [i..dp]-[cl_Vehicle.requestField, ligne 90                                     ] :       	*	SID[763.220122.4.0]	Data[Config Generale_Affichage temp exter]	Value[Avec affichage température extérieure (par défaut)]	Raw Value[16]
18/11/2019	08:34:51,27 [i..dp]-[cl_Vehicle.requestField, ligne 4                                      ] :       	>	:requestField(ecu[51],request[objet][DataWrite.Config Generale_montre]	Virtual[0]
18/11/2019	08:34:51,28 [i..dp]-[cl_Request.build_sendBytes, ligne 9                                   ] :         	SID[763.2E012100.4.0]	Data[Config Generale_montre]	Value[sans affichage montre ( par défaut)]	Raw Value[0]
18/11/2019	08:34:51,28 [i..dp]-[cl_Vehicle.sendELM, ligne 3                                           ] :         	>	:sendELM(elm[objet], ecu_request[objet], command[2E012100], auto[1]
18/11/2019	08:34:51,28 [i..dp]-[cl_ELM.port_write, ligne 3                                            ] :                     	CanBus[1]	>	Ecriture ELM : data[042E012100<\n>]
18/11/2019	08:34:51,30 [i..dp]-[cl_ELM.port_write, ligne 49                                           ] :                     	CanBus[1]	<	Ecriture ELM : Ok
18/11/2019	08:34:51,68 [i..dp]-[cl_Vehicle.requestField, ligne 4                                      ] :       	>	:requestField(ecu[51],request[objet][DataWrite.Config Generale_Affichage temp exter]	Virtual[0]
18/11/2019	08:34:51,68 [i..dp]-[cl_Request.build_sendBytes, ligne 9                                   ] :         	SID[763.2E012210.4.0]	Data[Config Generale_Affichage temp exter]	Value[sans affichage température ext]	Raw Value[0]
18/11/2019	08:34:51,68 [i..dp]-[cl_Vehicle.requestField, ligne 54                                     ] :       	Construction Bytes To Send	Ecu bytes to send (original) = [2E012210]	ECU bytes to send (calculé) = [2E012200]
18/11/2019	08:34:51,68 [i..dp]-[cl_Vehicle.sendELM, ligne 3                                           ] :         	>	:sendELM(elm[objet], ecu_request[objet], command[2E012200], auto[1]
18/11/2019	08:34:51,69 [i..dp]-[cl_ELM.port_write, ligne 3                                            ] :                     	CanBus[1]	>	Ecriture ELM : data[042E012200<\n>]
18/11/2019	08:34:51,70 [i..dp]-[cl_ELM.port_write, ligne 49                                           ] :                     	CanBus[1]	<	Ecriture ELM : Ok
18/11/2019	08:34:51,95 [i..dp]-[cl_Vehicle.requestField, ligne 4                                      ] :       	>	:requestField(ecu[51],request[objet][Commande reset soft]	Virtual[0]
18/11/2019	08:34:51,95 [i..dp]-[cl_Vehicle.sendELM, ligne 3                                           ] :         	>	:sendELM(elm[objet], ecu_request[objet], command[1103], auto[1]
18/11/2019	08:34:51,95 [i..dp]-[cl_ELM.port_write, ligne 3                                            ] :                     	CanBus[1]	>	Ecriture ELM : data[021103<\n>]
18/11/2019	08:34:51,96 [i..dp]-[cl_ELM.port_write, ligne 49                                           ] :                     	CanBus[1]	<	Ecriture ELM : Ok
18/11/2019	08:34:52,26 [i..dp]-[cl_Vehicle.requestField, ligne 4                                      ] :       	>	:requestField(ecu[51],request[objet][DataRead.Config Generale_montre]	Virtual[0]
18/11/2019	08:34:52,26 [i..dp]-[cl_Vehicle.sendELM, ligne 3                                           ] :         	>	:sendELM(elm[objet], ecu_request[objet], command[220121], auto[1]
18/11/2019	08:34:52,27 [i..dp]-[cl_ELM.port_write, ligne 3                                            ] :                     	CanBus[1]	>	Ecriture ELM : data[032201211<\n>]
18/11/2019	08:34:52,28 [i..dp]-[cl_ELM.port_write, ligne 49                                           ] :                     	CanBus[1]	<	Ecriture ELM : Ok
18/11/2019	08:34:52,40 [i..dp]-[cl_Vehicle.requestField, ligne 90                                     ] :       	*	SID[763.220121.4.0]	Data[Config Generale_montre]	Value[sans affichage montre ( par défaut)]	Raw Value[0]
18/11/2019	08:34:52,41 [i..dp]-[cl_Vehicle.requestField, ligne 4                                      ] :       	>	:requestField(ecu[51],request[objet][DataRead.Config Generale_Affichage temp exter]	Virtual[0]
18/11/2019	08:34:52,41 [i..dp]-[cl_Vehicle.sendELM, ligne 3                                           ] :         	>	:sendELM(elm[objet], ecu_request[objet], command[220122], auto[1]
18/11/2019	08:34:52,41 [i..dp]-[cl_ELM.port_write, ligne 3                                            ] :                     	CanBus[1]	>	Ecriture ELM : data[032201221<\n>]
18/11/2019	08:34:52,43 [i..dp]-[cl_ELM.port_write, ligne 49                                           ] :                     	CanBus[1]	<	Ecriture ELM : Ok
18/11/2019	08:34:52,54 [i..dp]-[cl_Vehicle.requestField, ligne 90                                     ] :       	*	SID[763.220122.4.0]	Data[Config Generale_Affichage temp exter]	Value[sans affichage température ext]	Raw Value[0]
18/11/2019	08:34:54,74 [i..dp]-[cl_Vehicle.doNextRequest, ligne 15                                    ] :     	Next request_sSID to request[763.220121]
18/11/2019	08:34:54,74 [i..dp]-[cl_Vehicle.requestField, ligne 4                                      ] :       	>	:requestField(ecu[51],request[objet][DataRead.Config Generale_montre]	Virtual[0]
[... cut ...]
18/11/2019	08:35:09,55 [i..dp]-[cl_Vehicle.sendELM, ligne 3                                           ] :         	>	:sendELM(elm[objet], ecu_request[objet], command[1103], auto[1]
18/11/2019	08:35:09,55 [i..dp]-[cl_ELM.port_write, ligne 3                                            ] :                     	CanBus[1]	>	Ecriture ELM : data[021103<\n>]
18/11/2019	08:35:09,57 [i..dp]-[cl_ELM.port_write, ligne 49                                           ] :                     	CanBus[1]	<	Ecriture ELM : Ok
`

FredLeudon, 2019-11-18:
Hi all,

New iOS beta release (4.0.0.36) ais available for beta-testing

In this version, it is possible to show/hide temperature and/or clock on cluster (tested on my 2013 Zoé).

I implemented the ability of writing TPMS IDs. (Have to be tested on Zoe equipped with TPMS, feedback appreciate)

Link to download the beta is : https://testflight.apple.com/join/nhMX6f5n

yoh-there, 2019-11-18:
Interesting, I never realized some mild WRITING was performed by issuing commands starting with "2". I assumed writes had to start with "3". Assumptions, the mother of all F-ups. ;-)

FredLeudon, 2019-11-18:
[x10_[51]_[Rx763-Tx743]_ECU_Tableau de bord_20190807110222900.dump.log](https://github.com/fesch/CanZE/files/3858320/x10_.51._.Rx763-Tx743._ECU_Tableau.de.bord_20190807110222900.dump.log)

Attached file contains ALL requests and datas from ECU Cluster

yoh-there, 2019-11-18:
Like you I use the XML files from ddt :-) I just convert them for convenience to internal CSVs for the app as they are way to bulky. Good stuff Fred!

AndreasBahrt, 2019-11-18:
I have intalled the Testflight iOS app. To join the CanZE iOS beta release, I miss an invitation E-mail, a public link or an activation code.

Thanks!

FredLeudon, 2019-11-18:
https://testflight.apple.com/join/nhMX6f5n

Frédéric RICHARD

AndreasBahrt, 2019-11-23:
SUCCESS! With CanZE iOS beta release 4.0.0.39
[image]

AndreasBahrt, 2019-11-23:
... The pressure readouts of 3,5 Bar are not correct though.

yoh-there, 2019-11-23:
Great news!

@FredLeudon do you send any diagnostic session start or tester present messages before you write the values (ie 50c0 or 7e01)? If so, which ones please?

yoh-there, 2019-11-23:
@FredLeudon The 3501.15 doesn't make sense to me as the multiplier is 13.725 according to my data. Even 255 from the sensors doesn't get to that. Do you have different definitions?

FredLeudon, 2019-11-23:
@yoh-there, I don't send any other than the request.

Will try to send data in development tool and send the full data sent.

Tires pressure are read from request 765.2174.15 / 16 / 17 / 18

Will send info later (I have to analyze the request)

FredLeudon, 2019-11-23:
Request[765.2174] : Read_TPMS___mesure_pression
		RawData : [61 74 04 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF ]
		< Receive    [765.2174.10.0] : RearRightWheelState - (0) Pressure OK
			Values list : 
				0	Pressure OK
				6	Tyre under inflated
				5	Flat Tyre
				2	Tyre over inflated
				1	No pressure information (sensor missing)
		< Receive    [765.2174.11.0] : RearLeftWheelState - (0) Pressure OK
			Values list : 
				0	Pressure OK
				6	Tyre under inflated
				5	Flat Tyre
				2	Tyre over inflated
				1	No pressure information (sensor missing)
		< Receive    [765.2174.12.0] : FrontRightWheelState - (0) Pressure OK
			Values list : 
				0	Pressure OK
				6	Tyre under inflated
				5	Flat Tyre
				2	Tyre over inflated
				1	No pressure information (sensor missing)
		< Receive    [765.2174.13.0] : FrontLeftWheelState - (0) Pressure OK
			Values list : 
				0	Pressure OK
				6	Tyre under inflated
				5	Flat Tyre
				2	Tyre over inflated
				1	No pressure information (sensor missing)
		< Receive    [765.2174.14.0] : Right/LeftImbalance - (0) normal
			Values list : 
				1	front right-left imbalance
				2	rear right-left imbalance
				3	front and rear right-left imbalance
				0	normal
		< Receive    [765.2174.15.0] : RearRightWheelPressure - (3501.15) 3501.15 mBar
		< Receive    [765.2174.16.0] : RearLeftWheelPressure - (3501.15) 3501.15 mBar
		< Receive    [765.2174.17.0] : FrontRightWheelPressure - (3501.15) 3501.15 mBar
		< Receive    [765.2174.18.0] : FrontLeftWheelPressure - (3501.15) 3501.15 mBar
		< Receive    [765.2174.3.2] : SSPP Défaut acq vitesse véhicule (RF) - (0) 0
		< Receive    [765.2174.3.3] : SSPP Défaut acq vitesse véhicule (IHM) - (0) 0
		< Receive    [765.2174.3.4] : SSPP Jeu1 présence 4 codes - (0) 0
		< Receive    [765.2174.3.5] : SSPP Jeu0 présence 4 codes - (1) 1
		< Receive    [765.2174.3.6] : SSPP Jeu1 codes identiques - (0) 0
		< Receive    [765.2174.3.7] : SSPP Jeu0 codes identiques - (0) 0
		< Receive    [765.2174.5.0] : SSPP Defaut coherence accel roue 1 - (0) 0
		< Receive    [765.2174.5.1] : SSPP Defaut coherence temp roue 1 - (0) 0
		< Receive    [765.2174.5.2] : SSPP Defaut coherence pression roue 1 - (0) 0
		< Receive    [765.2174.5.3] : SSPP Defaut pile roue 1 - (0) 0
		< Receive    [765.2174.5.4] : SSPP Defaut mesure accel roue 1 - (0) 0
		< Receive    [765.2174.5.5] : SSPP Defaut mesure temperature roue 1 - (0) 0
		< Receive    [765.2174.5.6] : SSPP Defaut mesure pression roue 1 - (0) 0
		< Receive    [765.2174.5.7] : SSPP Defaut chks roue 1 - (0) 0
		< Receive    [765.2174.6.0] : SSPP Defaut coherence accel roue 2 - (0) 0
		< Receive    [765.2174.6.1] : SSPP Defaut coherence temp roue 2 - (0) 0
		< Receive    [765.2174.6.2] : SSPP Defaut coherence pression roue 2 - (0) 0
		< Receive    [765.2174.6.3] : SSPP Defaut pile roue 2 - (0) 0
		< Receive    [765.2174.6.4] : SSPP Defaut mesure accel roue 2 - (0) 0
		< Receive    [765.2174.6.5] : SSPP Defaut mesure temperature roue 2 - (0) 0
		< Receive    [765.2174.6.6] : SSPP Defaut mesure pression roue 2 - (0) 0
		< Receive    [765.2174.6.7] : SSPP Defaut chks roue 2 - (0) 0
		< Receive    [765.2174.7.0] : SSPP Defaut coherence accel roue 3 - (0) 0
		< Receive    [765.2174.7.1] : SSPP Defaut coherence temp roue 3 - (0) 0
		< Receive    [765.2174.7.2] : SSPP Defaut coherence pression roue 3 - (0) 0
		< Receive    [765.2174.7.3] : SSPP Defaut pile roue 3 - (0) 0
		< Receive    [765.2174.7.4] : SSPP Defaut mesure accel roue 3 - (0) 0
		< Receive    [765.2174.7.5] : SSPP Defaut mesure temperature roue 3 - (0) 0
		< Receive    [765.2174.7.6] : SSPP Defaut mesure pression roue 3 - (0) 0
		< Receive    [765.2174.7.7] : SSPP Defaut chks roue 3 - (0) 0
		< Receive    [765.2174.8.0] : SSPP Defaut coherence accel roue 4 - (0) 0
		< Receive    [765.2174.8.1] : SSPP Defaut coherence temp roue 4 - (0) 0
		< Receive    [765.2174.8.2] : SSPP Defaut coherence pression roue 4 - (0) 0
		< Receive    [765.2174.8.3] : SSPP Defaut pile roue 4 - (0) 0
		< Receive    [765.2174.8.4] : SSPP Defaut mesure accel roue 4 - (0) 0
		< Receive    [765.2174.8.5] : SSPP Defaut mesure temperature roue 4 - (0) 0
		< Receive    [765.2174.8.6] : SSPP Defaut mesure pression roue 4 - (0) 0
		< Receive    [765.2174.8.7] : SSPP Defaut chks roue 4 - (0) 0
		< Receive    [765.2174.9.2] : SpeedPressureMisadaptation - (0) No speed pressure misadaptation
			Values list : 
				0	No speed pressure misadaptation
				1	Speed pressure misadaptation

yoh-there, 2019-11-23:
@FredLeudon Thank you re. setting ID's.

For the pressure, you are 0bviously receiving 0xFF (=255). First wheel starts at bit 112 = byte 15 (counting from 0). Probably FF means unavailable and I think we should translate it into 0 or a dash (-).

As for the pressure, I see you use 13.73 from the XML. Another source states it as 13.725 (step 549, divideby 40), so that explains the difference. Anyway, it's so small we can ignore it.

@AndreasBahrt Do you realize the pressures are only valid if the car has driven a few 100 meters? I noticed this when doing research and driving in a TPMS equipped ZOE. Could you either try again or confirm that this reading was done while driving? Thank you!

FredLeudon, 2019-11-24:
beta teste reports Pressure is correct while driving

[image]

yoh-there, 2019-12-01:
@FredLeudon Only FYI: I don't know if you have followed the other TPMS thread ( #577 ) but there has been a lot of confusion about wheel positions versus fields. I am sure it's 100% correct in Android now (and I documented it too in the source). It often goes wrong as the car has only one antenna and if the user doesn't know the real ID of the tire, but just believes what the dealer did, well, it can be messy.

The XMLs are correct, but two testers used actual 433 MHz receivers to ensure the proper positions.

FredLeudon, 2019-12-01:
Yes, I saw. 

Waiting for iOS users returns. 

And I will have a look at the code on github to check iOS and android version as saying same thinks. 

Frédéric RICHARD

yoh-there, 2019-12-14:
Just out of interest: are you happy and if so, should we close this issue? No pressure from my end at all, but I do tend to stay on top of things ;-)

AndreasBahrt, 2019-12-15:
Yes, I am very satisfied and thanks for your great effort J

closed 2019-12-15 by fesch