fesch/CanZE #588: New Zoe Z.E. 50 and iOS CanZE.

FredLeudon, 2019-11-30:
Hi, all

I could approach a new Z.E. 50 Zoe yesterday.

After adding the first 9 digits of the vin to my online vin decoder database, I could identify the new Zoe with the latest iOS beta.

As the car was only in the showroom, I couldn’t start it but it was possible to CanZE to identify at least one ecu.

I wish I can spend one hour next week to plug my dongle in a new Zoe and use CanZE to identify all ecu’s

If it work’s, I’ll publish the results ;-)

yoh-there, 2019-11-30:
Nice. What command did you send? 7df (broadcast)? That is what the tool of the general mechanic did and he indeed pulled out the VIN but nothing more

FredLeudon, 2019-11-30:
First, to read the VIN, 

For each address (26, 27, 51, 67, 68, 77, 95), I send an isotp request (xxx.2181) on can network with can protocol first and if vin is not read, same request with kwp2000 protocol.

Next step is to analyze ecus

For each address available (my vehicle database comes from ddt2000 files)

I first send start session can (1003)
Next I send 22F1A0 request
Next 22F18A
Next 22F194
And finally 22F195

So I get Diagversion, supplier, soft, etc

Next step is to find  ecu or approximate ecu or a list of available ecu corresponding to data I get.
(I have build a database of ecu by analyzing and aggregating different ddt 2000 databases I found)

And finally, download xml ecu definition from my server and building internal database ecu -> request -> send & receive data items -> datas 

Don’t know if I my explanations are limpidity :-)

Frédéric RICHARD

yoh-there, 2019-11-30:
Yes it makes sense :-) I am very curious if all the ECU's responded and if not, which one did.

I am pretty sure there are no free frames on the Diagnostic bus and it looks like the MM bus has been removed.

yoh-there, 2019-12-01:
What exactly do you mean with "for each address"? I don't understand as those (26, 27, ...) are not ECU CAN addresses. I would expect 765/745, 7bc/79c etcetera.

FredLeudon, 2019-12-01:
Each address can be mapped into rx and tx address.

I think I got this list on a vehicle.xml file in ddt2000 database. 

I check this tomorrow if I have time.

Frédéric RICHARD

yoh-there, 2019-12-01:
Oh I see what you mean, I think you meant eculist.xml.

The universal car diagnostic computer issues a broadcast. But I couldn't see which ECU responded with the VIN. Oh well, next time ;-)

FredLeudon, 2019-12-01:
There is no link between ecu list and vin

I use vin as primary key to store ecu information on my database

This is to avoid to rescan ecus after each install or reinstall.

Frédéric RICHARD

FredLeudon, 2019-12-01:
Vin is necessary to know witch vehicle CanZE is talking with ;-)

Frédéric RICHARD

yoh-there, 2020-01-01:
Someone in the German forum suggested the ZE50 uses 29 bit CAN addressing. If true I would be very surprised as I would not have suspected such a serious change. Then again, if it's true , the problem is probably simpler to solve than we thought, and my sniffer might just have been ignoring all the frames. Do you know more already Frédéric?

FredLeudon, 2020-01-01:
Happy new year !

At this time, no infos.

I will meet ZE50 user on Monday January 13.

Wait and see. 

Regards,

Frédéric RICHARD

yoh-there, 2020-01-01:
Same to you and the (new) family ;-)

Cool (Monday). If you bring any sniffer material, remeber the 29 bits stuff. My current sniffer (CanSee) does throw numbers other than standard 11 bits addressing away, so maybe I have been working with completely wrong assumptions all along. I will try to have a bit more confirmation before Monday and will post here if so.

FredLeudon, 2020-01-01:
Thanks. 

At first time, i will do my Ecu scan procedure to load right ecu xml file. 

Once done, I will analyze request, data items , data’s ... and I will try to add Z.E. 50 to my CanZE sid array.

Regards 

Frédéric RICHARD

yoh-there, 2020-01-01:
Can you maybe send me the very very most recent xml's you could find please? Google does drive me nuts on this one. I would appreciate it.

KarelSvo, 2020-01-01:
> Someone in the German forum suggested the ZE50 uses 29 bit CAN addressing. If true I would be very surprised as I would not have suspected such a serious change. 

All new Renault, Dacia, Nissan on C1A electronics platform use 29 bit for the drive control units.

yoh-there, 2020-01-01:
Thanks @KarelSvo , good to know. The guy from the German forum suggested only the motor controller used extended addressing, the remainder standard. Mixing is allowed on the network level of course but ick!

yoh-there, 2020-01-12:
The 11 <> 29 bits addressing is quite a bit more complicated. Many new or upgraded boxes, some 11 bits, some 29, some both. I will amend the massive spreadsheet (ECU tab) as soon as I can.

yoh-there, 2020-02-16:
First Beta is out, actually more alpha regarding the ZE50 stuff. The ECU tab in the spreadsheet is slowly being ammended,

arqueox, 2020-02-24:
There is something i can do to use a beta version of ios app on my ZE50? I could try and give you my feedback about.

arqueox, 2020-02-24:
> First Beta is out, actually more alpha regarding the ZE50 stuff. The ECU tab in the spreadsheet is slowly being ammended,

No IOS beta app yet?

FredLeudon, 2020-02-24:
Hi,

A new iOS beta app will be available soon 

Regards

Frédéric RICHARD

arqueox, 2020-02-26:
> Hi, A new iOS beta app will be available soon Regards Frédéric RICHARD
> […](#)

Oh thnx! I’ll wait then

Steckmaestro, 2020-02-28:
I can borrow this.

I tried CanZE for Android using Samsung S8. Appversion 1.52beta4.

Tried with Zoe R135 (ZE50) - Swedish car.

I used two different ODBII dongles (first a bluetooth one I bought previously and after reading your recommendation I ordered Konweii and tried with that one.

Hopefully debug logs can tell you something. I'm really looking forward to be able to see more detailed information about my Zoe :) 

[debug-2020-02-27-17-38-40.log](https://github.com/fesch/CanZE/files/4266390/debug-2020-02-27-17-38-40.log)

KarelSvo, 2020-02-28:
> I'm really looking forward to be able to see more detailed information about my Zoe :)
Yes there is a python kivy app that can show such information ;) 
tested on new Zoe and connected to over 20 ecus

yoh-there, 2020-02-28:
Getting there. Seems like the braking system has changed significantly, so we might need to defer the torques, but there is progress. Getting info out is "cracked". Getting the same info presented consistently is a different game.

ik2wxx, 2020-03-02:
Hello, how can I apply for iOS beta test for Zoe ZE50 ?
Thank you

kkornel13, 2020-03-02:
Hey guys,
A month ago I used CanZE on Reanult Zoe Q90 with LELink iOS bluetooth adapter. I connacted it with my iPad, and it worked really well. But I would like to export the real-time consumption datas (the numbers), but the app does not send the log files. So my question is how can I export the log files? Unfortunatelly I am not exactly a tech expert, so every help is greatly appreciated.
Thank you very much

abrensch, 2020-03-23:
Hi,

I have a couple of question on CanZE / new ZOE / Bluetooth Dongle, hoping someone can steepen my learning curve.

I'm expecting my ZOE R 110 Z.E. 40 next week (seems I made it before the production shutdown at 16.3.2020 ...). I was occasional User of CanZE with car-sharing rents of ZOE 

- am I right that, though it's not ZOE 50, your thoughts here apply, because it's the face-lifted version where the ODB2 connector is near your left foot ?

- can I expect that CanZE will (someday) work on a ELM327 clone, or do I need the CanSee hardware?

I am working on integrated range assistence that need to read realtime data from the ZOE. It's only a very small subset of what CanZE can read (just speed, SOC, amps and volts). I append below some pseudo code, that is working with a KONNWEI Bluetooth dongle and an "old" ZOE 40. Any chance to get that running on the "new" ZOE40?

thanx, Arndt 

here the pseudo-code (that was inspired by what's inside CanZE):

            reader.writeToElm("atz");  // full reset
            reader.writeToElm("ate0");  // no echo
            reader.writeToElm("ats0");
            reader.writeToElm("ath0");  // header off
            reader.writeToElm("atl0");
            reader.writeToElm("atal");
            reader.writeToElm("atcaf0");
            reader.writeToElm("atfcsh77b");
            reader.writeToElm("atfcsd300000");
            reader.writeToElm("atfcsm1");
            reader.writeToElm("atsp6");

            // now periodic commands
            for (; ; ) {
                .... cycle wait ...
                reader.writeToElm("atar");
                reader.writeToElm("atsh7e4");
                reader.writeToElm("atfcsh7e4");

                reader.writeToElm("03223203");
                p = reader.getResult().get(0);
                reading.lbcvolt = read2(p,4) * 0.5;

                reader.writeToElm("03223204");
                p = reader.getResult().get(0);
                reading.lbcamps = (32768 - read2(p,4))*0.25;

                p = reader.readFreeFrame( "5d7", 7 );
                reading.speed = 0.01 * read2(p, 0);

                p = reader.readFreeFrame( "427", 0 );
                reading.aenergy = (read2(p, 6) >> 6 )* 0.1;
            }

yoh-there, 2020-03-23:
If it's the ZOE with the OBD2 on the far left side, consider it for CanZE purposes, a ZE50, sorry

You do not need a CanSee dongle. It nice and fast, but decent ELMs do work for the ZE50.

That code won't work at the moment. The free frames are not present on the OBD2 connector. The others work "in principle". LBC commands work for sure, though they are different (unless your car uses the "old" battery computer, I don't know that). You need to open the secure gateway (easy, just one command) and then do exactly the same. That is already in the code base.

KarelSvo, 2020-03-29:
> here the pseudo-code (that was inspired by what's inside CanZE):
>
> ```
>         reader.writeToElm("atz");  // full reset
>         reader.writeToElm("ate0");  // no echo [...]
Hi
The free frame 5d7, on phase 1 transmits from the ABS but on phase 2 with it Cell voltages sent by the BMS. Phase 2 uses the new Renault Nissan electronics platform.
You must to build your project for phase 2 from scratch.  For more information, give an email address.

abrensch, 2020-03-29:
Thank you guys for making me hopeful that I can get that working

my email is : Arndt dot Brenschede at web dot de

> The free frame 5d7, on phase 1 transmits from the ABS but on
> phase 2 with it Cell voltages sent by the BMS. Phase 2 uses the
> new Renault Nissan electronics platform. You must to build your
> project for phase 2 from scratch. For more information, give an
> email address.

For speed (that I was reasding from free frame 5d7) I have the GPS-speed as a fallback, just in case.

The available energy read from free frame 427  is 0.1 kWh in resolution, which is not enough for my use case, so I added some averaging/filtering with the electric power. I have a similar adapter working for Nissan Leaf (at least the older models) where there's a 6-digit "micro-soc" that worked out-of-the box. So I am curious what's possible for ZOE ph2.

abrensch, 2020-04-05:
> I am working on integrated range assistence that need to read
> realtime data from the ZOE. It's only a very small subset of what
> CanZE can read (just speed, SOC, amps and volts).

I got that working (with some Help from Karel)

Reading 5 fields now per cycle from the EVC only, in that order:
- 22303E kWh per SOC %
- 2220BE State of charge
- 2220FE volts (measured at the battery)
- 2221CC amps (measured at the battery)
- 223064 Motor RPM ( rpm*0.012325 = speed/kmh )

It's about 400ms (80ms per field).

Actually, it works better then my solution for Phase1 ( faster, less latency, better resoltion SOC)

Thanx again

ZE135, 2020-06-04:
Hi, I would like to use CanZE as well with my new ZE50.

@abrensch: Can you describe how to did read out? I am sure many ZE50 users are very much interested. 

Thanks in advance

TomC-Zoe, 2020-08-02:
Hello, I have made a CanSee dongle, it is recognized by CanZe (beta), however I have only "fail" messages with my Zoe ZE50, and I cannot find any dump file created by CanZe. Did I forget something ? Do you have an idea to break the deadlock ?
Thank you

FredLeudon, 2020-08-02:
Hi, 

CanSee dongle is still under development.

COVID stopped CanZE development for my part. I plan to work again on CanZE and CanSee integration on September. Sorry for the delay.

Regards.  

Frédéric RICHARD

KarelSvo, 2020-08-02:
>  Do you have an idea to break the deadlock ?

Applications for Android, Windows and Linux have been around for a long time and can read ZE50 data without any problems.
Only a compatible Elm interface is required.

TomC-Zoe, 2020-08-04:
The problem is that I gave my ELM dongle to a friend, as I thought that CanSee was operationnal, or at least could read some data from the ZE50. So I will wait until further developments.

g7pfv, 2020-09-01:
Where is the OBD2 PORT ON THE ze50 R135 ?

i cant seem to locate it !

yoh-there, 2020-09-01:
https://canze.fisch.lu/ze50/

g7pfv, 2020-09-01:
Thankyou - just tried Powercruisecontrol app and works perfectly

 
 

https://canze.fisch.lu/ze50/

MarkRotNF, 2020-10-01:
Are there any news for the ZE50 support? Information in the app about the charging speed would be very nice :D

FredLeudon, 2020-10-01:
Hi,

I have now a new Zoe so I can check data on Ecus.

I’m working hard, but there is a lot of work.

I will give infos asap

Cordialement,

Frédéric RICHARD

miguel-negrao, 2020-11-18:
Hi

I have a zoe ze50, a konnwei bluetooth dongle and programming knowledge. Is there anything I can do to help get ze50 support ?

Leobaillard, 2020-11-18:
Same here and I have an oscilloscope, logic analyser and CAN hardware as well.

yoh-there, 2020-11-19:
@miguel-negrao @Leobaillard My advice would be to fork the project into Android studio and get a feeling on how it works. All the ZE50 specific code is done (I think), but you probably want to add specific debug code to figure a new problem.

The issue is more getting things done field-by-field. I have XML files that describe the new ECUs, but only part of the fields work. It's really painstakingly defining and testing each single one on the car.

If you don't fancy digging into Android Studio, you can also use the Research screen in CanZE to check fields, but it's results would still need to be incorporated in the app.

Of course you could concentrate immediately only on getting some parameter from the car, but you would have to make custom code for your dongle to get to that; just typing commands would not work as things would time out before you got everything done. Especially the ELMs is a deep rabbit hole just by itself. If you really want to go this route, I strongly advise building your own CanSee dongle as it's super fast and fully customizable.

For Bob and myself, the main issue is lack of both time and a car. It's proven to be truly impossible to get this done without having both. I had a ZE50 on loan for a week, and then time collapsed. When I have time, I don't have a car.

If there is anything I can do to help, please do reach out.

Leobaillard, 2020-11-19:
@yoh-there thanks for all the pointers and your hard work! I might have a go at it this weekend. I'm in France and we are on lockdown, so I guess I have time!

yoh-there, 2020-11-19:
Hah, yeah I know the deal :-( Needed, but no fun. Good luck and feel free to ask. It might be a bit uphill, but with the car available it should be much easier to give it a go.

ik2wxx, 2020-11-19:
My wife has a ZE50 so please count me in for ZE50 test and development.
I prefer iOS but I can try to help on Android.
Here in Italy there is already an Android app that can read the SOH for ZE50 (but costs a lot).
I have 3 OBD dongle but don't know very much of CAN bus.

yoh-there, 2020-11-19:
@ik2wxx Great, feel free to follow the pointers given above. Source code is in this githib repository.

How silly, I think if you download the CanZE beta, SOH actually works. The battery was about the easiest ECU to get working. So no need to pay!

I have no access to the iOS source. You might want to reach out to @FredLeudon for that.

KarelSvo, 2020-11-19:
> Here in Italy there is already an Android app that can read the SOH for ZE50 (but costs a lot).

There is Android software that can read all ZE50 data and it costs ZERO.

ik2wxx, 2020-11-19:
> How silly, I think if you download the CanZE beta, SOH actually works. The battery was about the easiest ECU to get working. So no need to pay!

Unless it only works while charging or other situations, for me it's not working.
But I'm here if I can help.
If anyone can tell us how we can help, it will be great.
I don't know anything about the app and I know a very little of CANBUS.

ik2wxx, 2020-11-19:
> There is Android software that can read all ZE50 data and it costs ZERO.This APP also writes logs to learn from.All you need is a WLAN or Bluetooth ELM that supports many AT commands.

What App ? 
CanZe ?

KarelSvo, 2020-11-19:
[2.2.apk](https://onedrive.live.com/?authkey=%21AN5YoZDHSTG0DQQ&id=864CCA38B98740D3%219133&cid=864CCA38B98740D3)

miguel-negrao, 2020-11-19:
@you-there I've built the app from source, using the commit 8640b0f76dc. I've installed it on my phone and connected to the bluetooth unit, but most reading are giving error. For instance, battery SOH gives error. I remember battery temperature and percentage were giving a value. Will try to look more into it.

yoh-there, 2020-11-19:
Great, you're moving at highway speed :-) In the "stupid questions" department: you DID select ZE50 as car didn't you? It will then select the ZE50 definition CSV file(s). I am fairly sure I got that working, but it's been like 7 months ago......

miguel-negrao, 2020-11-19:
@yoh-there yes  I did select ZE50. I had to checkout  8640b0f to be able to do it.

[image]

yoh-there, 2020-11-19:
Fair dinkum. But as you can see, it does actually successfully access the LBC, digging up temperatures. I think the heat maps actually both work, so the voltages graph here is just a definition problem in and a separate layout XML is probably the best way to go. 

I will drop my email address in your contact form. If you can reply I can send you some data that will I think will be of use. Thank you for the effort!

ik2wxx, 2020-11-21:
> In the "stupid questions" department: you DID select ZE50 as car didn't you?
Yes I did, but didn't get much info from CanZE, the only info is the accel pedal position.
I'm using a KONNWEI KW902 ELM327 V1.5, hope it's ok, I have other OBD dongle but this looks like it's the more compatible with CanZE.
Thank you

KarelSvo, 2020-11-21:
> EDIT yoh-there: be careful downloading apk's outside the Google Plays store.

Since the APP contains original data from Renault, it is not officially offered.
In contrast to CANZE, it works on the new Zoe.
CANZE also contains copyrighted material from Renault databases. You are walking on very thin ice.

yoh-there, 2020-11-21:
@KarelSvo I am warning other readers against bad practices, it was not aimed at you, though I would appreciate you not posting links to unchecked apk's. It's fine to post them on your own github account and link to those posts from here. Let me know when you've done this please.

Thank you for your warning re copyright.

yoh-there, 2020-11-21:
@ik2wxx I think you should at least get the two right hand graphs @miguel-negrao showed, and, but I am going from memory here, the two heat maps. Car must be switched on.

[22 more replies]

closed 2023-10-02 by yoh-there