fesch/CanZE #589: BT connection to CANSee freezes/crashes CANSee

rarisoft, 2019-12-03:
Hi, currently I can't connect to CANSee anymore.  CanZE doesn't allow to change to Bob Due.  The device type is greyed out.  When the connection between android and CANSee is established (blue light turns on) CANSee freezes.  The green CAN light stops blinking and my torque bar doesn't react.
CANsee works fine w/o CanZE connected.  As soon as I open CanZE, the blue light turns on and then it freezes.  The combination was working ok until 20201129. 
I resently updated to CanZE 1.49.  I tried to connect using KONNWEI - failed, no connection. Changing back zu CanSee not working.
Can I get an apk of older version to confirm the older version.

Regards, Ralph.

yoh-there, 2019-12-03:
Whoa, that sounds serious. We have not seen this yet (luckily!). Thing is, nothing has changed in the handling of CanSee, so it's confusing. Did you come from a (much) older version?

Before digging deeper and getting lost in a rabbit hole, may I suggest

- to ensure you have the latest CanSee firmware loaded
- to erase the settings data from CanZE (Android settings -> Apps -> CanZE -> Clear cache, clear data.

fesch, 2019-12-03:
I would suggest to remove CanZE, then re-install it.

rarisoft, 2019-12-03:
unfortunately, I have not logged the previous CanZE version I used.
for CanSee, as you know. I use an old one, a little updated to use the LED strip. Has there anything changed (last mod in github was 10 month ago).  If you say, there is a new CanSee mathing the newes CanZE, then I update CanSee.

But for now, I am happy to try out, going back, and see when CanZE works.  Just let me have the last 3 CanZE apks.

Cache clearing didn't help, don that anyway. I also completely uninstalled CanZE and re-installed it over the play store - still the same.
Regards, Ralph.

rarisoft, 2019-12-03:
... What worries me, I even couldn't get my KONNWEI to work with the 1.49

yoh-there, 2019-12-03:
I agree with the latter (KONNWEI), which suggest a BT problem with the phone (I am not trying to deflect from CanZE, just thinking out loud). There are too many people using it this way, it would have surfaced massively; there have been roughly 2000 updates to 1.49 already. Is there any chance to try with another phone?

Are you normally always following the update schedule? If so, I can try to get you a recent APK indeed. If you haven't been updating, when (roughly) was the last time you did?

rarisoft, 2019-12-03:
just to let you know, it was all working until Saturday, then I updated....

R240Pe, 2019-12-03:
CanZE worked very very well for 2 years till yesterday. Thousands of Screen-Copies with behavior of the ZOE. I did not change phone.
I accidently updated to Version 1.49 Build 2019.11.30@12:36
Now starting CanZE it starts "flickering" as it seems to be opend twice - the bluetooth-symbol is turning blue but the signs für connecting are not running. The blue BT-Led on ELM is lighted. 
Going to SETUP it turns off the BT-sign but suddenly starts BT and tells me the "ELM is ready".
Going back to Information screens BT is stopped. Tapping the BT-sign seems not to start BT. 
But suddenly BT starts and  "ELM is ready" appears and later again it stops. Found no workaround to get it work normal. 
Is there any chance to get old versions back as soon as possible?

rarisoft, 2019-12-03:
Thanks @R240Pe.

R240Pe, 2019-12-03:
Searching on my PC I found this link: https://apkpure.com/de/canze/lu.fisch.canze/versions
Deactivating SmartScreen-Filter I downloaded CanZE_v1.42_apkpure.com.apk - must see, how to install on phone.

rarisoft, 2019-12-03:
I found this apk too, but it is larger than 1.49 - can we trust this?

rarisoft, 2019-12-03:
Sorry, I found 1.44 of apkpure, 2.4M, current version 1.49 is only 1.6M

R240Pe, 2019-12-03:
Downloading
CanZE_v1.42_apkpure.com.apk is marked as unsafe 
CanZE_v1.41_apkpure.com.apk and CanZE_v1.44_apkpure.com.apk there is no warning

rarisoft, 2019-12-03:
I installed 1.44 of the apkpure link above, will try it out in 45 minutes...  I'll report back just after 1500h ( in 3h from now).  What I can say, I can select CanSee in the device type.  Later more.......

rarisoft, 2019-12-03:
1.44 connectsnok, no freeze. logging ok.
voltage map, heatmap: no data.

[image]

R240Pe, 2019-12-03:
1.44 has alredy the new colour-scheme. I did not use this version yet - so my last version was obviously 1.42.
Was ist necessary to uninstall CanZE 1.49 befor installing the elder vesion?

rarisoft, 2019-12-03:
well, I did, just in case... first uninstall 1.49, re-install "older" version.  It worked for me.
@Bob, @Jeroen - what is the verion before 1.49?  Ist there anything in between 1.44 and 1.49 (I think so, I remember a "Test" tile in the experimental screen :-) ).  Regards, Ralph.

rarisoft, 2019-12-03:
I see in the release logs, there was a 1.46 - how can I get hold of this version?

fesch, 2019-12-03:
Between 1.44 there were
1.45
1.46
1.47
1.48

;-)

yoh-there, 2019-12-03:
Gentlemen,

Looks like an Android version related problem. I tested the **** out of 1.49 on my J6 running Android 9 with CanSee and KONNWEI and it runs like a charm. I think I have 7 and 8 devices at home, will try that (but I won't be home this afternoon and evening).

Please do report your Android versions, to confirm (or deny) that this is the issue. Thanks.

rarisoft, 2019-12-03:
ok, possible but not very likely.  I run Android 9 on a OnePlus 5T (Oxygen OS 9.0.9). 
                           v1.49    v1.46    v1.44
CanZE+CanSee     nok         ?        ok

@Bob, thanks for the math lesson  :-)  Can you send me the v1.44 < apk_version < 1.49 please.  
You should have my private email from former issues with CanSee.  I'm happy to help.

Regards, Ralph.

R240Pe, 2019-12-03:
Is there any reason why version 1.42 is marked as unsafe while downloading on  WIN PC, 
is it ok to try installation of version 1.42? Is downloading from apkpure safe?
I run Android 6.0.1 - cannot change on this phone yet.

yoh-there, 2019-12-03:
Here's the issue though. Installed the production version (1.49 release master 2019-11-30 12:36), and it works fine on that J6. You have to trust me on that one. KONNWEI and CanSee.

If both fail on your devices, it's something that has a relation to your phone. I am not saying we can't address it in CanZE (obviously we can, as it works in the older versions), but it's kind of a weird problem to tackle.

I'll see if I can compile an 1.48

fesch, 2019-12-03:
https://www.fisch.lu/junk/V1.45.apk
https://www.fisch.lu/junk/V1.46.apk

yoh-there, 2019-12-03:
https://github.com/fesch/CanZE/releases/tag/V1.48

Try this too (ignore source code!)

rarisoft, 2019-12-03:
So, here my update
```
              |  v1.49  |  v1.48  |  v1.46  |  v1.45  |  v1.44  |
--------------|---------|---------|---------|---------|---------|
CanZE+CanSee  |  freeze |  freeze | no data |   ok    |   ok    |
```
however, no version returns the voltage nor the temperature heatmap.

R240Pe, 2019-12-03:
If there are problems in these versions - can we get a save copy of the CanZE Version 1.42  here?
V 1.42 had no such problems (on my phone).

rarisoft, 2019-12-03:
@R240Pe is working v1.45 with your setup?

R240Pe, 2019-12-03:
I uninstalled CanZE 1.49. After downlod V 1.45 from here I tried out. From normal start I don´t get bluetooth connection. I go to SETUP an back to MAIN or TECHNICAL-page and tap on bluetooth-symbol.
Then starts bluetooth Connection. I tried several times - once it doesn´t connect, i closed CanZE and started new. 
The Cell-voltages and cell-temperatures are shown (no NaN) at both.
Now I cannot switch to bright Mode - it always stays in dark mode. This is difficult to read (while driving).

yoh-there, 2019-12-03:
@R240Pe Dark mode is selected when either
- dark mode is selected in the settings screen
- Android is version 9 or 10 AND night/dark mode is selected
The idea is that CanZE follows your general settings, but for phones that do not support a dark mode (Android 8 and lower), it can still be selected.

If this is not what you experience, please make another issue with a full description (please do state Android version, brand, night mode setting if >=9, or an installed custom theme, and your setting screen, so I can configure an emulator accordingly. Let's keep this issue focused on Bluetooth, very easy to loose track already.

yoh-there, 2019-12-03:
I checked source changes between 1.45 and 1.46. There were a lot of changes to support ISOTP commands, but none to the BT on CanSee interface. The only change remotely relevant was the shortening of a timeout in the ELM327 (KONNWEI) driver.

rarisoft, 2019-12-03:
I don't think it's the BT stack, I simply think it gets overloaded and causes CanSee to hang up..  The Android app still reacts some how.  I had debug logging on but since the version is not included into the log, I lost track, what belongs to what - sorry, I changed around the versions to often and didn't write down the time when which was active.
I am away the next two days, back Friday.  I will do some logging then and keep a record which file belongs to which CanZE version.
If you let me know, what exactly I shall switch on/off to make it all meaningful, let me know.  Since we tracked the change 45/46, it should be possible to crush that bug,

gn8, Ralph.

yoh-there, 2019-12-03:
OK, Friday is a good plan. What you suggest though does not jive with KONNWEI not working either. It's very mysterious; I drove >100 km today with the KONNWEI in and CanZE re-loaded from the play store (production 1.49);, and it worked flawlessly.......

rarisoft, 2019-12-03:
agree, maybe there is another problem with the KONNWEI, since I was not successful with torque+KONNWEI either.  Those might be two issues - probably linked somehow. Maybe I didn't try hard enough and had a wrong setting for the KONNWEI somewhere, I was concentrating on the CanSee.
Let's focus on CanSee where we know, the version change of CanZE switches the problem on/off.  Let me know, what to log and which files shall be uploaded.
Bye, Ralph.

yoh-there, 2019-12-04:
I have thought about this issue a bit more and it doesn't make sense at all to me to be honest. For this moment, my assumption is there is (in @rarisoft  case) not a BT problem. This is a tricky assumption, as there were quite a few changes in the BT code, but that was only started after the release of 1.48.

Also, an assumption is (for now) that a KONNWEI does work. If it does, by inference, it is NOT an BT problem.

Of course both assumptions could be wrong, but given the discussions, it's a non unreasonable starting point.

I have checked the source file of CanZE, CanSee driver part. There has not been made any relevant change since March of this year, and that was released in 1.36 (4th of April). So that leaves CanSee itself.

My suggestion is to hookup a terminal to the USB connector, set to 115200 bps and reboot the ESP32. It should report the version and I am curious. Should be 002, though I fully admit I have been quite sloppy with upping it. I am perfectly happy to drop the latest compiled binary which I have reloaded just a few days ago (as I needed that ESP32 for something else for a day) and it works just fine. Let me know what you need for that please.

rarisoft, 2019-12-04:
I am quite familiar with arduino and wemos.  Yes, I have my own code of CanSee since I modified it to get a led bar, see https://canze.fisch.lu/page/3/
When I'm back home, I can try to enable debugging in the CanSee and log what comes from CanZE before it freezes.  And I can use the original code to verify its not my alteration.
is CanSee using dynamic memory allocation?

yoh-there, 2019-12-04:
I strongly suggest (but I know you know this ;-) ) to start with known-running-code as I am running it every day, if only just to verify where things diverge. Preferably even the binary, which I will make available for you. Then to it's exact source and include your tool chain, only then to your modifications.

I am sure you appreciate it is extremely hard for @fesh and me to support fixing this nasty crashing when we're actually not running the same code.

I can't make promises about the libraries, but everything we programmed is completely static memory wise. ESP32 is big enough to store everything in global arrays, and while we should never exclude stupidity / silliness, I am reasonably sure we do decent bound checks when accessing those arrays, but I admit I am going from memory here. Memory allocation on the stack is pretty marginal.

rarisoft, 2019-12-04:
no speculation, I will do some logging, then we'll see.  
My CanSee dongle is running fine with pre 1.45 CanZE and even without ;-) to drive the led strip. 
 
off topic: I currently map the torque to the strip, so one can see if its running or not. I can map other signals, whatever one likes.

R240Pe, 2019-12-05:
Workaround that BT works on my phone with here downloaded V1.45
ELM dongle is switched on - start CanZE - the mobile phone displays "existing connection", BT symbol on the mobile phone is blue with "))" behind it.
When you start CanZE on the phone, the blue LED on the dongle lights up - the yellow data LED remains dark. There is no data transfer even after minutes.
Next, go to "Settings" and a little 30 sec. waiting. Then leave again "settings" - and usually then builds up either the same or with some patience the connection. Takes and is awkward - but at least I can get to the data somehow.
Have a complete documentation with about 2000 lines with 50 parameters of data while driving and loading my Q90 for 2 years - want to continue  my documentation.

yoh-there, 2019-12-05:
Thank you for your persistence and workaround-description, it might give hints to a solution. Also I am curious if the newest beta is better.

BTW, when one if my kids start a sentence with "I want", my answer is always "I want a Ferrari" ;-)

@rarisoft can you deny/confirm on your end?

R240Pe, 2019-12-05:
OT: Ferrari is good, TESLA would be better ... 
Spent so many work and time for documentation how accu  (RENAULT software) works -  so it´s really more than only "I would like to ..."

yoh-there, 2019-12-05:
OT: In that case I might ask for a counter-favor. I am not convinced the modelling for the ZE40 battery is very good. if you can come up with a mathematical correlation between on one side the maximum power (DC) the battery will charge with, and on the other side the SOC and the battery compartment temperature, that would be great. A second order polynomial would be more than enough in terms of precision.

now:
maxDcPower = cons + (tempx * intTemperature) + (tempxsocy * stateOfChargePercentage * intTemperature) + (socy * stateOfChargePercentage);

where for a ZE40: cons=14.93,  tempx=1.101, socy=-0.145, tempxsocy=-0.00824);

rarisoft, 2019-12-08:
here some logs from within CanSee dongle via serial.
older versions seem to run more stable, 1.45 seems best.  1.48 has a lot of NANs
I also have debug, field and EVC logs from CanZE of the various versions (all in the zip).

[ZOE_logs.zip](https://github.com/fesch/CanZE/files/3936128/ZOE_logs.zip)

Hope this helps. let me know if any special procedures are needed.  In all logs, I first tried to access the charging history, braking, Tires, sometimes" ECV all data".

Let me know, Ralph

rarisoft, 2019-12-08:
btw, no log managed to get voltage heatmap nor temperature heatmap, not the TPMS.  TPMS was working yesterday when I was driving around (v1.45)

yoh-there, 2019-12-08:
Thank you.

@fesch I haven't checked all logs but in 1.48 the scheduler seems to get stuck in a loop 1f8.28 (ElecBrakeWheelsTorqueApplied). That thing is only used in the datalogger????

I also see a request to 7bc.624b7e which I can find nowhere in the source code, and nowhere in the cansee log (though the latter may be an overlow in the logging). This makes no sense to me at all.......

fesch, 2019-12-08:
Having the MainActivity being a singleton (or at least make as if it where one), all this might be related to the double-MainActivity-Startup-Bug that came with forcing dark mode. I've made some changes. On my dash it looks now better ...

rarisoft, 2019-12-08:
the very first logs might have ELM, so thus is not working, I use CanSee, this was for testing only.
I just drove a longer distance again with 1.45 and after a while, I got TPMS and even Heatmaps..I only can provide CanZE logging, I didn't take the laptop.
Tomorrow, I'll get the car back (winter tires will get fitted) and I log with v1.45

rarisoft, 2019-12-08:
...reading this, yes, I put it in dark mode where available

yoh-there, 2019-12-08:
@rarisoft @R240Pe I am rolling out 1.50beta4 which includes the changes Bob speaks of as we speak. If not already done, please subscribeto the Open Beta in the play store and you will get it in a few hours at most (if google agrees!)

rarisoft, 2019-12-08:
done

yoh-there, 2019-12-08:
Great. One of my devices already updated, so it is "out there"

rarisoft, 2019-12-08:
I can't choose CanSee in CanZE 1.50 b4. :-(
[image]

yoh-there, 2019-12-08:
Please switch back and forth in the Device type between your KONNWEI and CANsee. Type should adjust automatically if you select CANsee.

Also, please switch off all three log ticks before moving on with the connection issue. Fields I use often myself, so that could be switched on once A-OK. Yes, I know it should not interfere, but I m trying to establish a base line here. Thanks again!

yoh-there, 2019-12-08:
Nil my first sentence. Seems you (or me) changed the BT name of the device from CANSee to CANsee. Only CANSee makes CanZE think it is a CANS/see device.

See config.cpp line 35 for default. I will make the comparison case-unaware, but for now please rename your bluetooth name.

Remainder of previous post stands.

rarisoft, 2019-12-08:
with the change, quite possible.  I even think to attach that function to the nane is not good practice. A functional handshake would be better - question, answer .. that sort of thing.
For the time beeing, what exactly is CanZE expecting? Than I change it to that name.
Currently I'm not connected. "Device type" is greyed out as soon as I enter the settings.
Regards,  Ralph.

yoh-there, 2019-12-08:
CANSee (case sensitive)

rarisoft, 2019-12-08:
changing the name in the android settings is enough.  CanZE changes the type as described above...

yoh-there, 2019-12-09:
Yes, we try to get what is called the Alias. If the pure name is used, it's mighty confusing as dongles are usually named OBDII or KONNWEI only. In my case they are names like "old konnwei, new konnwei, blue cheap" etctera. Under the hood the mac address is used.

While I agree auto-type-detect would be nice, I am not going to implement that. 99% of the users run an ELM based dongle. I don't think it is a huge problem to insist a cansee dongle is named cansee. I did make some post beta4 changes though, it's far more consistent now. Type is now fully automated, http url decently and separately saved, etc.

Let us know how things are now. If there is still failure, please retry with your KONNWEI, and also with all logging off.

rarisoft, 2019-12-09:
I connected successfully with the  1.50 b4 earlier using CANSee.  No laptop connected.  Seems to work :-) 👍 
TPMS read out ok, charge history ok.  I didn't check while driving but brake data look plausible.
Well done.

yoh-there, 2019-12-09:
Good to hear and thank you for the testing effort, yay. Now if the other issue which also involved dark mode is reported solved too I think we can work towards a production release.

R240Pe, 2019-12-11:
V 1.50beta4 Build 2019.12.08 @ 16:57:
Still problems with BT. Starting CanZE  the BT-symbol stays grey - nothing happens even in I wait several minutes.
Going to Settings and back BT symbol is turning blue and "))" appears.
Morning - at the ZOE - I will see wether connection establishes.

[3 more replies]

closed 2019-12-11 by yoh-there