fesch/CanZE #590: Disable dark mode in CanZE does not work

R240Pe, 2019-12-04:
After updating to CanZE V 1.49 bluetooth connecting  ongle does not work properly. Activating and deaktivating dark mode was no problem. Made some screen copies in normal mode. 
Due to BT problems I deinstalled CanZE V 1.49 and installed from github manually downloaded version V 1.45.  BT problems are the same.
But this time CanZE started in dark mode. Changing button "force dark mode" in SETUP-settings does not do anything - CanZE remains in dark mode. I restarted app and also phone - CanZE stays in dark mode. Leaving SETUP-menu with activated or deativated dark-mode-button does not change colors - it stays in dark mode.

My Phone: SONY Experia Z3 (D6603)
Android: V6.0.1 - no dark mode settings in this version.
Theme is SONY standard - I never changed it. 
I did not change any phone-settings between updating CanZE to V1.49, deinstallation and manually installing V 1.45 CanZE-apk-file.

fesch, 2019-12-04:
I remember that one of the versions had a bug in the "force dark mode", which is why we needed to push the next release quite soon after that. I will do more tests ...

yoh-there, 2019-12-04:
@fesch 1.44 was immediately (within a day) superseded by 1.45. The only difference was that 1.45 ditched Android 4 as some renderings are not supported by Android 4. If you want have nothing to do with dark mode at all, you need 1.43.

I just loaded 1.45 in a stock Android 6 emulator (Nexus 6 on API 23). It started in light mode. I switched  back and forth a few times using Settings and it worked as expected. I have no idea why this would not be the case on the OP's phone.

R240Pe, 2019-12-05:
Afternoon at 16:10 looking on my phone, CanZE was suddenly displayed in "bright" mode. I did no changes on the phone. Now 17:16 it is in dark mode again - without any changes on the phone. Disabling "Force dark mode" does not change the display - as I already reported.
Does it perhaps use settings from play store? Playstore shows also in dark mode now. I did not control this seriously yet, but play store in dark mode happend here already serval times (on my Android 6 phone). 
Is there an automatically change to "dark Mode" by play store or by time of day? ... in CanZE ? 
As written above V1.43 should be last CanZE version without handling dark mode.  For testing I could not find V 1.43 anywhere for downloading. Is there a chance to get this?

yoh-there, 2019-12-05:
This must be a setting/theme on your phone, if even "build in". I bet tomorrow in the day it will be light. CanZE nor the play store does anything like that.

I can compile a 1.43 version for you, but I am not entirely happy about that solution. You'd miss out on tons of bug fixes. And the dongle handling is a lot better now.

Maybe I can think of a "force light mode". Let me think about that for a bit.

yoh-there, 2019-12-05:
OK, I made that three way setting. Please
- remove CanZE
- install CanZE from the play store
- Make yourself a beta tester in the play store.

I will upload this change in a couple of minutes and usually that is rolled out to users in the Open Beta in a couple of hours.

R240Pe, 2019-12-05:
Was never a beta tester - will try.
As I mentioned here [https://github.com/fesch/CanZE/issues/589](url) for my use (all accu data) it worked very well. But printouts of dark screencopies are very worse and "dark" driving is dangerous. 
Of course i will try further newer version.
BTW: 
While driving the car, the mileage is not updated in CanZE, so the ZOE must first fall asleep or start charging. After waking up, the mileage is correct again. Could not find anything about this.

yoh-there, 2019-12-05:
Thanks

You might want to consider experimenting with the new "Research" screen for your use case, see https://canze.fisch.lu/screen-for-researchers/

Dark mode is IMHO not dangerous at all, in fact especially at night it is much safer as it doesn't overwhelm eye perception. But let's agree to disagree: if the new setting works, you're a happy man.

What you are referring is the battery odometer. The car transfers it's own odometer to the battery computer regularly. It's not intended as the car's odometer, but indeed, it is never  off for more than one charge, mostly less. It has nothing to do with CanZE, it's is simply the field I choose to display as it's all _battery_ parameters.

R240Pe, 2019-12-06:
Thanks for information.
I have to drive 2000 km with a lorry - nobody has license. Will be back home on tuesday.
Sorry, CanZE must wait.

yoh-there, 2019-12-10:
I realize after 2000 km lorry driving, wheels are the last thing on your mind, but I would appreciate a quick spin of the latest beta4. If it doesn't run, you might try the new dark mode "Light".

If it works, I will push out a release candidate beta tonight.

R240Pe, 2019-12-11:
I´m back home again. Installed V 1.50beta4 Build 2019.12.08 @ 16:57.
Great - I am back in light again - tried several times using all modes - works like selected.
"System" turns on dark mode - although Android 6 doesn´t have dark mode.
Selecting Light turns off dark mode again.
Everything works fine.

yoh-there, 2019-12-11:
Great. Thanks for reporting.

closed 2019-12-11 by yoh-there