fesch/CanZE #591: Charging history, is column "energy taken from the grid" possible?

DompaEV, 2019-12-04:
I was about to ask if it would be possible to have column with number of kWh taken from grid in Charging history screen per each charge, when I came over already closed issue #519 which is about displaying total energy taken from the grid over some longer period (e.g month or year).
You mentined in comments there that potentialy usefull variable 7ec.6233db.24 "varies wildly".
If you can remember from your investigation back then, was that variable/field reliable during one single charge session or it was really wild even then?

If it is stable during one charge session, then it might be usefull to show it in that charging table history. Of course, CanZe would need to be active at least during begining of charge and at its end.

Another way would be to take samples of "Mains active power consumed" variable (shown in Charging screen) regularly (e.g. every minute) and then calculate Wh out of it and add it to temporary variable that will hold total kWh vaule for that single charge. 
For more accurate calculation, rectangle method of integration can be used:
X = (current - previous) / 60
Tot1charge = Tot1charge + X
This would require CanZe to be active during whole charge session.
Thank you.

yoh-there, 2019-12-05:
The history is taken from the car; we don't keep any energy memory. As the car does not store grid energy in the history, we cannot display it.

I have never managed to get a reliable grid power or energy figure, but feel free to research using the All Data or the Research screen (see https://canze.fisch.lu/screen-for-researchers/ ). Best candidates are EVC and BCB.

Calculating power from leads current and voltage is not very useful as indeed CanZE should remain open and in that specific screen, and the drawing substantial phase-shifted current due to the large filter capt (about 6 amps for the Q models). It would be too unreliable and user-unfriendly.

DompaEV, 2019-12-06:
Thank you for response and guidance.
I will try to dig there when I find some spare time.

yoh-there, 2019-12-06:
Please do. I will close this issue, but if you find something that works, just reopen please.

closed 2019-12-06 by yoh-there

DompaEV, 2019-12-06:
OK, understood :-)