fesch/CanZE #592: I found something very cool

FredLeudon, 2019-12-06:
Hi all.

When searching a more usable field for speed cam activity (cluster odometer is given only km per km) I found the request 76D.2121 who seems to be a large part of free frames we can read on can bus. 

For speed cam activity, il will use this request only as : 

7d6.2121.12.3 -> speed 
7d6.2121.4.3 -> trip distance

yoh-there, 2019-12-06:
In Android we use

```
,7ec,24,55,.0001,0,3,km,2233DE,6233DE,1ff,This variable allows to memorize the reseted distance
```
In theory this is on 10cm precise, hehe, but I doubt it. I never checked it but that should be easy.

Seems like you are taking the data from the USM. I extracted these fields for the 2121 frame (see below). You are using 
```
,76d,26,53,.01,0,2,km,2121,6121,ff,DistanceTotalizer
,76d,91,106,.01,0,2,km/h,2121,6121,ff,VehicleSpeed
```
Would it really be 10 meter / 0.01 km/h?

```
,76d,16,17,1,0,0,,2121,6121,ff,StartingMode_BCM_R,0:Engine stopped request;1:No request;2:Ignition request;3:Starting request
,76d,18,19,1,0,0,,2121,6121,ff,BCM_WakeUpSleepCommand,0:Go to sleep;1:reserved;2:reserved2;3:Wake Up
,76d,20,22,1,0,0,,2121,6121,ff,FrontWipingRequest,0:In Stop Position;1:Forced to Stop;2:One time low speed requested;3:One time high speed requested;4:Unavailable;5:Reserved;6:Low Speed requested;7:High Speed requested
,76d,23,23,1,0,0,,2121,6121,ff,HeadLampsWasherRequest,0:Headlamps washer not requested;1:headlamps washer requested
,76d,24,25,1,0,0,,2121,6121,ff,ClutchSwitchMaximumTravel,0:Not used;1:Maximum clutch travel switch not active;2:Maximum clutch travel switch active;3:Unavailable
,76d,26,53,.01,0,2,km,2121,6121,ff,DistanceTotalizer
,76d,54,77,1,0,0,mn,2121,6121,ff,AbsoluteTimeSince1rstIgnition
,76d,78,78,1,0,0,,2121,6121,ff,WakeUpType,0:Customer wake up;1:Technical wake up
,76d,79,79,1,0,0,,2121,6121,ff,PositionLightsRequest,0:Position Lights not requested;1:Position Lights requested
,76d,80,80,1,0,0,,2121,6121,ff,LowBeamRequest,0:Low beam not requested;1:Low Beam requested
,76d,81,81,1,0,0,,2121,6121,ff,HighBeamRequest,0:High Beam not requested;1:High Beam requested
,76d,82,82,1,0,0,,2121,6121,ff,FrontFogLightsRequest,0:Front Fog not requested;1:Front fog requested
,76d,83,90,1,40,0,°C,2121,6121,ff,EngineCoolantTemp
,76d,91,106,.01,0,2,km/h,2121,6121,ff,VehicleSpeed
,76d,107,108,1,0,0,,2121,6121,ff,EngineFanSpeedRequest,0:Not requested;1:Low Speed requested;2:Middle Speed requested [Nissan];3:High Speed requested
,76d,109,109,1,0,0,,2121,6121,ff,ACCompAuthorized,0:Ac compressor not authorized;1:Ac compressor authorized
,76d,110,110,1,0,0,,2121,6121,ff,ACCompClutchRequest,0:Ac compressor not requested;1:Ac compressor requested
,76d,111,117,.01,0,2,A,2121,6121,ff,ACCompValveRequest
,76d,118,118,1,0,0,,2121,6121,ff,ACCompWorkingMode,0:Running in idle mode or stopped;1:Running not in idle mode
,76d,119,120,1,0,0,,2121,6121,ff,EngineStatus_R,0:Stopped;1:Reserved;2:Running;3:Cranking
,76d,121,128,40,0,0,W/s,2121,6121,ff,AllowedPowerGradient
,76d,129,130,1,0,0,,2121,6121,ff,CrankingAuthorisation_ECM,0:Cranking not allowed;1:Cranking allowed;2:Cranking stopped required;3:No request
,76d,131,132,1,0,0,,2121,6121,ff,CrankingAuthorisation_AT,0:Cranking not allowed;1:Cranking allowed;2:Reserved;3:No request
,76d,133,140,40,0,0,watts,2121,6121,ff,AlternatorMaximumPower
,76d,141,156,.125,0,1,tr/min,2121,6121,ff,EngineRPM
,76d,157,164,1,40,0,°C,2121,6121,ff,ExternalTemp
,76d,165,165,1,0,0,,2121,6121,ff,DeliveryModeInfo,0:customer mode;1:delivery mode
,76d,166,166,1,0,0,,2121,6121,ff,DayRunningLightRequest,0:No DRL requested;1:DRL  requested
,76d,167,167,1,0,0,,2121,6121,ff,DiagMuxOn_BCM,0:counting and storage of CAN mute/absent failures  masked;1:counting and storage of CAN mute/absent failures not masked
,76d,168,169,1,0,0,,2121,6121,ff,ClutchSwitchMinimumTravel,0:Not used;1:Minimum clutch travel switch not active;2:Minimum clutch travel switch active;3:Unavailable
,76d,170,170,1,0,0,,2121,6121,ff,FrontDefrostRequest,0:Front defrost not requested;1:Front defrost requested
,76d,171,172,1,0,0,,2121,6121,ff,FloatingInhibition,0:Unavailable;1:Floating forbidden;2:Floating not forbidden;3:Not used
,76d,173,173,1,0,0,,2121,6121,ff,ACCompClutchActivation,0:no compressor activation req;1:compressor activation req
,76d,174,176,1,0,0,,2121,6121,ff,VehicleState,0:VEHICLEASLEEP_ENGINESTOPPED;1:VEHICLEAWAKE_ENGINESTOPPED;2:ENGINE_READY_TO_START;3:STARTING_IN_PROGRESS;4:VEHICLEAWAKE_ENGINERUNNING
,76d,177,178,1,0,0,,2121,6121,ff,VehicleStateSSExtension,0:VEHICLE_NOT_IN_AUTOMATIC_STATE;1:VEHICLE_IN_AUTOMATIC_STATE;2:VEHICLESTATE_SS_EEXTENSION_NOT_USED;3:VEHICLE_EXTENSION_UNAVAILABLE
,76d,179,180,1,0,0,,2121,6121,ff,GenericApplicativeDiagEnable,0:Only faults reports and alerts that MUST remain active during cranking or sleeping periods are enabled;1:All faults reports and alerts not related to APC power level are enabled;2:All faults reports and alerts are enabled;3:Unavailable
,76d,181,183,1,0,0,,2121,6121,ff,BrakeInfoStatus,0:;1:PEDAL_NOT_PRESSED;2:PEDAL_PRESSED;4:PEDAL_CONFIRMED_PRESSED;7:UNAVAILABLE
,76d,184,184,1,0,0,,2121,6121,ff,DriverDoorState,0:DRIVER_DOOR_CLOSE;1:DRIVER_DOOR_OPEN
,76d,185,186,1,0,0,,2121,6121,ff,EngineHoodState,0:ENGINEHOOD_UNAVAILABLE;1:ENGINEHOOD_OPENED;2:ENGINEHOOD_CLOSED;3:ENGINEHOOD_NOT_USED
,76d,187,191,4,0,0,cm,2121,6121,ff,AccurateOdometer
,76d,192,193,1,0,0,,2121,6121,ff,AC_StopAutoForbidden,0:UNAVAILABLE_REQUEST_CLIM;1:STOP_AUTO_FORBIDDEN_CLIM;2:STOP_AUTO_AUTHORIZED_CLIM;3:CLIM_REQUEST_NOT_USED
,76d,194,197,.25,-46,1,V,2121,6121,ff,MinimumVoltagebyAC
,76d,198,199,1,0,0,,2121,6121,ff,PumpActivationRequest,0:UNAVAILABLE_VALUE;1:NO_PUMP_ACTIVATION_REQUESTED;2:PUMP_ACTIVATION_REQUESTED;3:PUMP_NOT_USED
,76d,200,203,.25,-46,1,V,2121,6121,ff,MinimumVoltagebySteer
[... cut ...]
,76d,235,236,1,0,0,,2121,6121,ff,BatteryChangedActivation,0:Not Used;1:NO battery parameter reset request;2:Battery parameter reset requested;3:Unvailable Value
,76d,237,237,1,0,0,,2121,6121,ff,StartAutoSystemFailure,0:NO_START_AUTO_SYSTEM_FAILURE;1:START_AUTO_SYSTEM_FAILURE
,76d,238,241,.25,-46,1,V,2121,6121,ff,MinimumVoltagebyBCM
,76d,242,242,1,0,0,,2121,6121,ff,CrashDetected,0:No crash detected;1:Crash Detected
,76d,243,243,1,0,0,,2121,6121,ff,CrashDetectionOutOfOrder,0:Crash Detection OK;1:Crash Detection Out Of Order

```

yoh-there, 2019-12-09:
Conclusion after some testing:
7ec.6233de.24 has 10 meter resolution
76d.6121.26 has 10 meter resolution. But indeed nice it doesn't roll over
76d.6121.91 has 0.01 km/h resolution.

Not bad!

fesch, 2019-12-10:
If 76d.6121.26does not roll over, this would be a good candidate for the speed-cam activity ;)

yoh-there, 2019-12-10:
Yes, and for the Driving screen (it is used for trip distance consumption). But as the 76d is the fuse box, it seems a rather "indirect" approach. For speed and distance there are also a few good candidates in the EVC. I will experiment with these too and post here.

yoh-there, 2020-12-03:
We use 7ec.6233de.24 for Driving (to calculate  trip distance and average consumption). It's also used in speedcontrol.

Closing this issue, thank you for the tip!

closed 2020-12-03 by yoh-there