fesch/CanZE #593: Bluetooth connection issues

yoh-there, 2019-12-11:
Reported by @R240pe
V 1.50beta4 Build 2019.12.08 @ 16:57:
Still problems with BT. Starting CanZE the BT-symbol stays grey - nothing happens even in I wait several minutes.
Going to Settings and back BT symbol is turning blue and "))" appears.
Morning - at the ZOE - I will see wether connection establishes.
For Information: Everytime I installed another CanZE version - the Type of ZOE is resetted to Q210 

confirmed @rarisoft
I confirm the BT behaviour by @R240Pe. When BT is lost, one has to go to "settings" to retrigger the connection. But I consider this as a different issue. Does it have anything to do with background execution?

Extra info so far
R240pe uses Android 6, rarisoft Android 9
I am honestly baffled. I tested the disconnect scenario many times in the last few weeks. My car is on the driveway, my office at the other end of the house. Did a lot of walking back and forth. When connected: stable blue icon with arcs. Walking away (loosing connection): arcs stat animating (connection lost) or icon grey (bluetooth lost). Walk back, taking roughly 5 seconds to go from animation to connnected. Queue handling then also restarts (debug line shows that). Tested with KONNWEI and CANSee.

Initial connection also no issue. Sometimes it takes a few seconds to go from grey to animated, but always goes to connected in seconds.

Questions, especially to establish the same testing scenario
- do you test with walking away, or switching off the dongle? (I did not test the latter scenario)
- do switch on the dongle after starting CanZE? (mine is always on)

Calling @fesch as I am stumped.

yoh-there, 2019-12-11:
I might have an idea. It might have to do with an improper MAC address. Let me do a new beta. Once it arrives, please change the device to http gateway and back, then save and retest.

fesch, 2019-12-11:
I do not had any of these issues with the latest dev branche... Sent from my Huawei phone-------- Original message --------From: yoh-there <notifications@github.com>Date: Wed, Dec 11, 2019, 11:02To: fesch/CanZE <CanZE@noreply.github.com>Cc: Bob Fisch <bob@fisch.lu>, Mention <mention@noreply.github.com>Subject: [fesch/CanZE] Bluetooth connection issues (#593)Reported by @R240Pe
V 1.50beta4 Build 2019.12.08 @ 16:57:
Still problems with BT. Starting CanZE the BT-symbol stays grey - nothing happens even in I wait several minutes.
Going to Settings and back BT symbol is turning blue and "))" appears.
Morning - at the ZOE - I will see wether connection establishes.
For Information: Everytime I installed another CanZE version - the Type of ZOE is resetted to Q210
confirmed @rarisoft
I confirm the BT behaviour by @R240Pe. When BT is lost, one has to go to "settings" to retrigger the connection. But I consider this as a different issue. Does it have anything to do with background execution?
Extra info so far
R240pe uses Android 6, rarisoft Android 9
I am honestly baffled. I tested the disconnect scenario many times in the last few weeks. My car is on the driveway, my office at the other end of the house. Did a lot of walking back and forth. When connected: stable blue icon with arcs. Walking away (loosing connection): arcs stat animating (connection lost) or icon grey (bluetooth lost). Walk back, taking roughly 5 seconds to go from animation to connnected. Queue handling then also restarts (debug line shows that). Tested with KONNWEI and CANSee.
Initial connection also no issue. Sometimes it takes a few seconds to go from grey to animated, but always goes to connected in seconds.
Questions, especially to establish the same testing scenario
do you test with walking away, or switching off the dongle? (I did not test the latter scenario)do switch on the dongle after starting CanZE? (mine is always on)
Calling @fesch as I am stumped.

R240Pe, 2019-12-11:
To avoid work I did the following:
For fundamental testing I disabled BT and shutdown my phone and switched off dongle. I switched on the dongle and restarted my phone and aktivated BT. Then I started CanZE App ... and after a few seconds BT connection establishes - without "Settings"-workaround.
Then I stopped the CanZE app completely - the dongle stay on.
After a few minutes I went back to the car and started CanZE new. After a few seconds connection established again.
Maybe any leftovers from old installations. Now it seems to work perfectly.

yoh-there, 2019-12-11:
That is pretty great news, cause I have and had no idea what was going on.

I did publish the beta5 and will leave this issue open until the weekend. If no further issues reported, I will publish a production release and close this issue.

Thanks again for persistently trying.

R240Pe, 2019-12-12:
I updatet to V 1.50beta4 Build 2019.12.11 @ 11:10
Everything works fine.
But why did you remove the temperature-setting for heater from Temperature screen?
OK PTC-Info was never correct till now but setting-information was necessery to see if and how heater/ptc works.
With  outside 0°C - setting inside 18 °C the heat pump starts and ends within a few seconds - so you can´t see on screen that heating is on.   Nothing to see anywhere on that screen that heating is on ( when setting is higher than inside-temperature, assumed heater is on). While DC consumption on Charging-screen shows -2,5 kW (standing car) you see that PTC (s) is/are on.

yoh-there, 2019-12-12:
Believe it or not, I have not been able to find the cabin temperature setpoint as an ISOTP (diagnostic) command, and boy did I search! IOW, this removal is a victim of the ISOTP fields commands. This was such an improvement in dongle handling, and the dial in the car staring in your face that I think it is a reasonable choice.

The small graph should show the heat pump working?

R240Pe, 2019-12-12:
Yes - it shows - only - when the heat pump is working - but no information that PTCs are on or should be on because of  temperature setting is higher. Heating of ZOE has always been a bit tricky and and any information about that was very useful.

yoh-there, 2019-12-12:
I never saw the 3 PTC relays chance. Did you? If so I can put them back.

yoh-there, 2019-12-12:
Temperature issue to #594 please

rarisoft, 2019-12-12:
I had problems with the connection (v1.50b4 20191208@1657) under the following conditions:
settings: BT background mode not ticked.

when all is working, just 
- use task manager and switch to another running app (or home screen), 
- wait a bit (until 'BT connection lost message' briefly pops up),
- return to CanZE via task manager, (BT icon blinks, no connection can be (re-) established.

I record that from memory, I will confirm the behaviour when I'm back home (latest at the week end...)

rarisoft, 2019-12-14:
i tried this morning; v1.50 (release master) 20191213@12:56 reconnects BT to CANSee after "connection lost" without problems.

closed 2019-12-14 by yoh-there

cad0p, 2021-04-10:
It does not for me on two different phones!
If i leave the app for more than a few seconds, it just blinks continuously.. happy to help solve this!