fesch/CanZE #594: Cabin temperature setpoint not found in ISOTP

yoh-there, 2019-12-12:


yoh-there, 2019-12-12:
Comments from 593

@R240Pe
But why did you remove the temperature-setting for heater from Temperature screen?
OK PTC-Info was never correct till now but setting-information was necessery to see if and how heater/ptc works.
With outside 0°C - setting inside 18 °C the heat pump starts and ends within a few seconds - so you can´t see on screen that heating is on. Nothing to see anywhere on that screen that heating is on ( when setting is higher than inside-temperature, assumed heater is on). While DC consumption on Charging-screen shows -2,5 kW (standing car) you see that PTC (s) is/are on.

@yoh-there 
Believe it or not, I have not been able to find the cabin temperature setpoint as an ISOTP (diagnostic) command, and boy did I search! IOW, this removal is a victim of the ISOTP fields commands. This was such an improvement in dongle handling, and the dial in the car staring in your face that I think it is a reasonable choice.

The small graph should show the heat pump working?
@R240Pe
Yes - it shows - only - when the heat pump is working - but no information that PTCs are on or should be on because of temperature setting is higher. Heating of ZOE has always been a bit tricky and and any information about that was very useful.

@yoh-there 
I never saw the 3 PTC relays chance. Did you? If so I can put them back.

R240Pe, 2019-12-12:
I think they always showed "on", but I´m not sure. Have about 2k Screenshots - must have a look at those  when I´m home tonight.
As I wrote - any information what´s going on with heating would be useful.

yoh-there, 2019-12-12:
@FredLeudon Never mind, I found it

`,764,29,36,1,0,0,,2145,6145,ff,OH_ClimTempDisplay`

8 = min, 10 max, other = setpoint * 2.

@R240Pe This will go to the upcoming release

closed 2019-12-12 by yoh-there

R240Pe, 2019-12-13:
That´s fine. Thank you.
BTW: PTCs were always displayed as "on" in every of my temperature-screenshots 2019 with heater on or off +30°C/-10°C.

yoh-there, 2019-12-13:
Right. Those won't come back, as I don't think those are truly implemented in the / this diagnostic