fesch/CanZE #595: Zoé R110

neuesfernsehen, 2019-12-21:
Which car should I select in the settings when I have a Zoé R110 and want to use CanZe iOS version?

FredLeudon, 2019-12-21:
It is preferable to use the beta version of CanZE.

You can install beta version through Apple’s TestFlight app by following this link : https://testflight.apple.com/join/nhMX6f5n

Be sure to allow CanZE to use cellular network; so CanZE will be able to scan ECUs and download correct definitions form server.

yoh-there, 2019-12-22:
If the question is answered sufficiently, please close the issue, thanks!

FredLeudon, 2019-12-22:
I close this issue. 

Reopen if you can’t install beta version.

closed 2019-12-22 by FredLeudon

neuesfernsehen, 2019-12-23:
Hello,
Thank you very much for the swift reply to my question.
I’ve already installed the beta version but didn’t test it until now.
Closing the issue is fine. I was off over the weekend so I didn’t react in the time you would have preferred.
Best regards
Mark