fesch/CanZE #596: Can't read SOH and some others info

Toug19, 2019-12-21:
Hi,
I just bought a R240 Zoé and I wanted to retrieve the SOH information. But I can't get it. Some others parameters are correctly read as the motor instant power and the current battery charge %. But the SOH still shows 'NaN' and other parameters so.
I'm using an OBD Bluetooth dongle named "MotoDia ELM327".
The CanZE application version is 1.50.
Somebody knows this problem?

yoh-there, 2019-12-21:
This is almost certainly a "crappy dongle issue". Please do get a KONNWEI 902. More info here https://canze.fisch.lu/hardware/

Without wanting to be rude, I am going to close this issue. Many dongles unfortunately are exremely crappy and the generic ones are impossible to assess as their content will change overnight. Sorry.

closed 2019-12-21 by yoh-there

Toug19, 2019-12-21:
This dongle works perfectly with Torque Pro and EVnotify.