fesch/CanZE #599: Settable Startup screen

yoh-there, 2020-01-03:
Goingelectric.de user "Athlon" suggested to use a configurable startup Fragment.

yoh-there, 2020-01-03:
This is actually a pretty good idea (IMHO), but then I would go a bit further and not only allow a startup Fragment, but also an activity. However, the flow should not be disturbed, so if i.e. charging tech is used as startup screen, a "Back" operation should go to the Technical fragment.

In technical terms, this means MainActivity should be started normally always, initial fragment should be determined, also if an activity is choosen.

I programmed this today. There are two issues with this prototype:
- the used classnames are not always compatible with the end-user name (i.e. AVG SPEED <> SpeedControl)
- since MainActivity does some (re)initialisation in onResume(), and then also on a separate thread, this starting of an activity can happen while not everything is properly initialized.

yoh-there, 2020-01-03:
Released on 151beta1

fesch, 2020-01-04:
I would suggest do make an configurable fragement, as this would integrate best into the actual sliding system too and would not disturb anything done in MainActivity.

R240Pe, 2020-01-04:
Since Version 1.50 I use CanZE often when I´m driving because it shows data faster now.
Moving items from the Start-Menu to the second is not a goog idea for me. 
To select Items I have to scoll up an down to activate them - even very difficulty while driving (and was yet not necessary)  - while the start menu is now half empty (Sony Experia). In Samsung Tab 10" there it is no problem - every button is displayed, but it is to large to use in the car.
So I don´t like to install this new version onm my phone  - perhaps you will change to the old menus again. Getting a personalized Start-Menu would also be a good thing.
Removing functions - like discussed on other blogs - nobody is forced to use this functions - so please keep them - all we have in CanZE is useful.

yoh-there, 2020-01-04:
Strictly my _opinion_:
- I agree there is a size unbalance between Main and Technical. However, since we have Crashlytics installed we see, to my surprise I add, that the main use of CanZE **BY FAR**, is charging and charging tech. My beloved Consumption screen is low on the ranking, as is driving. Both are great to control range.
- I understand the annoyance towards change, but everything is far more logical now. Everything that is reasonably used on the main screen, Tech left is almost exclusively charching, Tech right is almost exclusively hardware. Tech in general (except charging tech) is hardly ever used except for a few techy die-hards. Which is exactly what I intended.
- A configurable (4th ?) fragment is an interesting idea. But I am on the fence on putting so much effort in the menu system. Not to be stubborn, but I am not going to code it myself. Too low fun factor and I have my hands full on the ZE50 research.
- Facilitating changing-while-driving is, IMHO, a dubious starting point. However, if that is the goal, fragmenting all activities (sliding them left-right) might be a better option. Brrrr.
- @R240Pe which screen from the technical menu are you activating while driving? It's an honest question as I can't think of a single one that is useful, and if it is, let's think about moving that. I moved climate to Main for exctly this reason. No need to die in complexity.
- As for removing functions, no worries, that is not going to happen I'd say. Weird comment (not by you @R240Pe) anyway as those extra's were moved to Tech and Experimental. Seems to be more a matter of annoyiance towards coders putting effort in things they like instead of what the annoyed like. There is a lot to learn about "getting what you want" in a labor-of-love product, hehe.

Good discussion.

fesch, 2020-01-04:
I've pushed a base concept for a fourth customizable fragment with a maximum of 14 buttons on it.

More to come later -..-.-

R240Pe, 2020-01-04:
My opinion is: screen left - screen right and directly selecting a button is easy and can be done blindly. 
I know: I want the third button on the right - no closer look is necessary - driving is safe.
Otherwise scrolling within a screen and searching for the button needs more attention. 
"ELM-Test" and "All Data" (probably not only) I never used while driving . So they could be moved to a third Screen and all buttons of the "technical screen" would be directly accessible again - for example.
"Customizable" solves all needs of course. Once I configurated it I know blindly again, where my buttons are situated. Do you know the configurable "scan my tesla"-app?
... if you are really bored, you can do the same.
I am also very (more) interested in your "ZE50" activities - I think I will get the R135 in March.

fesch, 2020-01-04:
Do you know a good sponsor for getting me a ZE50 for free? ;-) :-P

fesch, 2020-01-04:
I've just pushed the custom fragment into the dev branch. Everything is implemented. 

@yoh-there : Startup Activity is cool to. Could you extend it and put the custom fragment also into the list?

 @yoh-there : I didn't get the custom fragment into the top menu ... where the hell was that again?

yoh-there, 2020-01-04:
Will do

fesch, 2020-01-04:
- STOP -

I've found it myself ;)

fesch, 2020-01-04:
pushed!

You can test now ;-)

yoh-there, 2020-01-04:
I think activity_settings_costum.xml is missing?

fesch, 2020-01-05:
yes, sorry ...

closed 2020-01-12 by yoh-there