fesch/CanZE #600: Pilot formula and source

landswellsong, 2020-01-05:
Just noticed, the pilot amps reading is way off. Here's an example, please ignore the mess:
[image]
27A is clearly not what 3 phase 43 kW should be (around 64A per phase). 

I'm unsure when did it work correctly so I'm rather asking for debugging hints, apparently this comes from "Free Frames", so do I just need to monitor the bus without doing any actual request? Maybe you have recently changed something that would affect this field?

I also have a custom battery (i.e. custom LBC), but I was fairly sure such a field should come from BCB instead.

yoh-there, 2020-01-05:
Pilot amp has changed when in ISOTP mode, as there we could only get the duty cycle. I will have a look. It either is there, or I made a mistake when switching between ISOTP and free frames. Assuming you are using an ELM, could you please try in both modes?

landswellsong, 2020-01-05:
Yes, using ELM. I'll test tomorrow on 3 stations with known pilot values and report back.

yoh-there, 2020-01-05:
Great, thanks, that helps.

landswellsong, 2020-01-06:
Ok this really seems to be an ISOTP bug.

|  Power (kW) | Pilot (ISOTP) | Pilot (No ISOTP) |
|--------|---------------|------------------|
|  12    |  10.5         | 20               |
| 20     | 15.9          | 30               |
| 40.2   | 26.7          | 59               |

yoh-there, 2020-01-06:
Thank you, I willtry to fix that ASAP.

Bluetooth question I must defer to @fesch , but I am pretty sure that if no other BT using app is active, the connection is supposed to be shut down. Let's decide to open it as a new issue or not after @fesch 's answer.

landswellsong, 2020-01-06:
App in question is Bluetooth terminal. When I stop CanZE (by going to task list button and swiping it out), the terminal starts working normally. Maybe I'm misunderstanding how to Android though :)

yoh-there, 2020-01-06:
Oh I see. AFAIK the dongle can only handle one serial session, it's unable to multitask, nor can Bluetooth "split" the input/output.

landswellsong, 2020-01-07:
Yeah, I get that. But CanZE only started "holding" onto the serial when not active a release or so ago. I assumed that if I home/back from the app it's counted as background?

yoh-there, 2020-01-07:
Pilot: the definition file stated the to get the Duty Cycle was to multiplied by 0.5 and this seems to incorrect. My own charger showed half current too. I also did back-and-forth calculation on your data and assuming this error, the table should be as follows

20 A (13.8 kW) -> real DC 33% -> 10.5A suggests DC 17.5% -> correction * 2 = 35% -> 21 A
30 A (20.7 kW) -> real DC 50% -> 15.9A suggests DC 26.5% -> correction * 2 = 53% -> 32 A
59 A (40.7 kW) -> real DC 86% -> 26.7A suggests DC 44.5% -> correction * 2 = 89% -> 66 A

The "times 2" correction seems good enough for me. I will remove the 0.5 factor in the field definitions in the next release. Commit #4f80388a

landswellsong, 2020-01-07:
89% is actually 62.5A, even closer than 66. I'll compile and test tomorrow, thanks for a quick fix.

fesch, 2020-01-07:
If you press the home button or if you are in the main screen and go back, CanZE is being put into background. That is correct. In that case the BT connection is being given up and other apps should be able to use it too. This is how it _should_ work and worked , but I can recheck on that ...

landswellsong, 2020-01-07:
@fesch thanks. Then it totally doesn't, I can see CanZE's messages about BT reconnection, ELM being ready, etc etc being thrown all over other apps, the setting is off. I tried to set it on to see if it might have been accidenty inverted, but seems that's not the case.

landswellsong, 2020-01-08:
@yoh-there sorry took quite some time to build the new source with the Crashlytics/Google Play and stuff :)

That did help, 59A shows 66A instead of 27A now, seems good enough.

I'm however puzzled why must it be so? If the formula for higher amps is `(DC - 64) * 2.5` then 59A is 87.6%.  66A is then 90.4%, not a negligible difference. Is there a way I can sniff the raw DC data with CanZE? I'm unsure if the car can be this off, such a difference on lower amps is sure to trip the breakers.

yoh-there, 2020-01-08:
I know huh? Thing is, in NON isotp, we get the already calculated max amp from a frame that one of the computers sends on the bus repeadedly. I have not been able to find that same value as a diagnostic (query-response) field, so instead I take the raw duty cycle (which is available) and calculate the max amp from that. We could simply subtract 10%, from the duty cycle but hmm, I am not very happy about that either.

landswellsong, 2020-01-08:
I figured so, but the question then is whether the DC formula is off, the DC calculation on the car is off or something else. 

Is the DC shown anywhere on the app right now? Or if not, can you make a branch for it to do so?

I can plug the car emulator and the scope into my portable EVSE and try to figure out the real DC value, then use the same EVSE with the car and collect data from it. Also can ask the EV tuning pals to help out with several Zoes in case that's a car problem.

yoh-there, 2020-01-08:
DC is not shown but you can very easily revert it's value from the shown amperage, see https://github.com/fesch/CanZE/blob/Development/app/src/main/java/lu/fisch/canze/actors/Fields.java line 594.

I don't think it a specific car problem. I also see differences in my car, but less obvious as I charge at 16A

Alternatively, you can use the _Research facility (see blog for explanation)

landswellsong, 2020-01-08:
Thanks, will report back on weekend.

By car specific I meant there could be a software bug between pre-2015 and post-2015 models when they introduced the R engine, I know for a fact the BCB firmware versions are different and they reworked several key things there, e.g. the earthing detection protocol (which is a huge issue in our 2.5-nd world country)

landswellsong, 2020-01-11:
Okay, I ran a few tests and the results are puzzling? Seems like it's the old pilot that's consistently off by a few amps, I did notice that before but I assumed it was my non-OEM EVSE that had a bug, and since 9A and 10A are basically the same thing I never bothered to verify. Now that I have though, the EVSE seems to be absolutely standard compliant and even the slight error seems regular i.e. some of my own hickups probably, the duty cycle is relatively not noisy even on my makeshift car emulator setup with a cheap scope (gray is measured duty cycle, red is expected value, was raising from 6A to 16A with 30 sec intervals):
[image]

The following are my measurements, *P* is power, measured one is what the EVSE reports (it seems to have a voltmeter for the accounting purpose). Calculated current is convered from scope-measured DC, calculated DC is converted from CanZE ISOTP mode current. I also had the weird non-ISOTP values double checked too, the power really stays 2.1 kW all the way down. The non-rounded current in ISOTP mode seems to be the artifact of DC being rounded to integer, but not universally.

|   I (req) |  I (ISOTP) |  I (old) | I (calc) | P (meas) | P (ISOTP) | P (old) | DC (exp) | DC (meas) | DC (calc) |
|-----------|------------|----------|----------|---------|----------|--------|----------|-----------|-----------|
|   6       |  6.0       | 6        |  5.81    | 1.36    | 0.5      | 2.1    |  10.00   | 9.68      |  10.00    |
|  7        | 7.0        | 6        | 6.82     | 1.58    | 0.9      | 2.1    | 11.67    | 11.36     |  11.67    |
|  8        | 8.0        | 7        | 7.81     | 1.81    | 1.1      | 2.1    | 13.33    | 13.02     | 13.33     |
|  9        | 9.0        | 8        | 8.82     | 2.03    | 1.4      | 2.1    | 15.00    | 14.69     | 15.00     |
| 10        | 10.2       | 9        | 9.82     | 2.26    | 1.7      | 2.1    | 16.67    | 16.37     | 17.00     |
| 11        | 10.8       | 9        | 10.83    | 2.49    | 1.8      | 2.1    | 18.33    | 18.04     | 18.00     |
| 12        | 12         | 11       | 11.83    | 2.71    | 2.2      | 2.1    | 20.00    | 19.71     | 20.00     |
| 13        | 13.2       | 12       | 13.83    | 2.94    | 2.5      | 2.4    | 21.67    | 21.38     | 22.00     |
| 14        | 13.8       | 13       | 13.83    | 3.16    | 2.7      | 2.7    | 23.33    | 23.04     | 23.00     |
| 15        | 15.0       | 14       | 14.83    | 3.39    | 3.1      | 3.0    | 25.00    | 24.72     | 25.00     |
| 16        | 16.2       | 16       |  15.84   | 3.62    | 3.4      | 3.3    | 26.67    | 26.40     | 27.00     |

Lastly, speaking of the [calculation code](https://github.com/fesch/CanZE/blob/Development/app/src/main/java/lu/fisch/canze/actors/Fields.java#L594) I believe there's an error in the higher range, given the TI's datasheet for some EVSE and OpenEVSE docs, the formula is:
[image]
This gives quite a bit of error in that range.

yoh-there, 2020-02-16:
Thanks again. It's in the beta.

closed 2020-02-16 by yoh-there