fesch/CanZE #601: Renault ZOE R135

tsknightstorm, 2020-01-09:
Today I got the first R135 on my hands. Do you have any idea how to get in the CAN-Bus on the R135 ZOEs? I think there is no OBDII plug anymore, becaus this car does not require lampda tests. Maybe I can help somehow? For a privat mail connection I would be thankfull.

yoh-there, 2020-01-09:
Hi. See https://canze.fisch.lu/ze50/ You can use your email address when posting a comment on that blog post and I will reply.

Coming Saturday I'll have a second chance to fiddle with a ZE50, and this time with a real CLIP tool in our hands.

R240Pe, 2020-01-09:
I hope Saturday will be a successful day!

yoh-there, 2020-01-11:
Logging session successful. Few observations.

- The ZE50 is a completely different beast than the ZOEs up until now.
- Free frames are a thing of the past (as expected)
- Mixed 11 and 29 bit mode (as expected since some recent friendly comments), some ECUs even have double addressing
- Security gateway hides car from OBD2 connector
- Unclear if the gateway needs a special command to open up, need to write a few test scripts for that

More later.

KarelSvo, 2020-01-11:
> Today I got the first R135 on my hands. Do you have any idea how to get in the CAN-Bus on the R135 ZOEs?
[There is a photo.](https://www.goingelectric.de/forum/viewtopic.php?p=1156774#p1156774)

yoh-there, 2020-01-11:
Yes, but not enough info from the gent so far.

yoh-there, 2020-02-16:
I will close this issue; lets continue ZE50 stuff here https://github.com/fesch/CanZE/issues/588
 I will make the tags there less iOS specific. Efforts should be combined anyway

closed 2020-02-16 by yoh-there