fesch/CanZE #603: Meaning of charge ending strings?

ldti, 2020-01-30:
Hi. Small question:

What is the meaning of each of the reasons of the charge ending, such as fault 22, fail and so on?

yoh-there, 2020-01-30:
Fair question, but we only get a few numbers from the car and what we show is unfortunately all that we have from the documentation we got.

ldti, 2020-01-30:
So the strings are mapped to numbers, but you don’t really know their meaning?

yoh-there, 2020-01-30:
Correct. Some numbers we have better descriptions and then we show. If it's just a number, we have not gotten better documentation.

ldti, 2020-01-30:
Ok, I understand. Kind of a shame, though..

KarelSvo, 2020-01-31:
Take a screenshot, maybe I have an idea ;-)

ldti, 2020-01-31:
You can find all of the strings here:

https://github.com/fesch/CanZE/blob/master/app/src/main/res/values/strings.xml

KarelSvo, 2020-02-02:
There is no screenshot of your error 22 on this page ;-)

ldti, 2020-02-02:
I know, because there's nothing to really show. All the strings are there in the link I attached.

closed 2020-02-16 by yoh-there