fesch/CanZE #605: TPMS sensor ID reading in single wheel not moving possible

SMCinc, 2020-02-09:
Hello Everybody,
Using the following tools reading TPMS sesor data wiht out driving is
possible:

  *  EL- 50448 tool an 125 kHz tranceiver orginally for GM cars
    stimulates the TPMS to send. (about 10 Euro @ ebay)

  * using a DVB-T USB Stick with RTL2832 chipset, rtl_433 from Github
    and Zadig driver.
  *  Fequency 434 MHz Protocol 123 Jansite rtl_433 -f 434000000 -R 123

[image]

yoh-there, 2020-02-11:
Hi. Great! You mind if I close this issue and copy the text it to the blog? With or without attribution, what you like.

SMCinc, 2020-02-15:
Hi, of couse you can copy the cnotent to the blog. 
I am afraid I fond an answer and some new questions:
- The calibrated ID of the sensor is A3FFDA and working fine. The three most significant bytes are used. The tire is an original Renault and came with the car an Q210 from 12/2014. My summer and winter tires sensors are showing the same behaviour : 3 1/2 bytes in the ID (lowest nibble equals zero), highest bit =1 and the three highest bytes are the right ID calibrated in the ZOE. 
&7F FF FF  | 0x 80 00 00 doesn't work here. Maybe if the lowes nibble = 0 ?

An othe questtion; in DDT4all BCM 95 SW25 V1 TMPS Extra Screeen/ Data recorder_seui received sensor IDs can be seen which are not in the list. may those information could be use in CANZE to find an algorithm to read unknown sesors and add them to the car?

Have a nice weekend,
Christian

yoh-there, 2020-02-16:
Great research, thank you. I will publish on the blog soon (a couple of days to let the beta sink in)

Honest answer: I am reluctant to add the "learn tire" function, for 2 reasons:
- it needs additional hardware and however loud it is shouted on the website, it always leads to blowback. We have even seen this re needing a dongle, I kid you not.
- it is another function (after previous tires, and now the massive ZE50 changes) that should be developed without having the actual car. It is a pain in the behind and leads to buggy releases. Tires took over half a year to get it right.

What does @fesch think?

fesch, 2020-02-16:
For me personally, not having at my disposal the right hardware (in this case the car, the sensors, ...) is some kind of KO-factor. I hate doing blind programming and not being able to test. This is no funny stuff at all :-(

yoh-there, 2020-02-16:
@SMCinc Just to be sure, what ID did CanZE show for the sensor you showed? A3FFD0?

SMCinc, 2020-02-16:
@yoh-there A3FFDA is the ID that can be seen in CanZE and is working when tire is mounted .1

yoh-there, 2020-02-16:
I understand. But now I am confused a bit. Maybe I am misreading A3FFDA seems a proper match with taking the 3 MSB's, setting the most significant bit (it's already set). Where does the 0 nibble come in?

SMCinc, 2020-02-16:
exact. 
But maybe I misunderstood the rule/example  provided at goingelectric:
https://www.goingelectric.de/wiki/Renault-ZOE-Andere-RDKS-Sensoren-anlernen/

'[SENSOR-ID] & 0x7FFFFF | 0x800000

Oder im Klartext: Nur die niederwertigsten 23 bit der Sensor ID werden verwendet und das 24. Bit wird fest auf 1 gesetzt so dass die ID auf 3 Bytes (=6 Hex-Ziffern) eingekürzt wird.

Beispiel:
```
Sensor ID: 0x3F3EA642
0x3F3EA642 AND 0x7FFFFF => 0x3EA642
0x3EA642 OR 0x800000 => 0xBEA642'
```                -> I understood 3 LSBs with bit.23 = 1 

A3FFDAd0 seems to be the whole ID an I found the low nibble of byte 0 always equal 0 in my sensors. 
So I was confused because of the dercription I found ( 3 LSBs) and what I have measured (3 MSBs).

yoh-there, 2020-02-16:
Right right. Well, let's be kind to the gentleman who wrote that then and assume it has to do with the way the ID was printed on the sensors he got. Seems to me actually receiving it is the safest bet, so using your method, it's ALWAYS the formula applied on the left hand three bytes, and with printed ID's it might be either left or right.

closed 2020-08-27 by yoh-there