fesch/CanZE #606: Settings keep clearing.

KotekBesar, 2020-02-14:
Morning!

Seems like every morning I come to my car to find that CanZE doesn’t work and, when I open Settings, I find that it’s because they have cleared (back to safe driving mode).

Not sure why? If no one else is reporting this issue, perhaps it’s my phone? Using Samsung A80 (not rooted). 
[image]
[image]

yoh-there, 2020-02-14:
I have no idea and never seen this. @fesch an idea? Is maybe the app transferred to SD (Inever tested this)

yoh-there, 2020-08-27:
I hope issue went away. If not, feel free to reopen

closed 2020-08-27 by yoh-there