fesch/CanZE #608: CanZE IOS version connection problems

nunosoares73, 2020-04-18:
Good afternoon,

I bought KONNWEI KW903 Wireless WIFI ELM327 OBD-II Mini Car Auto Diagnostic Scan Tool, but I am not able to connect to the CanZE APP for Iphone.

Some screen shots of my phone and the product:
[image]
[image]
[image]
[image]
[image]
[image]
Can you understand why?

Tks

FredLeudon, 2020-04-18:
Hi, 

Up address is the dongles IP address.

Like this : 

On settings app, 

Configurar IP : manual
Endereço IP : 192.168.0.11
Máscara ... : 255.255.255.0
Router : leave blank

So when connected to the dongle, your phone will use cellular to go to the web. 

Regards. 

Frédéric RICHARD

nunosoares73, 2020-04-19:
Hi Frédéric,

First of all thanks for the help.

If I understood what you said, in the CanZe app remove the IP Address from the "router" field and leave it blank. The problem that in the app settings, I can only change the ip address or the port!
The "router" field is not visible.

[image]
In the Iphone settings, I changed the setting to manuel and it looked like this:

[image]

But the problem remains.

[image]

In other words, the CanZe app always assumes the "router" field with the ip address 192.168.0.10

What am I doing wrong?

FredLeudon, 2020-04-19:
Hi,

Enter 192.168.0.10 in CanZE app. Port would be 35000.

Assume your car is running (key in, break & start pressed)

If CanZE can’t connect, find the test elm button and send log.

Otherwise, there is a v2 still in development.

You can install this public beta here : https://testflight.apple.com/join/nhMX6f5n

Thanks,

Regards.

Frédéric RICHARD

nunosoares73, 2020-04-19:
Hi Frédéric,

With the normal version of CanZe and following steps:
- Enter 192.168.0.10 in CanZE app. Port would be 35000.
- Car is running (key in, break & start pressed)

The result was:
[image]
[image]
[image]

With the beta version:

[image]
[image]
[image]

While reading the car was on, the foot brake and accelerator on the bottom.

With the beta version there are no more communication problems with the OBD device, however it seems that the application cannot communicate with the car.
I also notice that in the beta version settings there is no field to choose the Zoe model.

Any more suggestions or configuration changes?

Once again, thanks for your help.

Nuno

FredLeudon, 2020-04-19:
Ok, same issue with the beta version. 

In the beta version, tap the top left icon (sliding menu) and select activities.

Then swipe to « technical data » and « test elm »

It seems your dongle doesn’t respond correctly.

Then go to settings and swipe to find « manage log files »

Then send logs. 

Frédéric RICHARD

nunosoares73, 2020-04-20:
Hi Frédéric,

I completed the ELM test, was sent by email to ioscanze@icloud.com

CanZE_Debug_20200420_15535856
CanZE_ECU_20200420_15535856
canze_livedata.fic

Best regards,

Nuno

FredLeudon, 2020-04-20:
Ok, thanks, will have a look soon. 

Regards. 

Frédéric RICHARD

nunosoares73, 2020-04-24:
Hi Frédéric,

Sorry for the insistence, but you have news from the logs?

Regards,

Nuno

FredLeudon, 2020-04-24:
Sorry, I didn’t have time to have a look at logs. Will tri before Monday. 

Regards. 

Frédéric RICHARD

FredLeudon, 2020-04-25:
Hi, 

I had a look at your log files.

Its looks like Your dongle is not fully compatible.

You can have a look at https://canze.fisch.lu/elm327/ <https://canze.fisch.lu/elm327/> for more info.

This one would  work (no guarantee)  https://www.amazon.fr/Lecteur-scanner-Diagnostic-travail-Android/dp/B07B93MYX9 <https://www.amazon.fr/Lecteur-scanner-Diagnostic-travail-Android/dp/B07B93MYX9> 

There are lots of link to find ODB2 dongle in this thread : https://renault-zoe.forumpro.fr/t8555-canze-pour-iphone <https://renault-zoe.forumpro.fr/t8555-canze-pour-iphone>

Regards

Cordialement,

Frédéric RICHARD

nunosoares73, 2020-04-29:
Hi Frédéric,

Unfortunately, neither of the two links you shared from Amazon (FR) delivers to Portugal.

From your experience and opinion, do you think this will be a good option?

https://pt.banggood.com/KONNWEI-KW902-WiFi-ELM327-OBD2-OBDII-Car-Diagnostic-Scanner-Reader-For-iOS-Android-p-1461010.html?gmcCountry=PT&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=lijing&utm_campaign=ssc-ptg-0724-usw-all&gclid=EAIaIQobChMI75jtm66N6QIVzEPTCh1lZgdTEAQYASABEgKRLvD_BwE&ID=233&cur_warehouse=CN

Regards,

Nuno

yoh-there, 2020-12-03:
I assume this is resolved. If not, feel free to reopen

closed 2020-12-03 by yoh-there