fesch/CanZE #610: "Carry our Overhaul" warning message after writing TMPS IDs.

Demolin-L, 2020-05-10:
Hi,

First, thanks a lot for this App in general and adding the option to save 2 TMPS tire sets, very convenient!

I am using Android phone, CanZE 1.51 Build 2019-12-28 12:08:57 and orange KW902 dongle.

My issue:
After changing from winter to summer tires on my wifes Zoe R90 2018 a week ago, immediatly after writing the new TMPS IDs, the display started showing a big orange warning to "Carry out overhaul".
Everything seems to still work just fine.

When writing the TMPS i get many error message (too many and being listed too fast to see properly), but this is the same as when i did it last fall, and TPMS IDs are still being written correctly and everything seems to work ok.

My suspicion is that CanZE app might have some settings for newer firmwares than what is currently in the car and might try to write some data that the car does not understand/support, or something like that, and that is causing this warning.

Has anyone else seen this and have any idea what is causing it and what to do about it?
Is it safe to keep driving?

Notice-Jeroen-Meijer, 2020-05-10:
That is an unknown to me. The TCU involved is the BCM. You can always check it's Version through the Firmware screen.

The BCM is not super critical, it s not involved in the drive train or brake system. These types of errors have a tendency to simply disappear after a while. If not, you can try to clear the BCM's DTC's. Also what MIGHT work is to scroll through the messages using the up-down buttons until the orange warning pops up and then hold the button for 10 seconds. Some errors are cleared this way.

Unfortunately, I don't have a '90 model here to test on.

yoh-there, 2020-05-10:
That is an unknown to me. The TCU involved is the BCM. You can always check it's Version through the Firmware screen.

The BCM is not super critical, it s not involved in the drive train or brake system. These types of errors have a tendency to simply disappear after a while. If not, you can try to clear the BCM's DTC's. Also what MIGHT work is to scroll through the messages using the up-down buttons until the orange warning pops up and then hold the button for 10 seconds. Some errors are cleared this way.

Unfortunately, I don't have a '90 model here to test on.

Demolin-L, 2020-05-11:
Not sure if its helpful, but the cars BCM versions are, :

DiagVersion: 04
Supplier: 001
Soft: 140E
Version: 2700
Debug: 765.6180

ElbilsChrippe, 2020-05-15:
I have a zoe r110
Diag 04
Supplier 037
Soft 2396
Ver 0E40

And can write fine.

yoh-there, 2020-05-15:
My own Q210 was the sane as @Demolin-L , but a lmuch lower version: 4 / 001 / 140E / 2010. And this is without TPMS of course. Sorry I don't have much to say about this. Were you able to clear the error with the stalk?

Demolin-L, 2020-05-30:
Thanks for replies and suggestions!
Finally got around to trying it today.

After a lot of fiddling around in variuos menus looking for these buttons and trying and failing to clear the DTCs from CanZE app, I finally got it cleared by holding one of the cars menu buttons for 10+ish seconds. (in front of the cars stearing wheel on the right side, near the Start button, on the end of the light/wiper switch stick?)
This also seemed to reset the Service interval counter as a side effect, but its going in for service next week anyway so thats not a problem.

Both reading and trying to clear DTCs in the app failed, probably because i could not figure out how to enable it?
-Enter key card - check
-Gear to D - check
-Hold start button for 5 sec until remove card appears - check
-Gear to P - check
-Wait until poller thread is stopped - nothing visible is happening, is somehting supposed to happen?

Anyways, the big orange light is gone and the wife is happy! :)

closed 2020-08-27 by yoh-there