fesch/CanZE #611: Tpms write and still pressure warning

ElbilsChrippe, 2020-05-15:
Hi,
I've read out my winter tires id:s and written them in canze but still get tpms warning. Is the positions saved in the zoe? I. E I have to have exactly right sensor ID on right tyre position?

Yes I've reset tpms sensor with pressure ref ongoing. 

Kind regards, 
Christer

yoh-there, 2020-05-15:
The car has only one antenna and does not "know" which valve is mounted where.

Be aware the valves only transmit when you are driving.

I have been told a few ZOEs use tire warnings based on rotational speed difference. If that's true you'd see no pressure and the car might know which wheel.

ElbilsChrippe, 2020-05-15:
Hi yoh-there,

Well the interface in canze has position so I think there is some connection, yes a omni antenna cannot detect direction, but could sence amplitude differences (if I would implement it that would have been a way to know the wheel id).

But then I'm confused since the wintertyres worked and the new sensor id should then work if the 4 tpms id's without position is saved in the BCM ECU system. Maybe there is a typo in the id's I'll check them again.

Anyone knows if position relates or what could otherwise go wrong, write operation to each position works. I've read out the wintertyres with the maxitpms pad - it was in decimal number but it is just a conversion in a programmers calculator to get it into hex.

yoh-there, 2020-05-15:
No it doesn't work like that. When the valves are learned by the car they are put one by one in the proper registers by the dealer.

Remember, ZOE only used 24 bits. Depending on your reader tool and brand (print) it can be first or last.

ElbilsChrippe, 2020-05-15:
Looking at the CANZE diagnistics I see pressure from all positions like Forward Right, Forward Left and so on so surely the vehicle can sense the 4 different valves RF transmissions. This is data sent from the BCM on the CAN-bus surely.

My question about proper register is if it is like:
FR = Sensor A, FL = Sensor B, AR = Sensor C, AL = Sensor D
Or if all IDs are interchangable and the position doesnt matter like:
Sensor IDs is in a list

In the MAXItpms pad I can sense the RF data: temperature, pressure, battery condition and ID. 
The ID is surely 24 bits when I read and write in CANZE and also from the MAXItpms pad, did you mean that the byte order can be little endian and big endian, like I have to byte-swap the numbers?

I think it is the case I have to have the right sensor ID on the right position as it is programmed by the dealer :( That will give me 4!=24 number of possible combinations to test here, i.e. it will take some time.

yoh-there, 2020-05-15:
Correct, but there have been comments from users that I copied in the blog, i.e. 

https://canze.fisch.lu/tpms-once-more/

Also check out the detailed comments in this issue https://github.com/fesch/CanZE/issues/577

ElbilsChrippe, 2020-05-16:
Thanks so much yo-there,

My mistakes where twofold. Reading out the sensor ID was in hex and not decimal. 
Then I didn't understand that you through diagnostics service write valves id's to the vehicle register. I was dumb and thought you wrote id's to valves. 

Now since I can read id's with the maxitpms pad and write to vehicle I'm home free 😊

closed 2020-05-16 by ElbilsChrippe