fesch/CanZE #615: IOS: Only first TPMS ID is written

niggli, 2020-07-31:
I am trying to store my summer tire IDs in the Zoe. If I change all four TPMS IDs in the "Tires" menu, and click on "write", after a short time a popup tells me the operation was successful. However, if I read them again, only the first one is written. Also of course after driving around the car starts complaining and in CanZE I only see the pressure of the front left wheel.

After randomly trying around in this menu for half an hour, suddenly I was able to write the second and third ID also (probably depending on timing or order of the buttons pressed), but the last one seems to be impossible.

The car is a R240, the dongle is a Konnwei Wifi. CanZE is 4.0.0.42 with testflight. I read that you don't have a car with TPMS, Is there any kind of logfile I can extract to help you diagnose the problem?

niggli, 2020-08-20:
With the new version, it now shows an error message and probably doesn't write any IDs at all. I have sent the corresponding log files with the built-in logfile-to-mail function.

FredLeudon, 2020-08-20:
Thanks.

Logs will help I hope. 

Regards. 

Frédéric RICHARD

yoh-there, 2020-12-22:
Stale

closed 2020-12-22 by yoh-there