fesch/CanZE #616: Using CanZe on a Zoe model Q210

jenkinsonandy, 2020-08-04:
So, I've installed the App, bought 3 ODB dongles the last being a Konnwei KW902, and still it doesn't work.  I get "Close this screen and make sure your device (is) paired and connected". Konnwei recommends using a Torque app and that connects fine, though is useless for a Zoe. So what can I do now?
[image]
My phone is a Samsung GS7

madamic, 2020-08-11:
I'm using the CanZE app with my Zoe Q210 as well, with the Konnwei KW902, and it works perfectly fine. It's just a little fussy when you first make the connection, sometimes it loops itself into a "No connection" forever, and you have to reset the whole thing. What you should do: Switch the dongle (KW902) on, then open the app, and allow it to use Bluetooth. Then, go to app Settings and select your Konnwei device. That should work and you will see the "ELM is now ready" notification. Give it a little time, sometimes it takes a while, like 10 seconds or so. If that doesn't work, close the app and reset the dongle again. Like I said, this very first time when you change the app Settings is very fussy, it took me like 15 mins to get it working and since then, no problems.

yoh-there, 2020-08-27:
See https://canze.fisch.lu/qa/, 9th question. You need to pair it in Android and then choose the proper device in CanZE's settings screen

closed 2020-08-27 by yoh-there