fesch/CanZE #617: DTC Readout - "Get DTCs field does not exist"

madamic, 2020-08-08:
Hello,

First of all, I would like to thank this community for the amazing work, the app is awesome and really useful!

To the point: I can't manage to read the DTCs in my 2015 Q210 Zoe. Everything else works perfectly and I can access all the parameters of the car, including "All data" and "Firmware" from each individual onboard computer. However, when I query DTCs, I get "Get DTCs field does not exist" and consequently no DTCs. I find it extremely hard to believe that there are no recorded DTCs in any of the computers, I even had a "CHECK Electric System" flashing today! :O

I'm using the recommended Konnwei KW902 dongle and the latest version (1.51) of CanZE on Android. Any ideas would be greatly appreciated.

-Michel

yoh-there, 2020-08-27:
Confirmed, thanks

yoh-there, 2020-08-31:
Fixed in 1.52 8640b0f76dc9ed94c8ab7dfa48112f8ebd3cbc7a

closed 2020-08-31 by yoh-there

madamic, 2020-08-31:
Great to hear, thank you!!

madamic, 2020-09-18:
Hello,

So today I finally tested the DTC readout on the updated 1.52 version of CanZE. Most of the computers respond properly, which is awesome. However, the EVC, PEB and sometimes CLIM now return this error message:
"Query sent, but unexpected result received:[-E-ISOTP rx unexpected 1st nibble of 1st frame:NO DATA]"

It's weird because CLIM, for example, sometimes returns this error and sometimes not. The EVC and PEB, on the other hand, always respond with the error.

Some computers (AIBAG, UPA, LBC2, BCB ...) are still responding with "Get DTCs field does not exist"

It would be nice to get the EVC and PEB working, I'm not that much interested in the rest.

Thanks again!