fesch/CanZE #620: Research screen shows NaN for certain IDs

Gosa7777, 2020-09-06:
Works for some (e.g SOC), but others (e.g. accel pedal) show NaN.
In log file, results are correct.

yoh-there, 2020-09-06:
I assume you are modifying the research.csv file on the phone. This is
exactly why I made it, to check what works and what not. Or to check fields
that are not coded in Canze

Notice-Jeroen-Meijer, 2020-09-06:
Maybe I misunderstood. Are you saying the display on the screen is in error, but it actually fetched the data correctly? If so,please email me your csv file, thanks!

Gosa7777, 2020-09-06:
Hey Jeroen,
yes you understood correctly,For the attached file, it works correctly for all values but pedal position. 

 
 

Build software better, together

GitHub is where people build software. More than 50 million people use GitHub to discover, fork, and contribute ...
 |

 |

 |

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Research screen shows NaN for certain IDs · Issue #620 · fesch/CanZE

yoh-there, 2020-12-03:
I apologize for this stale issue. ZE50 stuff taking spare time. I never received a file; I don't think that works in replying to github emails. Since it is text, can you paste it here? Also please indicate your car type. Thanks

yoh-there, 2020-12-18:
Friendly reminder

yoh-there, 2020-12-22:
Closed (stale, and new versions available)

closed 2020-12-22 by yoh-there