fesch/CanZE #621: Zoé VIN decoding : Help Wanted

FredLeudon, 2020-09-06:
Hi all

I need infos on Renault's VIN

Please, reply with first 10 digits of Your VIN (VF1AGVYA06) and full identification of Your Zoe

In this example, this is :  Zoé Q210 2013 Intens

Thanks for feedback, help is needed to correctly identify ZE40 and ZE50

Notice-Jeroen-Meijer, 2020-09-06:
Q210 Life 2013 VF1AGVYA04

FredLeudon, 2020-09-06:
Thanks, I really need vin for ZE40 & ZE 50

My new ZE 50 R 135 is VF1AG0004

Frédéric RICHARD

fgmccrea, 2020-09-08:
R110 40 VF1AGVYF05

fgmccrea, 2020-09-08:
Version from that is AGVYF0

FredLeudon, 2020-09-08:
Thanks. 

Frédéric RICHARD

fgmccrea, 2020-09-08:
Another one 40 VF1AG00006

FredLeudon, 2020-09-08:
Please can you say if it’s a Q90, R90 or R110 plz

Frédéric RICHARD

fgmccrea, 2020-09-08:
ah sorry its also r110 younger

KarelSvo, 2020-09-11:
> Please can you say if it’s a Q90, R90 or R110 plz Frédéric RICHARD

All Zoe manufactured after February 20, 2018 have a consecutive VIN, independent of engine and battery.

fgmccrea, 2020-09-12:
I thought it was R110Its probably manufactured in 2018Just so I don't give you bad information, can you tell me how to be certain its R110?

madamic, 2020-09-18:
Zoe Q210 (2015) Intens, 22 kWh.
VF1AGVYA15

KarelSvo, 2020-09-19:
> Hi all
> 
> I need infos on Renault's VIN
> 
> Please, reply with first 10 digits of Your VIN (VF1AGVYA06) and full identification of Your Zoe
> 
> In this example, this is : Zoé Q210 2013 Intens

VF1AGVYA0**6**  just checked, this version does not exist in the Renault vehicle database.
there are
VF1AGVYA0E  year 2013
VF1AGVYA04 year 2012 and 2013
VF1AGVYA05 year 2013 to 2016

FredLeudon, 2020-09-19:
Yes sorry a gave 10 digits.

Only first 9 digits are useful to decode the VIN but some VINs doesn’t match the general rule. 

VF1 means Renault 
AGVYA0 means Q210
AGVYB0 means R240

I saw AGVYA1 and AGVYB1 ....

Next digits are the car’s serial number.

Frédéric RICHARD

KarelSvo, 2020-09-20:
> I need infos on Renault's VIN
What is the background of your question? would you like canze to determine the vehicle type using the VIN?
A automatic detection of whether you are connected to phase 1 or phase 2 is easier to implement.

FredLeudon, 2020-09-20:
Yes, this is what I want to do. 

Frédéric RICHARD

KarelSvo, 2020-09-20:
> Yes, this is what I want to do. Frédéric RICHARD

Selecting phase 1 vehicle types is unnecessary.
There are no different control units installed.
There are only differences between phase 1 and phase 2.
And that can be selected with a request to the EVC.
11 bit negative response = phase 2
29 bit negative response = phase 1
11 and 29 bit negative response = Fake Elm

or ask the UPC 74D about the DIAG VERSION
<IdFrame request="2180" response="6180"/>
<IdByte rank="6"  value="36"/> #phase 1
<IdByte rank="6"  value="3A"/> #phase 1
<IdByte rank="6"  value="40"/> #phase 2

FredLeudon, 2020-09-20:
Ok and how do you deal with :

- twizy 
- kangoo 
- fluence
- smart / twingo

Frédéric RICHARD

KarelSvo, 2020-09-20:
> Ok and how do you deal with : - twizy - kangoo - fluence - smart / twingo Frédéric RICHARD

You forgot the City-K ZE ;-)

MarkRotNF, 2020-10-06:
ZE 50 R110 Exp.
VF1AG000X

SMCinc, 2020-10-23:
VF1AGVYA05
Zoe Q210 Intens 2014

Thecurrentcollector, 2020-11-06:
Zoe 2020 ZE 50 R135 Intens
VF1AG00006

yoh-there, 2020-12-03:
@FredLeudon do you have enough data?

FredLeudon, 2020-12-03:
Yes, can be closed. 

Frédéric RICHARD

closed 2020-12-03 by yoh-there