fesch/CanZE #622: Control Pilot Error

Sebes1, 2020-09-10:
I realised an error of Control Pilot value in version 1.52 Build 2020-08-30 12:13:41. It shows double of the set value.
Additionally, the Control Pilot signal should be an integer between 6 and 32 (Amper) there is no decimal notation of the standard. Do you really read decimal values from the car.

Do you experience the same?

Thanks,
Sebes

Sebes1, 2020-09-10:
Sorry in the Q version the CP could be upto 64 (Amper).

Notice-Jeroen-Meijer, 2020-09-10:
Can you please provide your car type? (Q210 / R240 / ZE40 / ZE50)? We calculate the amps from the pulse width, maybe the charger reports in a different unit. As for decimals, well, basically all values are presented as one decimal. I'll see what I can do, but regard that as a feature request ;-)

yoh-there, 2020-09-10:
Can you please report your car type (Q210 / R240 / ZE40 / ZE50)? We calculate the current from the pulse width, maybe the charger reports in different units.

As for the decimals: we present all numbers basically with one decimal. I;ll see what I can do, but please regard that as a feature request ;-)

Sebes1, 2020-09-11:
First of all I would like to thank all of you the great job what you did about CanZE.

I have been using the OBDII dongle and CanZE about 2 years ago. I bought the ZOE in the end of 2018. I recorded all of my charging since then using CanZE. My ZOE is R240 (2015 69785km 98% 10charges/week 1.6-22kW)

If I remember well, I never saw any CP error in the previous versions, only in not connected state (88.0). I always update CanZE when my phone offers a new version.

When I first saw the CP error I thought that my collegue made something wrong about the EVSE firmware.
But I doblechecked our pilot signal with oscilloscope, and checked the ZOE charging current with CanZE, with current probe and of course with the internal current sensor inside the EVSE. All were OK, except CanZE CP value.
Yesterday, I tried a public charging station (3x32A) and CanZE CP value was 105.0. Without connected showed 340.0.

My ZOE got the last firmware and software update in the middle of July.

yoh-there, 2020-09-11:
Got it. I will have a peek this weekend. Thanks for the praise and extensive reporting

yoh-there, 2020-09-11:
I checked with my Q210 tonight on a single phase 16A charger. Displayed value: 16.2 A. Hmmm. Maybe the R cars report a different duty cycle. I'll check more later. Thing is, I never had reports like this earlier and there are so many R models around, and as you said, it was OK. I'll run a clean install from the play store on an old phone tomorrow.

yoh-there, 2020-12-18:
Is this issue still relevant? We're in the process of redoing some stuff on the charging tech screen (mainly CCS though). If so, would you be so kind to provide us with the following info?
- in settings, switch on Debug and get a charging tech session. Then send/paste the debug file in Android > lu.fisch.canze > Data > Files. Please also make a screenshot.
- a screenshot from the firmware screen, while having tapped the BCB.

Sebes1, 2020-12-20:
It is maybe not the biggest problem... However, if you can solve this than I try to help you by provinding data.
I have never used SD card data logging function. I tryed but it failed. The requested library is empty...

I attached Settings, Firmware, and Charging  screenshots and a picture from our prototype where you can see the current settings.
Please let me know what could be the problem with SD card logging.
Thanks.
[image]
[image]
[image]
[image]

yoh-there, 2020-12-20:
Thank you for the detailed info! I will check for resolution changes.

Logging data should be written now to Phone > Android > lu.fisch.canze > Data > Files > debug_xxxxxxxx (date coded). Are you sure you are looking in Phone? It's no longer written to the SD Card, which has to do with a change in Android itself. If you can find it, it would be extra helpful. If not, oh well ;-)

Sebes1, 2020-12-20:
Sorry, you have right. I looked in SD card only.

[BCB-2020-12-20-11-07-37[1].txt](https://github.com/fesch/CanZE/files/5720908/BCB-2020-12-20-11-07-37.1.txt)
[debug-2020-12-20-11-13-12[1].log](https://github.com/fesch/CanZE/files/5720909/debug-2020-12-20-11-13-12.1.log)

yoh-there, 2020-12-20:
Thank you, that really helps. The car reports 12.5% duty cycle = 7.5 amps, but unavailable too, alternating It absolutely makes no sense it showing 16.2. Not giving up yet. Issue remains open as it should be solvable, since the car reports good enough data.

yoh-there, 2020-12-20:
Found it. Will correct in next beta. I will keep this open and would appreciate if you could report back after that one is out (will be somewhere in the upcoming week). Just for kicks: it's probably showing an OK value if you switch off "Use ISOTP" now.

Sebes1, 2020-12-20:
Thanks. I will check and reply to you. Thanks for the tip.

Anyway, I realised that after latest BMS update the usable state of charge is different in dashboard in the upper region (around 2-3%, f.e. 97% in the dash and 94% in CanZE). eDrive experienced the same (Youtube). Do you noticed the same?
Of course there is a point where both shows the same (100% or less than 30% if I remember well). This is truth for charging and discharging states.

yoh-there, 2020-12-21:
1.55beta1 is out!

yoh-there, 2020-12-23:
TechCharging crashes

yoh-there, 2020-12-23:
I just uploaded 1.55beta2 and it should be rolling out in a few hours. It would be splendid if you could try with both ISOTP on and off. Thank you again for keeping on testing.Your use case has to do with a far more recent BCB firmware version than mine.

Sebes1, 2020-12-24:
I tested the pilot signal from 6A to 16A with 1A steps. Some values are perfect the others are 0.1A more or less. As I wrote to you there is no decimal notation in the standard, so you can round to get an integer. But this is your decision.
Anyway, it is perfect as it is.
I will test from 16A upto 32A next year in my workplace. 

Regarding to ISOTP (I do not know what it is), I swithed off in the Settings and I do not realized any difference. But when I opened Settings again ISOTP was on. I tried several times, but got the same result. What did I do wrong?

yoh-there, 2020-12-24:
Thank you, good news.

ISOTP mode is only relevant for Ph1 cars and voids using "free frames". It's more reliable but it "does things" on the canbus, where the non-free frame mode is only "listening" for most data.

I see that the ISOTP switch is not retained, which is an error. Thanks for reporting, I'll have a look at that.

yoh-there, 2020-12-24:
Sorry, slightly misleading info from my end. Since late 2019 we nudge ISOTP on for KONNWEI style dongles and off for CanSee. You can still change them but next time the setting screen is pulled up, they will get the nudge again. So, it's not a bug, it's a feature ;-) I think we can close this issue (again)?

Sebes1, 2020-12-25:
Thanks for the info.
I had the possibility to check upto 32A, while Zoe limited the charging power. So, it is working fine. It seems that there is somewhere (Zoe?) a rounding error, while only that numbers are integer, which can be divided by 3 without residue.
Anyway, thanks for the help. Of course you can close this issue.

yoh-there, 2020-12-25:
Decimals fixed in d594c346 . Closing :-)

closed 2020-12-25 by yoh-there