fesch/CanZE #623: DTC readout - some new errors

madamic, 2020-09-18:
Hi,

This is the continuation of thread #617 on the issues with DTC readout where I was getting "Get DTCs field does not exist" on my 2015 Q210 Zoe, Konnwei KW902 dongle, CanZE 1.51 on Android.

So today I finally tested the DTC readout on the updated 1.52 version of CanZE. Most of the computers respond properly, which is awesome. However, the EVC, PEB and sometimes CLIM now return this error message:
"Query sent, but unexpected result received:[-E-ISOTP rx unexpected 1st nibble of 1st frame:NO DATA]"

It's weird because CLIM, for example, sometimes returns this error and sometimes not. The EVC and PEB, on the other hand, always respond with the error. Therefore, I can't see any DTCs from there.

Some computers (AIBAG, UPA, LBC2, BCB ...) are still responding with "Get DTCs field does not exist".

It would be nice to get the EVC and PEB working, I'm not that much interested in the rest.

Thanks again!

Notice-Jeroen-Meijer, 2020-09-18:
The older issue was really "on us" so to speak, the definitions were omitted in a release unnoticed.

This is a harder one to crack as it is really the car not responding as expected. I'll see if I can replicate it on mine, luckily a Q210.

yoh-there, 2020-09-20:
Issue confirmed, but only for the KONNWEI. It works fine with CanSee (which is why it slipped through). So this is either an unknown issue with the driver for the KONNWEI, an incompatibility between ZOE's messaging and the KONNWEI (not very likely, but I can imagine i.e. an overflow, CanSee showed a lot of messages in the EVC on my car), or a combination.

This will take some work to find out. I am quite reluctant to fiddle with the KONNWEI driver code, it was quite hellish to get it stable, but I'll see what I can do.

madamic, 2020-09-20:
Wow, interesting. Thanks for taking a look at it!

Last time when my Zoe had a service in January, Renault printed out the list of found DTCs for me and the EVC showed 7 DTC error codes. I don't know if they cleared them or whether they are still stored. Do you think 7 codes are enough for an overflow? I slightly doubt it's an overflow because with the CLIM computer, I sometimes get the "unexpected result received" and sometimes just a regular "No active DTCs." So even with zero DTCs, this error can occur. And on some other computer (don't remember which), I get a "Timeout" from CanZE. So maybe it really is the KW902 having some issues with the Zoe.

By the way, the provided list of DTCs from Renault also showed the odometer count for the last occurence of the DTC that was related to a warning light. Is it possible to see the odometer count of each DTC triggered in CanZE or is there always just a listing of them?

Thanks again.

yoh-there, 2020-12-18:
We have fiddled a bit with the KONNWEI driver and DTC. And will even more in the next (beta) release. Could you please try again?

madamic, 2020-12-19:
OK, I tried again today and now we're back to "Get DTCs field does not exist" on all computers. :D Latest CanZE 1.54.

yoh-there, 2020-12-19:
Strange. I tested with my Q210. Back to the drawing board.

madamic, 2020-12-19:
Have you tested with the Konnwei KW902? I have an impression this dongle is rather slow, I have to wait quite a lot sometimes for various menus (multiple seconds). Maybe this "Get DTCs field does not exist" is some sort of a timeout issue? No idea.

yoh-there, 2020-12-19:
TBH, probably not. The dongle is slow indeed, but on the other hand, it should be able to handle this. Then again, should and is are different words :-/

madamic, 2020-12-19:
Haha, indeed. :D Well, what did you do in 1.52 to make it (almost) work?

yoh-there, 2020-12-19:
Problem found. Will be fixed in the next beta. Regression introduced by Ph2 implementation :-(

It might take a few days for the next beta to be released. I'll leave the issue open until that time, and would appreciate if you could confirm the fix by then.

madamic, 2020-12-19:
Sure, I'll be happy to check it out! Just let me know when the update comes out and I will try again.

yoh-there, 2020-12-21:
1.55.beta1 is out!

closed 2020-12-21 by yoh-there