fesch/CanZE #624: Communicating with the ELM327 manually

nobodyinperson, 2020-10-02:
Hi everyone,

I am desperately trying to find out how to communicate with the ELM327 directly on a Linux box via Bluetooth. The CanZE app refuses to connect to the dongle on all our phones (some other people's phones work, though). So I'd like to understand how it works under the hood and hack something together myself.

I figured out how to:

- connect to it via bluetooth (normal procedure)
- bind the bluetooth device to a serial port (`sudo rfcomm bind 0 MACADDRESS 1`)
- open a serial monitor to communicate with the ELM327 (`pio device monitor -p /dev/rfcomm0`)
- configure the ELM327 via AT-commands so that it's usable in a serial console (`ate1 ats1 atsp6 atat1 atcaf0 atfcsh77b atfcsd300010 atfcsm1`, adapted from [here](https://github.com/fesch/CanZE/blob/master/app/src/main/java/lu/fisch/canze/devices/ELM327.java))

Now what I don't understand: I assume that for example [this table](https://github.com/fesch/CanZE/blob/master/app/src/main/assets/FFC_Fields.csv) contains the OBD-codes for use with a ZOE. Now if I want for example the current [charging power](https://github.com/fesch/CanZE/blob/master/app/src/main/assets/FFC_Fields.csv), I assume I can use the code `042e`. This (always) gives the answer: `03 7F 2E 7F AA AA AA AA`. Ignoring that it's always the same, I assumed `startBit` and `endBit` would mean the position inside these bytes, which in this case would be bits 56 to 63, i.e. the last byte. But `0xAA` is `170`, times the `resolution` of `0.3` it would be `51`. Still does not make sense...

How exactly does one get the data out of the ELM327? Am I reading the table wrong? Am I interpreting the output incorrectly? I'd really like to be able to read out at least the current state of charge and the charging power...

Any help would be greatly appreciated!

Cheers,

Yann

Notice-Jeroen-Meijer, 2020-10-02:
I would advice to have a more detailed look at the ELM327. java driver. For frames under 0x700 it will apply a filter and then use ATMA to simply let every frame that matches pass on. Regard it as passing free, unsolicited frames to the serial link.

0x700 and up are diagnostic commands to the car. Ask something, get a reply. Unfortunately, there is a lot of flow-control / frame (dis)assembly going on. as you can see in the code. Read up on the ISO-TP protocol if you want to delve into that.

Bottom line is: the ELM327 is a B*tch. If you want to get quick and lazy results, you might want to consider our self designed CanSee DIY dongle as an alternative. Believe me, there is a reason we put in the effort...... ;-) 

https://gitlab.com/jeroenmeijer/cansee

KarelSvo, 2020-10-03:
> Hi everyone,
> 
> I am desperately trying to find out how to communicate with the ELM327 directly on a Linux box via Bluetooth. 

Give me an email address.

closed 2020-10-03 by yoh-there

KarelSvo, 2020-10-03:
open

nobodyinperson, 2020-10-04:
nobodyinperson at posteo de