fesch/CanZE #625: Any explanation for Flt4 and Flt5 error codes

Shrolk, 2020-10-16:
It seems there is no explanation for end of charge caused by Flt4 ou Flt5 codes.
I guess it's an error code but what are the meanings of thoses ?

yoh-there, 2020-10-23:
This mnemonic is the only thing that was listed in the documentation files we got our hands on. I have no idea what they actually mean. Silly I know, but I included them as some others do make sense and keep it complete.

closed 2020-10-23 by yoh-there

danielb7390, 2021-07-20:
A few months passed, i guess still nobody knows what they mean?
My zoe keeps stopping charge after a ~2h, i just unlock the car and open the door and it starts charging again...

[image]