fesch/CanZE #626: TPMS Can't read or write in Android

Luraholme, 2020-10-22:
Great app
But I can't read or write tires information.
The file path seems to be wrong, 
Read: OK
Save/Load to B : "Cant access file/storage/emulated/0/CanZE/_TiresB.cvs"
Write: "Could not write TPMS value 2-E-ISOTP rx unexpected 1 st nibble of 1 st frame_ NO DATA" (no file written)

Can't find any files in the phone, serched in the phone for new cvs or other new files.
Not even log files, otherwise I will mail it.
Tried restart, reinstall, all privileges, clear cache

1.53beta1
Build 2020-08-30 12:19:36
Android 10

Notice-Jeroen-Meijer, 2020-10-22:
@fesch do you know? From memory, I thought we queried Android for the "right" path to access internal/external files.

The Could not write is another issues entirely. Can you read data at all? It seems it's not too happy with the dongle. Please confirm dongle type (KONNWEI?) and car type (pre-ZE50 I presume)

Luraholme, 2020-10-22:
I can read all data
But I can't save or load tire data to the phone, no such file path in the phone. [Image]
[Image]
So I think OBD is ok

fesch, 2020-10-22:
Indeed, the path if _NOT_ hardcoded:
`Environment.getExternalStorageDirectory().getAbsolutePath() + "/CanZE/";`

SMCinc, 2020-10-23:
Hello,
I have observed the same behaviour with Android 10 / MIU12.0.1 : In error message the path to CanZE folder is the right one.
Neither logs nor _TiresA/B.csv files can be written or read.
SD card can not be chosen (tick disappears) in prferences menu.

With my elder device using Android 7 everything works fine.

Using 1.53beta1
CANSee dongle
Zoe Q210

yoh-there, 2020-10-23:
> 
> 
> Indeed, the path if _NOT_ hardcoded:
> `Environment.getExternalStorageDirectory().getAbsolutePath() + "/CanZE/";`

While I do have an Android 10 and it operates without issue on log files, Environment.getExternalStorageDirectory() is deprecated in API level 29 (=Android 10). Some vendor releases might have been a bit stricter or something.

When I have time I'll update the code to use mContext.getExternalFilesDir(null).getAbsolutePath(); as per https://stackoverflow.com/questions/57116335/environment-getexternalstoragedirectory-deprecated-in-api-level-29-java

Luraholme, 2020-10-23:
I read it on stackoverflow
Tricky and annoying that A11 will create the same problem.
Thank you for great code

tac2s, 2020-11-29:
Hello, so do have Android 10, and got same problem with writing down IDs of wheels.
File TierA.csv was successfully created after "read"
While write is not possible and after driving 20-30 km error "check tyre sensors" appears
[image]

tac2s, 2020-11-29:
Also do have screen recording

tac2s, 2020-11-29:
How can I upload video?

[Video about tpms](https://drive.google.com/file/d/19ooJBuQWnmdPfEcRhU-OmypZjpwSHWSW/view?usp=drivesdk)

yoh-there, 2020-11-29:
In the next Beta release (somewhere next weekend) this will be fixed by moving all files to the new framework. Unfortunately this will make the research facility fairly useless, but this is more important and I will try to fix that in another way, ie embedding a simple research editor in CanZE

yoh-there, 2020-11-29:
This issue is re. file handling. If there is an issue with reading/writing TPMS from/to the car, please open a new issue. Thanks

closed 2020-11-29 by yoh-there