fesch/CanZE #627: Can't find log files

miguel-negrao, 2020-11-24:
In my phone I can't find the log files of CanZE.

I've built the app with platform version 29. I see that getExternalStorageDirectory() is [deprecated](https://developer.android.com/reference/android/os/Environment#getExternalStorageDirectory()) in API 29.

I also don't have a /storage/0/emulated directory. My shared storage folder is in phone's flash memory is in /storage/sdcard0 . My actual physical sdcard is in /storage/0000-0000.

Phone is running Android 10.

Does this mean I should build with platform v. 28 ?

yoh-there, 2020-11-24:
Probably yes, or just recode it to a folder of your liking while testing. Google changed their storage model significantly and it's getting harder to work with "shared folders". To be honest, I did read up on it, but didn't dive in to it deep.

yoh-there, 2020-12-03:
This is fixed in the Beta release by moving all files to the new framework. See blog https://canze.fisch.lu/canze-file-location/

closed 2020-12-03 by yoh-there