fesch/CanZE #642: Android 10 stacked storage

landswellsong, 2020-12-07:
Hi

I recently switched phones and tried to retrieve the logs yesterday for the first time since. To my confusion, CanZE can't access `/storage/emulated/0/CanZE` anymore, verified with `_Research.csv`. I scratched my head and tried to look into the private storage of the app, but of course it's barred for non-root user and the error message for Research clearly states it tries to reach that normal storage folder. After poking the Interwebs about what might have broken it, I came across [this article](https://www.raywenderlich.com/9577211-scoped-storage-in-android-10-getting-started) explaining that storage works very differently in Android 10 and this will also be permanent in Android 11. If I understand it correctly, there won't be means to access the `CanZE` folder directly and the external storage path will instead lead to `/storage/emulated/0/Android/data/lu.fisch.canze` or something.

I wanted to make a quick PR with that line for the manifest before a more definite solution is made for the looming Android 11, but after building CanZE against Android 10 SDK it actually picked up the new path and it also works (i.e. I can put a file with `adb` there and CanZE can read it, so I assume logging also works, too cold to go outside and verify). I have a very blurry idea of the app publishing process and whether you can have multiple APKs targeting their own SDK version each so instead of making an odd PR I thought I should just report this all in text.

To reiterate the problem: logging and research is kaput on Android 10 when using the Google Play build.

yoh-there, 2020-12-07:
See issue #627 Please close if that is the solution to your problem, if not, please elaborate. Thanks!

landswellsong, 2020-12-07:
@yoh-there oops, I should have read the blog :) Good thing I didn't try to fix it myself. It's probably not the SDK but your changes instead than that fix it since I've built the `master`

closed 2020-12-07 by landswellsong

yoh-there, 2020-12-07:
Glad it works. Just to make sure: the master branch is a representation of all releases, including the (open) beta track. We don't maintain a branch for the true non-beta production. The development branch is every single change, very detailed and with often multiple updates per day if "something's brewing" ;-)

landswellsong, 2020-12-07:
Oh, good to know. I just assumed it was like development, because it's the default branch of the repo.

yoh-there, 2020-12-07:
I can imagine that. We released a beta (and thus updated master) about every other day the last week....... ;-)