fesch/CanZE #655: CanZE not working with R110 Experience ZE50 and KONNWEI

a20eac1d, 2020-12-19:
I recently got my new Renault Zoe R110 Experience ZE50 and wanted to read some information with CanZE.

I downloaded the latest version from the Google Play Store. From Amazon I bought this KONNWEI KW902 and plugged it in:

https://www.amazon.de/gp/product/B00W30RMEC

My phone recognizes the dongle but CanZE refuses to work with it. It says the Bluetooth keeps disconnecting and all the debug prints on the bottom "fail".

What can I do to help solve this issue?

a20eac1d, 2020-12-19:
After doing some research I think it may be because of a wrong resistor on the PCB. How do I find out if that is the cause?

Can a lower resistor (20 Ohm) do damage to the OBD system if its supposed to be 120 Ohm?

yoh-there, 2020-12-19:
See https://canze.fisch.lu/dongle-trouble-4-solution-sort-of/

a20eac1d, 2020-12-19:
> 
> 
> See https://canze.fisch.lu/dongle-trouble-4-solution-sort-of/

Thank you for your quick reply!

Yes, I found that already. I'm not too familiar with electronics but I can measure the resistance across pins 6 and 14. If the multimeter shows ~20 Ohm I know that I have the wrong resistor?

Also, I already plugged in the device without checking it first. Can a resistor this low damage the cars electronics at all? Low resistance basically means a short, right?

Also, why does cutting the resistor in the middle fix the issue? I'm creating a dead end and disabling some functions, no?

yoh-there, 2020-12-19:
1) yes
2) no, car won't be happy, but no permanent damage
3) no, you're creating a dead open!
4) no

yoh-there, 2020-12-21:
I am closing this issue. Feel free to reopen or comment on it while closed.

closed 2020-12-21 by yoh-there