fesch/CanZE #662: Support for Renault Captur Etech Plugin

jumpjack, 2020-12-22:
Is it possible to add support for Renault Captur Etech Plugin? I tried couple a dozen of apps, none can read electrical parameters, apart from battery voltage.

yoh-there, 2020-12-22:
The app is specifically made for ZOE. None of the team members own such a car. A car donation can change that of course but.......

Seriously though. If you want to have a peek at some parameters I would recommend having a peek at ddt4all ( https://github.com/cedricp/ddt4all ). A bit of a different beast but it might do exactly what you want.

There is nothing we can do in CanZE for you at the moment, sorry.

closed 2020-12-22 by yoh-there

jumpjack, 2020-12-29:
Already checked ddt4all, not working.